arendst/Tasmota #4949: Web interface behavior

lalo-uy, 2019-01-16:
Hardware used :
Sonoff Basic 
pre-compiled release 
5.12.0

STATUS 0
{"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"Ferm8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
{"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T21:52:15","Sleep":0,"BootCount":120,"SaveCount":262,"SaveAddress":"F6000"}}
{"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-07-30T11:41:40","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
{"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"osoplanta","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
{"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
{"StatusNET":{"Hostname":"Ferm8-2566","IPAddress":"192.168.100.8","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"84:F3:EB:49:6A:06","Webserver":2,"WifiConfig":2}}
{"StatusMQT":{"MqttHost":"192.168.100.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_496A06","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
{"StatusTIM":{"UTC":"Thu Jan 01 21:52:57 1970","Local":"Thu Jan 01 21:52:57 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1}}
{"StatusSNS":{"Time":"1970-01-01T21:52:57"}}
{"StatusSTS":{"Time":"1970-01-01T21:52:57","Uptime":"0T21:52:15","Vcc":3.170,"POWER":"OFF","Wifi":{"AP":1,"SSId":"osoplanta","RSSI":90,"APMac":"C8:3A:35:EA:48:B8"}}}

If I excute
> wget  -O - -q http://192.168.100.8/cm?cmnd/=Status%200

I got:
{"WARNING":"Enable weblog 2 if response expected"}

Even with WebLog in 2 or 4

ascillato, 2019-01-16:
Sorry. Duplicated issue. Please, search in issues. Already addressed. Thanks

meingraham, 2019-01-16:
You have an extra slash before the equal sign ('=') in your statement. It should be

`wget -O - -q http://192.168.100.8/cm?cmnd=Status%200`

closed 2019-01-16 by ascillato2