arendst/Tasmota #4950: Missing sensor in JSON messages when multiple sensors connected

qingz2004, 2019-01-16:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
Yes.
I have two different type of sensors in one device. Sometimes missing one sensor in MQTT message which causes a lot of error messages in the openhab logs.

**Describe the solution you'd like**
An option to send the previous good data of the sensor if the there is an error reading from the sensor.

**Describe alternatives you've considered**
No JSON message at all instead of partial message.

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

qingz2004, 2019-01-16:
It happens on my Sonoff Pow.
The messages with missing sensor are outside of the 120s teleperiod.

> 11:10:47 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:10:47","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":1,"Factor":0.00,"Voltage":119,"Current":0.000}}
> 11:10:57 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:10:57","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.000}}
> 11:11:51 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:11:51","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Period":0,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.000},"AM2301":{"Temperature":19.4,"Humidity":34.0},"TempUnit":"C"}
> 11:12:20 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:12:20","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":1,"Factor":0.00,"Voltage":119,"Current":0.000}}
> 11:12:30 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:12:30","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.051}}
> 11:13:54 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:13:54","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Period":0,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.000},"AM2301":{"Temperature":19.4,"Humidity":34.0},"TempUnit":"C"}
> 11:15:58 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:15:58","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Period":0,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.000},"AM2301":{"Temperature":19.5,"Humidity":34.0},"TempUnit":"C"}
> 11:17:28 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:17:28","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":1,"Factor":0.20,"Voltage":119,"Current":0.051}}
> 11:17:32 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:17:32","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":3,"Factor":0.60,"Voltage":119,"Current":0.045}}
> 11:17:43 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:17:43","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":0,"Factor":0.00,"Voltage":108,"Current":0.000}}
> 11:18:01 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:18:01","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Period":0,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.000},"AM2301":{"Temperature":19.5,"Humidity":34.0},"TempUnit":"C"}
> 11:18:34 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:18:34","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":2,"Factor":0.00,"Voltage":119,"Current":0.000}}
> 11:18:43 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:18:43","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.046}}
> 11:20:05 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:20:05","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Period":0,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.000},"AM2301":{"Temperature":19.4,"Humidity":33.9},"TempUnit":"C"}
> 11:20:46 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:20:46","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":1,"Factor":0.00,"Voltage":119,"Current":0.000}}
> 11:20:56 MQT: tele/POW/SENSOR = {"Time":"2019-01-16T11:20:56","ENERGY":{"Total":44.657,"Yesterday":0.203,"Today":0.009,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.045}}
>

Jason2866, 2019-01-16:
Sensors on POW are not supported since this very dangerous!
@ascillato2  or @andrethomas close and lock this.
THX

qingz2004, 2019-01-16:
It's not dangerous. Am2301 is isolated, and both POW and AM2301 are inside a cage. Nobody can touch it.

andrethomas2, 2019-01-16:
I cannot resist closing this Issue by quoting @arendst 

> If you continue on this path I would like to send you my best wishes in your afterlife...
> 
> As stated on the pow wiki page never connect a sensor to a pow. Your choosen sensor is my favorite as the casing is connected to Gnd/AC power on Pow.
> 
> May the force be with you (and wear a rubber underpants).
>

closed 2019-01-16 by andrethomas2

ascillato2, 2019-01-16:
Never connect a sensor to a pow.

>Have you look for this feature in other issues and in the wiki?
>Yes.

Please, next time do the search for real. There are big warning messages about this. 

It is too DANGEROUS