arendst/Tasmota #4952: MQTT TLS Verify Fingerprint Fail with Core 2.5.0 PlatformIO

majherek, 2019-01-16:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-01-16:
Are you using latest version from today of the core 2.5.0?

The core 2.5.0 is in beta. It is not released yet and it is in constant updating.

ascillato2, 2019-01-16:
Please, try the latest core. Remember that platform.io don't sync with the files from the repository. So, you have to mannually copy and paste the files. (this is due to the core is still in development)

andrethomas, 2019-01-17:
I think PR https://github.com/arendst/Sonoff-Tasmota/pull/4703 may bear reference

majherek, 2019-01-17:
@ascillato2: I have core 2.5.0 from few days ago. Visual Studio Code download if using git. I will try today to remove core and download it again. Should I take it from: https://github.com/Jason2866/platform-espressif8266.git#Tasmota?

@Jason2866: I will change to stage and also remove it before, to download new one. Also, should I take it from: https://github.com/platformio/platform-espressif8266.git#feature/stage?
Should I linked it using 1M? -Wl,-Teagle.flash.1m.ld?

ascillato2, 2019-01-17:
You should take the core directly from the source in arduino repository (https://github.com/esp8266/Arduino). There is the latest and the install instructions are there also. The other 2 repositories you say are old copies and may not work as they don't have latests fixes.

ascillato2, 2019-01-17:
>Should I linked it using 1M? -Wl,-Teagle.flash.1m.ld?

That only depends on the device you are using. The default works fine.

majherek, 2019-01-17:
I follow the instrucion [Using Arduino Framework with Staging version](http://docs.platformio.org/en/latest/platforms/espressif8266.html?utm_source=github&utm_medium=arduino-esp8266#using-arduino-framework-with-staging-version). Updated the Core to newest version. 

And it still not working. 
```
00:00:00 Project sonoff Sonoff 12 Version 6.4.1.9(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff-0476...
00:00:04 WIF: Connected
22:10:03 MQT: Attempting connection...
22:10:03 MQT: Verify TLS fingerprint...
22:10:03 MQT: Failed
22:10:06 CMD: STATUS 0
22:10:06 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff 12"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:10:06 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff-ota.majchrowski.waw.pl:8888/api/arduino/sonoff12.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:09","StartupUTC":"2019-01-17T21:09:57","Sleep":50,"BootCount":14,"SaveCount":16,"SaveAddress":"3F4000"}}
22:10:06 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.9(sonoff)","BuildDateTime":"2019-01-17T22:08:10","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
22:10:06 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":0,"SysLog":2,"LogHost":"192.168.199.251","LogPort":514,"SSId":["atomix","MAJCOMNET HTC"],"TelePeriod":60,"SetOption":["00008009","558180C0","00000000"]}}
22:10:06 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":466,"Free":2604,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","06082744","040003A0","000004C6","000000C0"]}}
22:10:06 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0476","IPAddress":"192.168.20.12","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.199.251","Mac":"EC:FA:BC:14:81:DC","Webserver":0,"WifiConfig":4}}
22:10:06 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.majchrowski.waw.pl","MqttPort":8883,"MqttClientMask":"sonoff12","MqttClient":"sonoff12","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:10:06 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 17 21:10:06 2019","Local":"Thu Jan 17 22:10:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99}}
22:10:06 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-17T22:10:06"}}
22:10:06 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-17T22:10:06","Uptime":"0T00:00:09","Vcc":3.362,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","BSSId":"06:8D:DB:DB:97:47","Channel":11,"RSSI":94}}}

[core_stage]
; *** Esp8266 core for Arduino version latest beta
platform                  = https://github.com/platformio/platform-espressif8266.git#feature/stage
build_flags               = ${esp82xx_defaults.build_flags}
                            -Wl,-Teagle.flash.4m1m.ld
; lwIP 1.4 (Default)
;                            -DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH
; lwIP 2 - Low Memory
;                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
; lwIP 2 - Higher Bandwidth
;                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
; lwIP 2 - Higher Bandwitdh Low Memory no Features
                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH
; lwIP 2 - Higher Bandwitdh no Features
;                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH
; VTABLES in Flash (default)
                            -DVTABLES_IN_FLASH
; VTABLES in Heap
;                            -DVTABLES_IN_DRAM
; VTABLES in IRAM
;                            -DVTABLES_IN_IRAM
; enable one option set -> No exception recommended
; No exception code in firmware
                            -fno-exceptions
                            -lstdc++
; Exception code in firmware /needs much space! 90k
;                           -fexceptions
;                           -lstdc++-exc

[core_active]
; Select one core set for platform and build_flags
;platform                  = ${core_2_3_0.platform}
;build_flags               = ${core_2_3_0.build_flags}
;platform                  = ${core_2_4_2.platform}
;build_flags               = ${core_2_4_2.build_flags}
;platform                  = ${core_2_5_0.platform}
;build_flags               = ${core_2_5_0.build_flags}
platform                  = ${core_stage.platform}
build_flags               = ${core_stage.build_flags}
```

ascillato, 2019-01-17:
Ok, thanks for testing and reporting :+1:

Jason2866, 2019-01-18:
Could this today merged PR maybe fix it?
https://github.com/arendst/Sonoff-Tasmota/pull/4967

majherek, 2019-01-19:
This PR #4967 use MDNS_HOSTNAME. 
And there is no #define MDNS_HOSTNAME in source code. 
This only change EspClient.verify (fingerprintX,  HOSTNAME).

```
majherek@SOKAR:~/win/Documents/Arduino/Sonoff-Tasmota/sonoff$ grep MDNS_HOST * -r
xdrv_02_mqtt.ino:    #ifdef MDNS_HOSTNAME
xdrv_02_mqtt.ino:      if (!strcmp(MDNS.hostname(i).c_str(), MDNS_HOSTNAME)) {
xdrv_02_mqtt.ino:    #endif  // MDNS_HOSTNAME
xdrv_02_mqtt.ino:#ifdef MDNS_HOSTNAME
xdrv_02_mqtt.ino:    else if (EspClient.verify(fingerprint1, MDNS_HOSTNAME)) {
xdrv_02_mqtt.ino:    else if (EspClient.verify(fingerprint2, MDNS_HOSTNAME)) {
xdrv_02_mqtt.ino:#endif  // MDNS_HOSTNAME
```

If I don't define MDNS_HOSTNAME it will check fingerprintX with Settings.mqtt_host, otherwise with MDNS_HOSTNAME . That's all.

And I want to use of USE_MQTT_TLS_CA_CERT. 

In Core 2.5.0 TLS verification doesn't work either with USE_MQTT_TLS_CA_CERT, nor with USE_MQTT_TLS only.

Jason2866, 2019-01-19:
@majherek  Open a issue in https://github.com/esp8266/Arduino with a example sketch.
Since this is not under control from Tasmota nothing can be done from Tasmota side.
THX!

andrethomas, 2019-01-19:
@fmeies Can you please confirm if this is core issue or use case issue since you made the original PR? Thanks

DavidFW1960, 2019-01-20:
Am I doing this wrong? I am downloading the latest Development Branch ZIP file of Tasmota and I edit the platformio.ini file to use core 2.0.5
When I then compile the firmware and flash it, in the Tasmota information it shows this:
[image]
Yet if I was to download the precompiled 2.5.0 binaries, the core is shows as 2.5.0 as per this:
[image]
So am I doing this wrong and getting an old version of core 2.5.0? Am I supposed to download arduino seperately? I thought Visual Studio Code with PlatformIO IDE automatically got the latest core 2.5.0 when it compiles the firmware?

ascillato, 2019-01-20:
You are doing all ok.

Core 2.5.0 Stage version is newer than core 2.5.0 beta2. Beta2 is a prerelease and stage contain the latest fixes. Use stage. It is better.

DavidFW1960, 2019-01-20:
OK thanks!
My build flags are set like this:
```
[core_active]
; Select one core set for platform and build_flags
;platform                  = ${core_2_3_0.platform}
;build_flags               = ${core_2_3_0.build_flags}
;platform                  = ${core_2_4_2.platform}
;build_flags               = ${core_2_4_2.build_flags}
platform                  = ${core_2_5_0.platform}
build_flags               = ${core_2_5_0.build_flags}
;platform                  = ${core_stage.platform}
;build_flags               = ${core_stage.build_flags}
```
but it's still getting stage so I guess that's all OK and I can stop worrying. Thanks.

ascillato, 2019-01-20:
Remember that platform.io will download stage only once. It will not update to latest commits. As it is the edge version of the core (the latest of the latest) you need to manually copy and paste the files of the core

majherek, 2019-01-20:
@fmeies I use USE_MQTT_TLS_CA_CERT in this case. But this problem exists both using USE_MQTT_TLS_CA_CERT and without using USE_MQTT_TLS_CA_CERT (using mqttfingerprint1 and mqttfingerprint2). And it exists on Core 2.5.0 (with Core 2.4.2 everything is OK). 

```
20:32:07 MQT: Verify TLS fingerprint...
20:32:07 MQT: Verified using FingerprintCA
20:32:07 MQT: Connected
```

I will open issue on [https://github.com/esp8266/Arduino](https://github.com/esp8266/Arduino)

Jason2866, 2019-01-20:
Is this related?

https://github.com/esp8266/Arduino/issues/5640

majherek, 2019-01-27:
I don't think so...

majherek, 2019-01-27:
I open an issue [#5680](https://github.com/esp8266/Arduino/issues/5680)

sislakd, 2019-01-28:
I was facing this issue right after Christmas when I've updated my 30+ devices to firmware based on Tasmota v6.4.1 with Arduino Core 2.5.0 beta 2. After few hour troubleshooting and deep dive into Arduino sources, I've found that the root cause is change of default TLS implementation in Arduino Core from legacy AxTLS to new BearTLS. Initially, I've changed Tasmota to work with new BearTLS, however this was not good as BearTLS implementation requires much more memory causing issues with OTA upgrades which I need. Thus, finally I've updated Tasmota to use legacy AxTLS. Here is diff you need apply to your Tasmota v6.4.1 code:

diff --git a/sonoff/sonoff.ino b/sonoff/sonoff.ino
index 5e51b219..d5e80ea6 100644
--- a/sonoff/sonoff.ino
+++ b/sonoff/sonoff.ino
@@ -33,6 +33,16 @@
 #ifdef USE_CONFIG_OVERRIDE
   #include "user_config_override.h"         // Configuration overrides for my_user_config.h
 #endif
+
+#ifdef USE_MQTT_TLS
+#ifdef ARDUINO_ESP8266_RELEASE_2_5_0_BETA2
+#define USING_AXTLS
+#include <ESP8266WiFi.h>
+#include "WiFiClientSecureAxTLS.h"
+using namespace axTLS;
+#endif // ARDUINO_ESP8266_RELEASE_2_5_0_BETA2
+#endif // USE_MQTT_TLS
+
 #include "sonoff_post.h"                    // Configuration overrides for all previous includes
 #include "i18n.h"                           // Language support configured by my_user_config.h
 #include "sonoff_template.h"                // Hardware configuration

At the moment, there is Arduino Core 2.5.0 beta 3, but I haven't found there any optimization around BearTLS. Maybe in the future, it will be usable.

Jason2866, 2019-01-29:
@majherek 
Could please test the latest TLS change in Tasmota? Please close, if it solves the issue.
Thx.

ascillato2, 2019-01-31:
Hi,

Closing issue as there is no feedback. Please, when you have time, tell us if your issue is solved. Thanks.

closed 2019-01-31 by ascillato2

majherek, 2019-01-31:
Please open it again, I will check it on weekend, because one I am in delegation. What can I do about this...

majherek, 2019-02-02:
@Jason2866 Still the same. I use now 2.5.0_BETA3. I saw some changes in support_wifi.ino made by @arendst, but commented out, similar to that made by @sislakd in this [issue](https://github.com/arendst/Sonoff-Tasmota/issues/4952#issuecomment-458097138).

@sislakd Thanks ;-) I use your diff and it works perfectly (after change to BETA3 in my case).

timstanley1985, 2019-05-08:
Hi, I can confirm this issue persists with 2.5.0 final.  I am having WiFi stability issues with 2.3 and 2.4.2 so am desperate to try 2.5 but I am suffering the TLS fingerprint issues detailed above

terba, 2019-09-11:
A quick hack is to set `MqttFingerprint 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`. At the first connection it will be calculated (not correctly) and stored, as I understood. Works for me.