arendst/Tasmota #4953: Add Lonsonho support

konbec, 2019-01-16:
Hello,

is it possible to add Lonsonho smart plug WiFi support for Tasmota firmware?

Thank you in advance 
Konstantin

Jason2866, 2019-01-16:
We dont have the device and no info about....

jziolkowski, 2019-01-16:
Or buy one and send it to one of the devs :)

ascillato2, 2019-01-16:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-16 by ascillato2

Astr0, 2019-02-09:
I have 3 Lonsonho 16A EU power meter smart plugs successfully running Tasmota and probably can write a wiki entry for them. 
In short - all were flashed using tuya-convert with modified Tasmota to make all free Blitzwolf SHP pins user-configurable.
Device profile: Blitzwolf SHP
GPIO0 - Red LED - PWM1
GPIO2 - Blue LED - PWM3
GPIO4 - Green LED - PWM2
Everything works fine, but I had to disassemble one of the sockets cause it had hardware issues from the factory (broken one of the steel poles, dunno how). Haven't probed ESP module for RX/TX/VCC/GND since it was flashed already, but there are a lot of pads from module to main PCB and some of them are not used (have no tracks on the main PCB, just solder pads) so there's a high chance that everything is there.
They have ESP8266EX with FM25F08 8Mb flash, HJL-01 for power monitoring, 4 RGB LEDs, AFE BPD-SS-105DM 16A 250VAC relay.
The case is glued together, then you have to desolder main output connections to get access to mains input poles which you also have to desolder (hard) to take the PCB out, on which relay is also glued to the case using some sticky hot-glue-style glue. Putting it back together is also tricky since everything fits back very tight. And of course, it won't have a factory look/safety after gluing the case back together.

ChocoTonic, 2020-05-22:
did you do this all OTA save for that one that was broken? @Astr0