arendst/Tasmota #4955: PrivateEyePi IOT Wifi Sensor

sqerk, 2019-01-16:
Can I flash Tasmota software to this device?:
http://projects.privateeyepi.com/WIFI-Sensor

Flash directions for its software:
http://projects.privateeyepi.com/WIFI-Sensor/wifi-sensor---firmware-upload

It got the DS18B20 temp sensor, no DHT22 sensor.

ascillato, 2019-01-16:
Hi, 

As this device has an Esp8266, Tasmota could work. You need to try it and set as generic module. Then configure the gpios of the board.

Please, then report back if it works. If works please, add it to the wiki under other devices. 

Thanks

ascillato2, 2019-01-16:
If you need further assistant please find us in the Tasmota Support Chat. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-16 by ascillato2