arendst/Tasmota #4956: Period Always 0 - TelePeriod=300

meingraham, 2019-01-16:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- Development/Compiler/Upload tools used: OTA
- Hardware used : HW-655 & Sonoff S31
- If a pre-compiled release or development binary was used, which one?: Precompiled 6.3.0.16 and 6.4.1
- You have tried latest release or development binaries?: Yes, Release

Period always 0 on TelePeriod generated messages.

HW-655 (Generic) with PZEM-004T attached
```
STATUS = {"Status":{"Module":18,"FriendlyName":["PZEM Power Monitor"],"Topic":"pzem-power_monitor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff-sensors.bin","RestartReason":"Hardware Watchdog","Uptime":"0T17:57:58","StartupUTC":"2019-01-16T03:52:35","Sleep":25,"BootCount":51,"SaveCount":124,"SaveAddress":"F7000"}}
STATUS2 = {"StatusFWR":{"Version":"6.3.0.16(a516b25-sonoff)","BuildDateTime":"2018-12-09T18:24:40","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":2,"LogHost":"myHost","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"SetOption":["00208108","558180C0","000000C0"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":520,"Free":480,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
STATUS5 = {"StatusNET":{"Hostname":"tasmota-dryer_monitor-8082","IPAddress":"192.168.1.nnn","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"<myMAC>","Webserver":2,"WifiConfig":4}}
STATUS6 = {"StatusMQT":{"MqttHost":"myHost","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_<myMAC>","MqttUser":"myUserID","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 16 21:50:33 2019","Local":"Wed Jan 16 16:50:33 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"07:34","Sunset":"17:38"}}
STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
STATUS10 = {"StatusSNS":{"Time":"2019-01-16T16:50:33","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.350,"Yesterday":0.001,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":119,"Current":0.000}}}
STATUS11 = {"StatusSTS":{"Time":"2019-01-16T16:50:33","Uptime":"0T17:57:58","Vcc":3.439,"SleepMode":"Dynamic","Sleep":25,"LoadAvg":39,"Wifi":{"AP":1,"SSId":"mySSID","BSSId":"<myBSSId>","Channel":1,"RSSI":68}}}
```
```
SENSOR = {"Time":"2019-01-16T00:02:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:07:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:12:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:17:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:22:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:27:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:32:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:37:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:42:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:47:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:52:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T00:57:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:02:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:07:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:12:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:17:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:22:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:27:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:32:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:37:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:42:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:47:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
SENSOR = {"Time":"2019-01-16T01:52:53","ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":15.349,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":12 [... cut ...] ,"Current":0.046}}
SENSOR = {"Time":"2019-01-16T16:51:22","ENERGY":{"TotalStartTime":"2019-01-15T04:12:42","Total":235.768,"Yesterday":0.037,"Today":0.027,"Period":0,"Power":1,"ApparentPower":5,"ReactivePower":5,"Factor":0.26,"Voltage":120,"Current":0.044}}
SENSOR = {"Time":"2019-01-16T16:56:27","ENERGY":{"TotalStartTime":"2019-01-15T04:12:42","Total":235.768,"Yesterday":0.037,"Today":0.027,"Period":0,"Power":2,"ApparentPower":7,"ReactivePower":7,"Factor":0.26,"Voltage":119,"Current":0.062}}
```

ascillato, 2019-01-16:
Hi,

It is correct. 

Period in that JSON correspond to the actual energy used since your last boot. It goes to zero every restart. So, it is like energytoday but if you restart the device it goes to zero. 

This was implemented a couple of month ago I think.

andrethomas, 2019-01-16:
Period reports the watts since the last teleperiod occurred. Since you have "Yesterday":0.001,"Today":0.001 it means that only 1 instance of Period = 1 would have been recorded for a 24 hour period.

In the test case below I have set teleperiod to 10 seconds and this is what I get

```
01:29:07 MQT: tele/ups/SENSOR = {"Time":"2019-01-17T01:29:07","ENERGY":{"TotalStartTime":"2018-11-19T12:02:10","Total":91.135,"Yesterday":4.845,"Today":0.240,"Period":0.550,"Power":194.069,"ApparentPower":194.069,"ReactivePower":0.000,"Factor":1.00,"Voltage":218,"Current":0.889}}
01:29:17 MQT: tele/ups/SENSOR = {"Time":"2019-01-17T01:29:17","ENERGY":{"TotalStartTime":"2018-11-19T12:02:10","Total":91.136,"Yesterday":4.845,"Today":0.241,"Period":0.530,"Power":166.800,"ApparentPower":203.674,"ReactivePower":116.900,"Factor":0.82,"Voltage":218,"Current":0.933}}
01:29:27 MQT: tele/ups/SENSOR = {"Time":"2019-01-17T01:29:27","ENERGY":{"TotalStartTime":"2018-11-19T12:02:10","Total":91.136,"Yesterday":4.845,"Today":0.242,"Period":0.540,"Power":166.000,"ApparentPower":198.435,"ReactivePower":108.700,"Factor":0.84,"Voltage":218,"Current":0.909}}
```

0.540W for the last teleperiod makes sense if the power = 198.435W / 60 minutes / 6 to get to my 10 second teleperiod.

```
"ApparentPower":198.435
"Period":0.540

198.435 / 60 / 6 = 0.551208
```

Seems close enough.

So the use case for Period in the telemetry data is to count the number of watts the device has observed since the last teleperiod occured.

andrethomas, 2019-01-16:
To add, in theory if the apparent power was exactly 100 W consistently then Period would report

100 / 60 / 6 = 0.277778

Each subsequent Period value will be the same so I need to add these up in my home automation system which is exactly what I am doing. I do not use the Yesterday, Today etc at all.

andrethomas2, 2019-01-16:
It's 01h40 and I'm still awake... this is not good.

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

andrethomas2, 2019-01-16:
I forgot to mention, to see the fractions like I do you need to perform

`WattRes 3`

closed 2019-01-16 by andrethomas2

meingraham, 2019-01-17:
I have logs for two devices in my original post - an S31 and an HW-655 with a PZEM.

But the S31 has small power consumption each `TelePeriod` all day long yet `Period = 0`. This is actually the device that alerted me to a potential issue.

The PZEM had 0 watts since last restart (BTW, it's restarting periodically due to a Hardware Watchdog - a problem for later). So, OK, `Period = 0` is correct. I just jumped to a conclusion because I saw `Period = 0` for both and thought I had a related issue. Let's put this one aside.

Mike

P.S. It would be beneficial to add a field to the SENSOR message to print out the value of the TelePeriod at the time of those readings. Yes, querying TelePeriod gives you that but not in one nice consolidated payload. Or you could just know what you've set it to. But if you have different devices and you've changed the value on some, then having that information along with the SENSOR readings would be valuable in calculating information such as Watt-Hours, etc.

meingraham, 2019-01-17:
I can confirm that it was a rounding issue. Setting WattRes to 2 shows Period values >0. Thanks.

Thanks for your help.

Mike

I'll post a feature request for adding TelePeriod value to the SENSOR message.

ascillato, 2019-01-17:
Teleperiod is a configuration and it is a constant. I don't think it is needed a constant in a sensor json message

meingraham, 2019-01-17:
@ascillato 

I tried to be more explicit in the feature request I just posted. It's really not `TelePeriod` but the actual interval.

I also just thought about this scenario...
- `TelePeriod` is set to 1500
- A `SENSOR` message is broadcast as a result of the "regular" mechanism when `TelePeriod` transpires.
- Within a few minutes (less than 15), a user implements your "trick" and sends a `TelePeriod 1500` command (could be done manually or as a result of some condition in a Rule or via a home automation system)
- This results in TASMOTA broadcasting a `STATE` and a `SENSOR` message
- Question: what interval is represented for the data in the `SENSOR` message?

andrethomas, 2019-01-17:
@meingraham It would return the data as at the time the "trick" is executed, so Period data, for example, will be reset and start measuring until either the next teleperiod occurs, or the "trick" is executed.

meingraham, 2019-01-17:
Thus the need to return the actual interval in the `SENSOR` message for which the measured data is for since it's not equal to TelePeriod.

andrethomas, 2019-01-17:
@meingraham The "Period" value is the number of watts accrued during the teleperiod. I explained the math part of it in comment https://github.com/arendst/Sonoff-Tasmota/issues/4956#issuecomment-454985573

meingraham, 2019-01-17:
Yes, I was a bit confused about what the Period field was because `Period`/`TelePeriod`/Interval got jumbled in what I was thinking and trying to convey.

What I'm trying to explain is a field that tells you the interval (in seconds) that transpired for which the totals collected and transmitted in the `Period` represents.

arendst, 2019-01-17:
The only value dependant of the Teleperiod elapsed time is the "period" value in Energy. All other values have no relation to TelePeriod elapsed time.

"Oh my, I wished I never invented the period value..."

meingraham, 2019-01-17:
@arendst - but you did ;-)

andrethomas, 2019-01-17:
Period is useful if you don't have anything else messing with the SENSOR output which would likely be the most use case scenario.

My POW's are set to teleperiod 60 and I collect the Period data to perform further calculations and graph in node-red on load, consumption etc as I do not fully trust the totals for Yesterday and Today and I prefer live changing graphs.

I've posted a comment on your other issue on how you can add what you are asking for yourself as a user-defined driver here https://github.com/arendst/Sonoff-Tasmota/issues/4962#issuecomment-455263606