arendst/Tasmota #4960: Tasmota uses hardcoded DNS and time servers - ignores DHCP values

sgofferj, 2019-01-17:
In my network, the DHCP server provides local addresses for DNS servers (option 6) as well as for the time server (option 42). Tasmota ignores both and tries to use hardcoded values (208.67.222.222 for DNS and various different for time) instead.
Consequently, Tasmote is unable to resolve DNS queries and to acquire the time in a closed network.
This is a security issue as Tasmota requires otherwise unnecessary access to the internet for proper functionality.

arendst, 2019-01-17:
From my_user_config.h
```
// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or enter a static IP address
#define WIFI_GATEWAY           "192.168.1.1"     // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.1.1"     // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

// -- Time - Up to three NTP servers in your region
#define NTP_SERVER1            "pool.ntp.org"       // [NtpServer1] Select first NTP server by name or IP address (129.250.35.250)
#define NTP_SERVER2            "nl.pool.ntp.org"    // [NtpServer2] Select second NTP server by name or IP address (5.39.184.5)
#define NTP_SERVER3            "0.nl.pool.ntp.org"  // [NtpServer3] Select third NTP server by name or IP address (93.94.224.67)
```
and yes, NTP servers will need to be configured if the defaults are not being used.

sgofferj, 2019-01-17:
Interesting. I'm using the release binaries. Apparently, 208.67.222.222 is preconfigured in them then.
My firewall is logging tons of outgoing DNS requests to that address and my DHCP server definitely provides the DNS option also to the IoT VLAN.

ascillato2, 2019-01-17:
Hi, NTP is not hardcoded. You can choose the want you want with ntpserver commands. If you see an IP is being dns resolved. You can put an IP also. See wiki and other issues due to this has been addressed before

closed 2019-01-17 by ascillato2