arendst/Tasmota #4962: Add TelePeriod value to SENSOR message

meingraham, 2019-01-17:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
It would be beneficial to add a field to the `SENSOR` message to print out the value of the `TelePeriod` (or more accurately stated, **_the actual interval_** for which the Period reading represents) at the time of those readings in one nice consolidated payload. Having that information along with the `SENSOR` readings would be valuable in deriving other metrics such as Watt-Hours, Average, etc.

**Describe alternatives you've considered**
You could just know what you've set it to... but it may not be the actual interval.

If you have different devices and you've changed the `TelePeriod` value on some, then keeping track needs some sort of separate table/database/hard-coded constants. This is cumbersome and could lead to errors if a change is made on the device and not in this other `TelePeriod` data repository... and it may not be the actual interval.

Querying `TelePeriod` gives you the value... but it may not be the actual interval. Having a Rule that when a `SENSOR` message is sent (what would the trigger be?) to also performs a "`TelePeriod`" to capture the response. This is a lot of manual setup and extra traffic.

Have a home automation hub store the "last `SENSOR` time" and calculate the interval when the new `SENSOR` message arrives. This is not all that difficult but requires one to have a ha hub. If, alternatively, someone is just capturing these messages (e.g., via SysLog) and wants to perform some data analysis, the same "current - last" could be used.

Having the interval in the message would simplify matters considerably.

andrethomas, 2019-01-17:
Do you mean you want the remaining time left on the current teleperiod to be reported?

arendst, 2019-01-17:
So this is only about the Energy used during "period" being the interval between two teleperiod messages as that is the only case this "period" is used.

As it stands it is just a nice to have value generated after TelePeriod seconds. I see no reason to supply this Teleperiod time within the message as it is pure overhead. If you want to see the teleperiod time add a rule which triggers on any energy value and shows teleperiod.

arendst, 2019-01-17:
BTW with low power loads and teleperiod set to 300 you won't hardly see any power usage.

ascillato2, 2019-01-17:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-17 by ascillato2

meingraham, 2019-01-17:
What I had in mind is that the field would be the number of seconds that transpired since the last `SENSOR` message (yes, usually equal to the _CURRENT_ setting of `TelePeriod`, but **_might_** not be) - in other words, the interval for which the metrics were collected for in this instance of the `SENSOR` message.

Yes, perhaps only for ENERGY. But might be useful for other sensors which are collecting totals. Just haven't thought about other possibilities.

andrethomas, 2019-01-17:
I agree with @arendst that it does not serve a purpose to many users... perhaps as an alternative, you could deploy a custom sensor driver using xsns_91_teleperiod.ino, for example, to append this information for your own purposes. This would work by just including that file in your sonoff folder and would not require any other changes in the main code.

andrethomas, 2019-01-17:
So create `xsns_91_teleperiod.ino` with the following content

```
#define XSNS_91 91

uint16_t tele_counter = 0;

void Xsns91_TeleCounterJSON(void)
{
  snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s,\"TelePeriod\":{\"teleperiod\":%u,\"telecurrent\":%u}"),mqtt_data, Settings.tele_period, tele_counter);
  tele_counter = 0;
}

boolean Xsns91(byte function)
{
  boolean result = false;
  switch (function) {
    case FUNC_EVERY_SECOND:
      tele_counter++;
      break;
    case FUNC_JSON_APPEND:
      Xsns91_TeleCounterJSON();
      break;
    }
  return result;
}
```

And you'll get it appended to the SENSOR JSON similar to this

```
19:43:55 MQT: tele/sonoff/SENSOR = {"Time":"2019-01-17T19:43:55","ANALOG":{"A0":0},"TelePeriod":{"teleperiod":10,"telecurrent":10}}
```

Have fun see'ing how many different combinations will give you instances where teleperiod != telecurrent :)

cyclemat, 2019-01-18:
have anybody here an idea to make a single json string for tele/SONOFFPOW/Periode my problem is in openhab i cant find a way to use Period out of the string wihout error when the Sting didnt include it.
so my idea was to create a single publishing for periode over  tele
/SONOFFPOW/Periode.

cyclemat, 2019-01-18:
okay thats too much for me i never had use JSON Transform in a Rule

meingraham, 2019-01-18:
// Separate rule to test for the presence of the PERIOD metric
rule "Power monitor JSON"
when
    Item power_monitor_JSON received update
then
    val myString = power_monitor_JSON.state.toString
    if (myString.contains("Period"))
        power_monitor_period.postUpdate(transform("JSONPATH", "$.ENERGY.Period", myString))
end