arendst/Tasmota #4964: BME680 compensation functions

brunofosi, 2019-01-17:
Is it possible to include the compensation functions that the [https://github.com/BoschSensortec/BME680_driver](url) uses?

> Regarding compensation functions for temperature, pressure, humidity and gas we have two implementations.
> 
> - Integer version
> - floating point version

ascillato, 2019-01-17:
Hi,

Have you checked the BME680 library included in Tasmota? Seems that it is already there, isn't it?

brunofosi, 2019-01-18:
well yes, the [](https://github.com/BoschSensortec/BME680_driver) library used, gots the compensation functions! but i think we need to compile the tasmota firmware whenever we need to compensate the sensor. maybe i got it wrong? I propose a way to compensate the sensor using the CLI or a GUI setting!

ascillato2, 2019-01-18:
Great, thanks for sharing your ideas :+1: 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-18 by ascillato2