arendst/Tasmota #4968: Switch turns off after loosing wifi connection.

sjev, 2019-01-18:
I have an issue with a switch turning off after a wifi reconnection. It was not possible for me to do fix it with changing powerretain parameter. 
I've removed the sonoff from Home Assistant, to make sure that the problem is located on the switch itself.
Configuration: Sonoff basic with Sonoff-Tasmota 6.4.1.8 

I'd appreciate some pointers on how to fix or work-around this issue.

```
08:58:54 MQT: stat/laders/RESULT = {"POWER":"ON"}
08:58:54 MQT: stat/laders/POWER = ON (retained)
09:01:20 MQT: tele/laders/STATE = {"Time":"2019-01-18T09:01:20","Uptime":"0T02:30:16","SleepMode":"Dynamic","Sleep":150,"LoadAvg":5,"POWER":"ON","Wifi":{"AP":1,"SSId":"fry","BSSId":"CC:5D:4E:91:1A:3C","Channel":4,"RSSI":52}}
09:02:00 MQT: tele/laders/UPTIME = {"Time":"2019-01-18T09:02:00","Uptime":"0T02:30:56"}
09:06:20 MQT: tele/laders/STATE = {"Time":"2019-01-18T09:06:20","Uptime":"0T02:35:16","SleepMode":"Dynamic","Sleep":150,"LoadAvg":5,"POWER":"ON","Wifi":{"AP":1,"SSId":"fry","BSSId":"CC:5D:4E:91:1A:3C","Channel":4,"RSSI":48}}
09:07:29 MQT: Attempting connection...
09:07:29 MQT: Connected
09:07:29 MQT: tele/laders/LWT = Online (retained)
09:07:29 MQT: cmnd/laders/POWER = 
09:07:29 MQT: stat/laders/RESULT = {"POWER":"OFF"}
09:07:29 MQT: stat/laders/POWER = OFF (retained)
09:10:27 MQT: Attempting connection...
09:10:27 MQT: Connected
09:10:27 MQT: tele/laders/LWT = Online (retained)
09:10:27 MQT: cmnd/laders/POWER = 
09:10:27 MQT: stat/laders/RESULT = {"POWER":"OFF"}
```

Jason2866, 2019-01-18:
Please fill out the issue template.  `status 0` is needed !

Yali0n, 2019-01-18:
Hi!

You can try
"wificonfig 5"

With this setup your sonoff wont restart at wifi-loss.

It also looks like you have some retained parameters set.

There is a great YouTube video how to fix this
    https://www.youtube.com/watch?v=31IyfM1gygo

ascillato2, 2019-01-18:
Closing this issue as it has been answered.

If you need further assistant please find us in the Tasmota Chat. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-18 by ascillato2

mr-varga, 2022-12-02:
I have the same problem, with some device, after wifi reconection the device turn off. Here is the log:
`01:01:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:01:52","Uptime":"0T00:10:12","UptimeSec":612,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}
01:01:52 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:01:52","Switch1":"ON"}
01:06:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:06:52","Uptime":"0T00:15:12","UptimeSec":912,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:07"}}
01:06:52 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:06:52","Switch1":"ON"}
01:11:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:11:52","Uptime":"0T00:20:12","UptimeSec":1212,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}
01:11:52 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:11:52","Switch1":"ON"}
01:16:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:16:52","Uptime":"0T00:25:12","UptimeSec":1512,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":45,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}
01:16:52 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:16:52","Switch1":"ON"}
01:21:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:21:52","Uptime":"0T00:30:12","UptimeSec":1812,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:07"}}
01:21:52 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:21:52","Switch1":"ON"}
01:26:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:26:52","Uptime":"0T00:35:12","UptimeSec":2112,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:07"}}
01:26:52 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:26:52","Switch1":"ON"}
01:31:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:31:52","Uptime":"0T00:40:12","UptimeSec":2412,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:07"}}
01:31:52 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:31:52","Switch1":"ON"}
01:35:43 WIF: Connecting to AP1 Mr-Varga_Admin Channel 13 BSSId 7E:8A:20:B7:F9:97 in mode 11n as lusala2...
01:35:45 MQT: Attempting connection...
01:35:45 MQT: Connected
01:35:45 MQT: tele/lusala2/LWT = Online (retained)
01:35:45 MQT: cmnd/lusala2/POWER = 
01:35:46 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:35:46","Uptime":"0T00:44:06","UptimeSec":2646,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"RSSI":100,"Signal":-47,"LinkCount":2,"Downtime":"0T00:00:09"}}
01:35:46 MQT: stat/lusala2/RESULT = {"POWER":"OFF"}
01:35:46 MQT: stat/lusala2/POWER = OFF (retained)
01:35:46 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:35:46","Uptime":"0T00:44:06","UptimeSec":2646,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"RSSI":96,"Signal":-52,"LinkCount":2,"Downtime":"0T00:00:09"}}
01:35:46 MQT: stat/lusala2/RESULT = {"Time":"2022-12-02T01:35:46","Uptime":"0T00:44:06","UptimeSec":2646,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"RSSI":96,"Signal":-52,"LinkCount":2,"Downtime":"0T00:00:09"}}
01:35:46 MQT: stat/lusala2/STATUS11 = {"StatusSTS":{"Time":"2022-12-02T01:35:46","Uptime":"0T00:44:06","UptimeSec":2646,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"RSSI":100,"Signal":-49,"LinkCount":2,"Downtime":"0T00:00:09"}}}
01:35:47 MQT: stat/lusala2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:44:07","StartupUTC":"2022-12-01T23:51:40","Sleep":50,"CfgHolder":4617,"BootCount":50,"BCResetTime":"2020-04-05T18:07:11","SaveCount":1741,"SaveAddress":"F8000"}}
01:36:51 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:36:51","Uptime":"0T00:45:11","UptimeSec":2711,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"RSSI":100,"Signal":-48,"LinkCount":2,"Downtime":"0T00:00:09"}}
01:36:51 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:36:51","Switch1":"ON"}
[... cut ...] 00:00","Timezone":99,"Sunrise":"08:23","Sunset":"16:55"},"StatusSNS":{"Time":"2022-12-02T01:46:25","Switch1":"ON"},"StatusSTS":{"Time":"2022-12-02T01:46:25","Uptime":"0T01:55:27","UptimeSec":6927,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}`

The device in a Sonoff basic coneccted to Unifi access-point.