arendst/Tasmota #4969: How to retrive whole Rule definition?

mkh595, 2019-01-18:
Rule can be up to 511 characters long but only 422 characters are displayed. Same result in console or http comand.
[image]
Is there way how to retrive whole definition? I need modify rule but don't want to lose part which is not displayed.

arendst, 2019-01-18:
Use MQTT

ascillato2, 2019-01-18:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-18 by ascillato2

mkh595, 2019-01-18:
It works.
Personaly I would prefere 5x 300ch rules instead of 3x 511.

Thank you.

bod-bod, 2020-07-29:
Hi. Is this a valid solution yet (8.3.1)? The mqtt RESULT message is not complete too.