arendst/Tasmota #4973: SonOff  doesn't go off after adding switch

UserSander, 2019-01-18:
Hello, i've been useing 3 Tasmota Sonoff s20's switch for a while now and it works as normal with HASSio. 
Yesterday i added an door switch (magnetic reed switch) to the GPIO pins. I solderd the RX GPIO3 and the ground to the switch. As espected the relay clicks when the door closes and opens. But i want the switch to be stand alone. i've followed this tutorial of The Hook up. And now the sonoff goes on but does not go off in HomeAssistant. It works then you toggle off/on in the WebUI. 
So i don't know whats wrong. 

I heard to avoid the GPIO3 at all. But does this somehow kill the sonoff?

The link of the tutorial:
[https://www.youtube.com/watch?v=qZqmkhpJ55A&t=2s](url)
My code
`binary_sensor.yaml  
- platform: mqtt
    name: "Achterdeur"
    state_topic: "cmnd/sonoff01switch/POWER2"
    payload_on: "ON"
    payload_off: "OFF"
    device_class: door

Lights.yaml
  - platform: mqtt
    name: "Lamp tv"
    state_topic: "stat/sonoff01/POWER"
    command_topic: "cmnd/sonoff01/POWER"
    availability_topic: "tele/sonoff01/LWT"
    optimistic: false
    qos: 2
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true`

ascillato2, 2019-01-18:
Hi, please provide the troubleshooting template.

UserSander, 2019-01-18:
Here it is

[home-assistant.log](https://github.com/arendst/Sonoff-Tasmota/files/2774491/home-assistant.log)

ascillato2, 2019-01-18:
Sorry, the troubleshooting template of Tasmota repository

ascillato, 2019-01-18:
The templeate is:

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-18:
So, summary:

Use latest Tasmota

Set the GPIO you use for a reed as switch2 (for example)

Set the `switchmode2 1` to follow mode (will result in giving you 1 for ON and 0 for OFF)

Set the rules:

```
rule1 1
rule1 on switch2#state=0 do publish stat/MY_SONOFF_TOPIC_FOR_REED/REED OFF endon on switch2#state=1 do publish stat/MY_SONOFF_TOPIC_FOR_REED/REED ON endon
```

and in Home Assistant:

```yaml
platform: mqtt
name: "Achterdeur"
state_topic: "cmnd/MY_SONOFF_TOPIC_FOR_REED/REED"
payload_on: "ON"
payload_off: "OFF"
```

closed 2019-01-18 by ascillato2