arendst/Tasmota #4974: Quick closing of issues, questions, etc

crispy78, 2019-01-18:
I really like Tasmota, the thing I dislike is the way people manage issues, help requests, etc.
It looks like some are eagerly trying to close as many “tickets” as possible, after reading two words they provide the person with an answer and close the “ticket”. Adding some clarification as a response is kind of pointless because the “ticket” has been closed and nobody’s interested anymore.
What platform should I use in case I have questions and need some help from people that are willing to do more than just quickly close a “ticket”?

meingraham, 2019-01-18:
Discord TASMOTA Chat - https://discordapp.com/channels/479389167382691863/479389167382691865

ascillato2, 2019-01-18:
Sorry for your frustration.

The issues you opened didn't have the requested template and also they were duplicated issues. You should follow the template and search in issues and in the wiki.

Github issues are for bug reporting.

For support there is the Tasmota Support Chat.

Please, chill and enjoy.

closed 2019-01-18 by ascillato2

ascillato2, 2019-01-18:
@meingraham 

The link you provided to discord is wrong. You need to provide the invitation link: https://discord.gg/Ks2Kzd4