arendst/Tasmota #4976: Gosund WP3 US Socket - Button Press Problem

meingraham, 2019-01-19:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- Development/Compiler/Upload tools used: OTA
- Hardware used: Gosund WP3
- If a pre-compiled release or development binary was used, which one?: 6.4.1 (2.3.0) Release
- You have tried latest release or development binaries?: Both 6.4.1 and 6.4.1.9
```
STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"tasmota-gosundPlug","ButtonTopic":"tasmota-gosundPlug","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/","RestartReason":"Software/System restart","Uptime":"0T00:06:06","StartupUTC":"2019-01-19T02:18:52","Sleep":250,"BootCount":15,"SaveCount":37,"SaveAddress":"F9000"}}
STATUS2 = {"StatusFWR":{"Version":"6.4.1(release-sonoff)","BuildDateTime":"2018-12-24T15:51:34","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"SetOption":["00008108","558180C0","000000C0"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
STATUS5 = {"StatusNET":{"Hostname":"tasmota-gosundPlug-5244","IPAddress":"192.168.1.nnn","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"<myMAC>","Webserver":2,"WifiConfig":5}}
STATUS6 = {"StatusMQT":{"MqttHost":"myHost","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_<myMAC>","MqttUser":"myUserID","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 19 02:24:58 2019","Local":"Fri Jan 18 21:24:58 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"07:33","Sunset":"17:40"}}
STATUS10 = {"StatusSNS":{"Time":"2019-01-18T21:24:58"}}
STATUS11 = {"StatusSTS":{"Time":"2019-01-18T21:24:58","Uptime":"0T00:06:06","Vcc":3.178,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":7,"POWER":"ON","Wifi":{"AP":1,"SSId":"mySSID","BSSId":"<myBSSId>","Channel":1,"RSSI":100}}}
```
**(Please, remember to close the issue when the problem has been addressed)**
[image]
I have also tried `Button1n` for GPIO4.

The issue is that when I press the button in the socket TASMOTA is seeing a "multi-press"
`APP: Button1 multi-press 1`
and it takes a long time for the socket to respond and toggle the state of the relay. Is there anything I can do to remove the lag and also not have the button cause a "multi-press"?

Thanks.

Mike

meingraham, 2019-01-19:
Based on the advice of @digiblur, I changed GPIO4 to Switch1 and set `SwitchMode1 5`.
[image]

closed 2019-01-19 by meingraham

ascillato, 2019-01-19:
You can try also

Setoption1 1 
Setoption13 1