arendst/Tasmota #4978: Tasmota Rules, Relay Status, MQTT

HobbytronicsPK, 2019-01-19:
I am working with Sonoff Rules and want to send the current relay status on a specific topic every minute. My rules is as given below.
`rule1 on Time#Minute do Publish watchdog {"id":16,"val":Current_Relay_State} endon`

May someone please help me to fill Current_Relay_State with actual state of relay? I am also thinking of setting a variable such as var1 on event of switching the state with the current state, but this variable may get lost, if Sonoff reboots (Am I correct?).

ascillato, 2019-01-19:
For sending the status of relays you don't need a rule. Just use teleperiod 60

ascillato, 2019-01-19:
The full list of variables and triggers are in the wiki at rules.

If you need the exact publish you did, you can use a var to store the power state.

```
Rule1 1 
Rule1 on Power1#boot do var1 %value% endon on Power1#state do var1 %value% endon on Time#Minute do Publish watchdog {"id":16,"val":%var1%} endon
```

ascillato2, 2019-01-19:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-19 by ascillato2

HobbytronicsPK, 2019-03-22:
@ascillato Can we change `on Time#Minute` to every 10 seconds?

ascillato, 2019-03-22:
Sorry, time#minute is just one minute.
If you want a trigger every 10 seconds you can use teleperiod

HobbytronicsPK, 2019-03-22:
@ascillato got it. I have one more problem. I have the following rule set.
`Rule1 on Power1#boot do var1 %value% endon on Power1#state do var1 %value% endon on Time#Minute do Publish watchdog {"idx":17,"nvalue":%var1%} endon`
Now problem is value of var1 is not set at boot. and state is publish as empty `"{"idx":17,"nvalue":}"`. However when i change the switch state then published value changed as expected `"{"idx":17,"nvalue":1}"`.

HobbytronicsPK, 2019-03-24:
Issue has been resolved by using `System#Boot` trigger. Here is an example rule for reference.
`Rule1 on System#Boot do var1 %value% endon on System#Boot do Publish watchdog {"idx":1,"nvalue":%var1%} endon on Power1#state do var1 %value% endon on Time#Minute do Publish watchdog {"idx":1,"nvalue":%var1%} endon`