arendst/Tasmota #4979: Typo in "Important User Compilation Information"

Mikka20, 2019-01-19:
The installation guide contains a typo:
Important User Compilation Information
....
To make compile time changes to Sonoff-Tasmota it can use the user_config_override.h file. It assures keeping your settings when you download and compile a new version. To use user_config.override.h you will have to make a copy of the provided **user_config.override_sample.h** file and ...

It should be:
Important User Compilation Information
....
To make compile time changes to Sonoff-Tasmota it can use the user_config_override.h file. It assures keeping your settings when you download and compile a new version. To use user_config.override.h you will have to make a copy of the provided **user_config_override_sample.h** file and ...

THX

ascillato, 2019-01-19:
The wiki is open. Please edit the typo. Thanks.

closed 2019-01-19 by ascillato2

Mikka20, 2019-01-19:
The changes must be made in README.md

ascillato, 2019-01-19:
Ah ok, so you can make a Pull Request.

Go to the readme and Click in the upper right the pencil. Do the edit and then click on make pull request. 

Thanks for helping :+1:

Mikka20, 2019-01-19:
Done ;-)