arendst/Tasmota #4981: DHT11 Sensor (Deg. C vs. deg. F) Tasmota displays incorrect data/HA different!

SirRedZ, 2019-01-20:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Wemos D1 Mini (clone). Pin D1 for Relay1, PinD2 DHT11, Pin D5 Switch2 (reed switch) 
- [X] _Development/Compiler/Upload tools used_ : OTA updated from 6.3.0 to latest 6.4.1. Previously flashed using Sonoff.bin (ESPEasy)
- [X] _If a pre-compiled release or development binary was used, which one?_ : Sonoff.bin  - release
- [X] _You have tried latest release or development binaries?_ : YES
- [X] _Provide the output of command_``status 0`` :
```

20:01:07 CMD: status 0
20:01:07 MQT: stat/garagedoor/STATUS = {"Status":{"Module":18,"FriendlyName":["Garagedoor"],"Topic":"garagedoor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"garagedoorstate","SwitchMode":[2,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
20:01:07 MQT: stat/garagedoor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:28:09","StartupUTC":"2019-01-20T00:32:58","Sleep":50,"BootCount":40,"SaveCount":84,"SaveAddress":"F8000"}}
20:01:07 MQT: stat/garagedoor/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
20:01:07 MQT: stat/garagedoor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wbhnet","wbhnet"],"TelePeriod":260,"SetOption":["000280A9","558180C0","00000288"]}}
20:01:07 MQT: stat/garagedoor/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
20:01:07 MQT: stat/garagedoor/STATUS5 = {"StatusNET":{"Hostname":"garagedoor-4039","IPAddress":"192.168.7.50","Gateway":"192.168.7.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"DC:4F:22:55:CF:C7","Webserver":2,"WifiConfig":4}}
20:01:07 MQT: stat/garagedoor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.7.11","MqttPort":1883,"MqttClientMask":"garagedoor","MqttClient":"garagedoor","MqttUser":"rksreygm","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:01:07 MQT: stat/garagedoor/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 20 01:01:07 2019","Local":"Sat Jan 19 20:01:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"-05:00","Sunrise":"02:35","Sunset":"11:26"}}
20:01:07 MQT: stat/garagedoor/STATUS10 = {"StatusSNS":{"Time":"2019-01-19T20:01:07","Switch2":"OFF","DHT11":{"Temperature":3.0,"Humidity":40.0},"TempUnit":"C"}}
20:01:07 MQT: stat/garagedoor/STATUS11 = {"StatusSTS":{"Time":"2019-01-19T20:01:07","Uptime":"0T00:28:09","Vcc":3.021,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wbhnet","BSSId":"EC:08:6B:E3:12:60","Channel":11,"RSSI":60}}}

```

**To Reproduce**
_Steps to reproduce the behavior:_

Set GPIO4 (Pin D2) to DHT11 module, reset. Incorrect temperature displayed (3.3 C - actual is minus 15-16 currently). Change HA configuration to deg. F -- Incorrect Temp is displayed in Tasmota Device as above, but correct temperature is displayed in Home Assistant. but in celcius, not F.

**Expected behavior**
- Correct temperature reading in tasmota / Match Home Assistant reading according to setting of Setoption8 (0=C, 1=F) Humidity reading way off, know issue with sensor so I am ignoring.

**Screenshots**

 -- Settings in HA/Tasmota set to Celcius

[image]
[image]
[image]

Changed only Setting in HA to F.
[image]
[image]
[image]
[image]

**Additional context**

Setoption8 (1) displays F as expected, incorrect data value (consistent with C-> F conversion) in Tasmota module, but does not affect HA output.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-20:
Hi,

It is indeed very weird what is happening with your device. I could not reproduce it. Could be a defective sensor? 

Can you test your sensor at indoor temp?

SirRedZ, 2019-01-20:
I am currently trying to reproduce on a secondary setup, using a different sensor as the GarageSensor is attached to my GD opener and I would rather not remove it again if I don't have to. I will experiment and post my results here. TY for very quick reply.

ascillato, 2019-01-20:
Any news on this?

SirRedZ, 2019-01-21:
Hello;

The Temperature sensor must be defective as you assumed - it is now -20.5 degrees celcius here and the sensor is showing -10.3 C. I will be closing this issue until I have received additional items to test with. 

Thank you Adrian!

closed 2019-01-21 by SirRedZ

MichaelMichaelMichaelMichaelMichael, 2020-11-12:
I have comparable results with a ESP01 + DHT11 --> The temperature shows around 10°C higher than it actually is. 
There's 2 tasmotized devices laying on the same table at around 23° C room temperature, the one with a DHT22 is displaying the seemingly correct temperature, the one with DHT11 has the +10°C difference. 

NodeMCU + DHT22, Firmware Version: 8.5.1:
```
AM2301 Temperature	23.1 °C
AM2301 Humidity	        49.6 %
AM2301 Dew point	12.0 °C
```

ESP01 + DHT11, Firmware Version: 9.1.0:
```
DHT11 Temperature	32.7 °C
DHT11 Humidity      	32.0 %
DHT11 Dew point     	13.8 °C
```

What's wrong here? How can I debug this?

Tollbringer, 2020-11-12:
Do you have a pull-up (4.7k or 10k) resistor? The DHT11 is known to be inaccurate, but not by that much. Try using ESPHOME and see if that generates better results (along with pull-up resistor)

Tollbringer

MichaelMichaelMichaelMichaelMichael, 2020-11-12:
[image]
[image]

I'm using this kind of board (tested several with the same result) and I've measured two of them to be 10k resistors. I assume they're wired accordingly, I cannot see that though.

HerraPlagu, 2020-11-25:
> I have comparable results with a ESP01 + DHT11 --> The temperature shows around 10°C higher than it actually is.
> There's 2 tasmotized devices laying on the same table at around 23° C room [...]

Did you solve this? i have exact same issue, Temperature show 10c too much and humidity shows 20% too low. Running tasmota with esp-01 and same shield that you seem to have.  I have these 2 sensors side by side: (screenshot from HA but it shows same values in tasmota webpage. 

[image]

pgk83, 2021-01-25:
[image]

Hi have same board with esp1s and you see this big celsius in the livingroom :) sometimes we are in 250C :) Humidity is normaly and temprature is good but sometimes measure too much value. I try another DHT11 board same issue