arendst/Tasmota #4982: Problems with MQTT Server

asantaclara, 2019-01-20:
Hi! This is my first time in the page.

I have a sonoff basic and a sonoff dual r2, both running the last version of tasmota. Today I entered the console and I realized that both sonoff are constantly reconnecting to the mqtt server. Here is a pice of the log. Can anyone help me to solve this? Sometimes sonoffs when i click the button on the hassio page. Thanks.

05:48:45 DNS: Initialized
05:48:46 MQT: Attempting connection...
05:48:46 MQT: Connected
05:48:46 MQT: tele/SonoffDualTest/LWT = Online (retained)
05:48:46 MQT: cmnd/SonoffDualTest/POWER = 
05:49:06 MQT: Attempting connection...
05:49:06 MQT: Connect failed to 192.168.0.71:1883, rc -2. Retry in 10 sec
05:49:09 DNS: Initialized
05:49:20 MQT: Attempting connection...
05:49:20 MQT: Connected
05:49:20 MQT: tele/SonoffDualTest/LWT = Online (retained)
05:49:20 MQT: cmnd/SonoffDualTest/POWER = 
05:50:46 DNS: Initialized
05:50:47 MQT: Attempting connection...
05:50:47 MQT: Connected
05:50:47 MQT: tele/SonoffDualTest/LWT = Online (retained)
05:50:47 MQT: cmnd/SonoffDualTest/POWER = 
05:51:06 MQT: Attempting connection...
05:51:06 MQT: Connect failed to 192.168.0.71:1883, rc -2. Retry in 10 sec
05:51:09 DNS: Initialized
05:51:20 MQT: Attempting connection...
05:51:20 MQT: Connected
05:51:20 MQT: tele/SonoffDualTest/LWT = Online (retained)
05:51:20 MQT: cmnd/SonoffDualTest/POWER =

markusb68, 2019-01-24:
Same here:

00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.4.1.10(sonoff)-2_5_0_BETA2
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 xxx in mode 11N as sonoff-2113...
...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-2113 with IP address 192.168.2.108
00:00:06 MQT: Attempting connection...
00:00:07 APP: (UTC) Thu Jan 24 23:00:52 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
00:00:56 MQT: Connect failed to 192.168.2.66:1883, rc -2. Retry in 10 sec
00:00:56 HTP: Main Menu

ascillato, 2019-01-24:
Please, see wiki at troubleshooting

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

markusb68, 2019-01-24:
Went to Core 2.3 and made a reset 5.
Still the same:

00:33:05 MQT: Attempting connection...
00:33:11 MQT: Connect failed to 192.168.2.66:1883, rc -2. Retry in 10 sec

What is very strange:
Can't ping tasmota (192.168.2.108) on the Raspi (where the Mosquito server runs)
Other Windows and Linux on the same net can ping 192.168.2.108.

Any idea why the raspi can't ping the 192.168.2.108?

Don't think it's a network problem. With core 2_5_0_BETA2 wifi was stable on the fritz.box

flexiti, 2019-05-02:
back to version 6.4.1.14 and core_2_5_0 helps 
(the problem only occurs sometimes)
In the latest version, when I have connected 10 devices, virtually mqtt restarts every 20 sek

smartboart, 2019-07-08:
I have also this issue but only with generic 18 modules.. Like wemo d1 mini or 5 slot powerpluggs... I use different firmware.. Self-made and the pre compiled firmware.. 6.4.1.18 core 2.5.0 and self made 6.5.0.16 core 2.5.2. All other sonoff devices do not have this issue..