arendst/Tasmota #4983: Strange behavior (PWM + MQTT)

AlexTransit, 2019-01-20:
There are several sonoff basic. firmware 6.4.1 (sonoff) -2_4_2
If in the settings specify PWM and MQTT. Through the Web, quickly change the value of PWM (2 times enough, but quickly) then the device restarts. He writes nothing to the log.
If you disable MQTT then everything works fine.
At first I thought that there was a problem with VCC. Filed 3.3V from a laboratory source. Problem still exists.
If in the address MQTT specify the name, then when you quickly switch PWM I get
`09:18:01 MQT: Connect failed to mqtt.local: 1883, rc -2. Retry in 10 sec`
(start after 2-3 reboot)
What could be the problem? (if I use Rule + MQTT + PWM, then I get a bootloop)

`09:07:32 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-01-06T11:46:23","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
09:07:32 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":0,"LogHost":"192.168.1.3","LogPort":514,"SSId":["AlexHome","AlexHomeR"],"TelePeriod":0,"SetOption":["00008008","558180C0","00000000"]}}
09:07:32 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1365","IPAddress":"192.168.1.159","Gateway":"192.168.1.3","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.3","Mac":"CC:50:E3:02:65:55","Webserver":2,"WifiConfig":0}}
09:07:32 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_026555","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
`

AlexTransit, 2019-01-20:
UPD 
Just flashed 5.14. everything works fine

mike2nl, 2019-01-20:
@AlexTransit 
- Had you flashed via OTA the new firmware?
- Had you done a reset 5 in the console via the web ui?

Jason2866, 2019-01-20:
Disable Sleep (`sleep 0`). PWM on ESP is software based since ESP doesnt have hardware PWM.
PWM needs a lot CPU power.
Since 5.14 there where a lot of stuff added needing resources, watchdog could be a result of from low memory...

Jason2866, 2019-01-20:
Have you measured temperature of ESP8266?  Could be a heat problem.
Some batches of Sonoff Basic does have heat issues

ascillato2, 2019-01-20:
As you are going forward and backwards in versions, may be some incorrect data is being left in flash. Please, try to erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins from development branch at http://thehackbox.org/tasmota/020500/
Thanks

AlexTransit, 2019-01-20:
@mike2nl
flashed via OTA or serial
send command reset 5 in the console
The problem does not solve.
@Jason2866
sleep 0
The problem does not solve.
@Jason2866
Could be a heat problem.
no. does not work immediately after power on. the chip is slightly warm.
@ascillato2
i erase all flash with esptool.py.
now done again.
erase and flash latest "Project sonoff Sonoff Version 6.4.1.9(a055908-sonoff)-2_5_0_BETA2"
The problem does not solve.
I can 100% emulate restart
with MQTT and PWM enabled, you need to quickly change the PWM value through the browser, quickly.
tested on two different devices.

Jason2866, 2019-01-20:
2.5.0 beta2 does NOT work reliable with PWM! You have to use latest stage core.
(@ascillato  see https://github.com/esp8266/Arduino/pull/5578 )
Your device has a problem. Many use PWM without a problem.
What device are you using? Wemos mini clone? There are fake chips around for ESP on it. 
 It can still be a heat problem. Most current is needed at startup. Are you measuring direct on chip or on metal case?

Jason2866, 2019-01-20:
I did moved the slider often very fast. No issue no glitch.
```
17:55:35 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":34}
17:55:35 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":20}
17:55:35 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":12}
17:55:36 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":2}
17:55:36 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":54}
17:55:37 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":84}
17:55:37 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":100}
17:55:37 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":75}
17:55:37 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":58}
17:55:38 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":39}
17:55:38 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":28}
17:55:38 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":18}
17:55:38 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":10}
17:55:39 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":4}
17:55:39 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":47}
17:55:39 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":62}
17:55:39 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":71}
17:55:40 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":82}
17:55:40 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":92}
17:55:40 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":97}
17:55:40 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":88}
17:55:41 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":68}
17:55:41 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":46}
17:55:41 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":36}
17:55:41 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":33}
17:55:42 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":18}
17:55:42 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":3}
17:55:42 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":6}
17:55:42 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":41}
17:55:43 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":60}
17:55:43 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":72}
17:55:43 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":89}
17:55:43 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":93}
17:55:44 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":87}
17:55:44 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":48}
17:55:44 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":26}
17:55:44 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":12}
17:55:45 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":8}
17:55:45 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":91}
17:55:46 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":53}
17:55:46 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":12}
17:55:46 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":86}
17:55:47 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":1}
17:55:47 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":89}
17:55:48 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":67}
17:55:48 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":37}
17:55:49 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":9}
17:55:49 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":81}
17:55:50 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":48}
17:55:50 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":6}
17:55:51 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":74}
17:55:51 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":90}
17:55:51 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":96}
17:55:52 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":9}
17:55:52 MQT: stat/sonoff-53CF5F5F/RESULT = {"POWER":"ON","Dimmer":16}
```
```
17:57:31 CMD: status 0
17:57:31 MQT: stat/sonoff-53CF5F5F/STATUS = {"Status":{"Module":18,"FriendlyName":["NFC"],"Topic":"sonoff-53CF5F5F","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:57:31 MQT: stat/sonoff-53CF5F5F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:21:43","StartupUTC":"2019-01-20T16:35:48","Sleep":50,"BootCount":6,"SaveCount":185,"SaveAddress":"F4000"}}
17:57:31 MQT: stat/sonoff-53CF5F5F/STATUS2 = {"StatusFWR":{"Version":"6.4.1.9(sonoff)","BuildDateTime":"2019-01-20T17:25:38","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
17:57:31 MQT: stat/sonoff-53CF5F5F/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"SetOption":["00008009","55C180C0","00000000"]}}
17:57:31 MQT: stat/sonoff-53CF5F5F/STATUS4 = {"StatusMEM":{"ProgramSize":492,"Free":508,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F8AA784","040103A1","80041496","00002CC2"]}}
17:57:31 MQT: stat/sonoff-53CF5F5F/STATUS5 = {"StatusNET":{"Hostname":"sonoff-53CF5F5F-3935","IPAddress":"192.168.2.163","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:53:CF:5F" [... cut ...]
17:57:31 MQT: stat/sonoff-53CF5F5F/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 20 16:57:31 2019","Local":"Sun Jan 20 17:57:31 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:56","Sunset":"16:51"}}
17:57:31 MQT: stat/sonoff-53CF5F5F/STATUS10 = {"StatusSNS":{"Time":"2019-01-20T17:57:31"}}
17:57:31 MQT: stat/sonoff-53CF5F5F/STATUS11 = {"StatusSTS":{"Time":"2019-01-20T17:57:31","Uptime":"0T00:21:43","Vcc":2.942,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":339,"POWER":"ON","Dimmer":16,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":88}}}
```

andrethomas2, 2019-01-20:
Closing this issue as it has been answered.

For now, PWM only works properly in the staged core - we hope to see a release of beta3 in the next few days then pre-compiled development binaries will also be moved to the beta3 core.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-20 by andrethomas2

AlexTransit, 2019-01-21:
@Jason2866
> What device are you using? Wemos mini clone? There are fake chips around for ESP on it.
This clones fail on high CPU load...

I use original devices.
ordered from the manufacturer's website.
https://itead.taobao.com/

what is your version of the firmware and core?

Jason2866, 2019-01-21:
Latest dev 6.4.1.9 with latest Stage Core 2.5.0  (https://github.com/esp8266/Arduino/tree/cee0796a87afc735760e1f88a9fd68629b0350ad)
core_esp8266_timer.c, core_esp8266_wiring_digital.c and core_esp8266_wiring_pwm.c in folder "sonoff" deleted

AlexTransit, 2019-01-21:
thank. I compared the old Sonoff and the new.
In the new little changed power supply. (minimum voltage 3.42V) 
if make the rule (turn on the relay at startup)
But on the old sonoff it works within a second, and on the new only after wifi is connected.
On the old there is no problem with restarting.
the problem is not in PWM
network problem. perhaps frequent switches send a lot of data and the buffer overflows.
It is not clear why the rule works only after connecting to the network?
take photos of a new device?

Jason2866, 2019-01-21:
This is the behaviour of the bad basic devices.... No need for photo since it is a basic
You can try the stripped software version for this problematic devices
http://thehackbox.org/tasmota/release/020300/sonoff-basic.bin
Here is nearly every sensor support excluded. Just a plain small version.
If this version doesnt work you can use just the old version since MANY features are here not included
and so the CPU has an "easy" job and needs low power only.

To be clear this is NOT a bug in Tasmota. It is bad (cheap) hardware design.