arendst/Tasmota #4985: Xingg Model: W-DE004S

Arkadije, 2019-01-20:
HI all, 
I have Xingg WiFi Smar Socket Model: W-DE004S
I am wondering is it possible to flash Tasmota on it.
Product:
https://www.aliexpress.com/item/EU-European-Smart-Wi-Fi-Power-Plug-Socket-Outlet-Adapter-For-Amazon-Alexa-Echo-Google-Assistant/32850566772.html?spm=2114.13010708.0.0.3a994c4dZvZUk5

[image]
[image]
[image]

ascillato, 2019-01-20:
If it has an esp8266 may be it is possible. You should try. Remember to backup the original firmware.

ascillato2, 2019-01-20:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-20 by ascillato2

ascillato, 2019-01-20:
If you need further assistant please find us in the Tasmota Chat

Arkadije, 2019-01-20:
Thanks for the fast reply.
I am new in all of this, trying to wrap my head around it. 
Flashed couple of Sonoff basic but that is about it.

Maybe it is nube question, but where I can find Chat?