arendst/Tasmota #4993: Sonoff RF for garage door restart

Fulanorro, 2019-01-20:
Hi, I've been struggling for a few days with 2 Sonoff RF devices I pretend to install in my garage door and the outside gate. Both devices show the same behaviour.

Raspberry PI with Openhab server and Mosquitto server.
Sonoff RF with the last Tasmota firmware.
Gpio02: with a momentary switch / button to know if the door is open or closed.

Using the console I set the following configurations:
Switchmode1 1
Switchmode2 2
switchtopic outgatestate 
Pulsetime 7 – so the relay is on 0.7 seconds and goes off
Poweronstate en 3
Powerretain off
Switchretain off
Buttonretain off

At first, when I restarted the Sonoff RF, or there was a power loss, after restarting it activated the relay. This was a big issue, concerning security. However, I managed to solve this with the above configuration and by modifying my mqtt.cfg, retain = false.

But now I'm facing a new issue when the device looses power.

If the Switch2 on Gpio02 is OFF (the door is open), the device starts as usual, no problems, relay stays off.
But if this switch is ON (door closed), the device doesn't start. It's dead. No led blinking. 

Hope you can help me. Thanks.

STATUS 0 OUTPUT HERE
21:00:23 CMD: status 0
21:00:23 MQT: stat/sonoffVA4/STATUS = {"Status":{"Module":2,"FriendlyName":["sonoffVA4","Sonoff2"],"Topic":"sonoffVA4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"outgatestate","SwitchMode":[1,2,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:00:23 MQT: stat/sonoffVA4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:08:56","StartupUTC":"2019-01-20T19:51:27","Sleep":50,"BootCount":48,"SaveCount":111,"SaveAddress":"F5000"}}
21:00:23 MQT: stat/sonoffVA4/STATUS2 = {"StatusFWR":{"Version":"6.4.1.9(sonoff)","BuildDateTime":"2019-01-19T09:30:01","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
21:00:23 MQT: stat/sonoffVA4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Telfy_2487","Telfy_2487"],"TelePeriod":300,"SetOption":["00008008","558180C0","00000000"]}}
21:00:23 MQT: stat/sonoffVA4/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
21:00:23 MQT: stat/sonoffVA4/STATUS5 = {"StatusNET":{"Hostname":"192.168.1.28","IPAddress":"192.168.1.31","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:C1:80:90","Webserver":2,"WifiConfig":4}}
21:00:23 MQT: stat/sonoffVA4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.28","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C18090","MqttUser":"fulanorro","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:00:23 MQT: stat/sonoffVA4/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 20 20:00:23 2019","Local":"Sun Jan 20 21:00:23 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:34","Sunset":"17:28"}}
21:00:23 MQT: stat/sonoffVA4/STATUS10 = {"StatusSNS":{"Time":"2019-01-20T21:00:23","Switch2":"OFF"}}
21:00:23 MQT: stat/sonoffVA4/STATUS11 = {"StatusSTS":{"Time":"2019-01-20T21:00:23","Uptime":"0T00:08:56","Vcc":3.459,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"Telfy_2487","BSSId":"50:1D:93:F7:8A:1C","Channel":10,"RSSI":98}}}

Fulanorro, 2019-01-20:
Please, an admin to erase this issue if he considers it. 
It was my fault. When I restart the sonoff via console the led blinks. If I unplug and plug it and the door is open (switch off), it blinks as well. But if the door is closed (switch on), the led doesn't blink and the console shows nothing. However, the device is working properly.

Jason2866, 2019-01-20:
Just close the issue. Thx.

closed 2019-01-20 by Fulanorro

Jason2866, 2019-01-20:
See why https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#usable-pins