arendst/Tasmota #4996: Touch switch double tap

Lawrencezarb, 2019-01-21:
I have a sonoff basic with a touch switch configured as switch1.

Reading through the wiki it seems I cannot configure it as a button, so I can obtain the double tap feature.

What is the reason for this, or is it possible the config a switch to have the double tap feature?

ascillato, 2019-01-21:
Hi, 

Why you can't configure your touch switch as a button?

Lawrencezarb, 2019-01-21:
According to: https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches

I can configure it as a button but must setoption13 to 1 = Enable button single press only for immediate response, when I need setoption13 to 0 = Enable button single, multi-press and hold

ascillato, 2019-01-21:
Have you tried setoption13 to 0 (the default)?

You can try also setoption1

Seems that should work without any issue.

ascillato2, 2019-01-22:
@Lawrencezarb 

Hi, any news on this?
I saw you were posting this issue all over internet.
Could you solve your issue?

ascillato2, 2019-01-25:
Closing issue as there is no feedback.

closed 2019-01-25 by ascillato2

Lawrencezarb, 2019-02-07:
Sorry for the late reply. The solution was in fact to set the touch switch as a button. But the wiki indicated that I could not do this...