arendst/Tasmota #4997: Weird chars in console and unresponsive device

ezraholm50, 2019-01-21:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : Sonoff SC
- [ ] If a pre-compiled release or development binary was used, which one? : 6.4.1 / core 2.5 and generic 6.4.1
- [ ] You have tried latest release or development binaries? : Yes
- [ ] Provide the output of command ``status 0`` : Can't this is my console, i noticed this also on another Sonoff Bridge that stopped working (6.4.1 / 2.5)
```
⸮Ɗ⸮̽X⸮Hؽ⸮)⸮⸮Ύ⸮JJJz⸮⸮H⸮g⸮⸮k⸮⸮⸮⸮9)g⸮⸮)⸮⸮JJJJJJ)⸮M⸮⸮g⸮⸮k⸮⸮⸮⸮9?⸮JJJ)⸮M(⸮g⸮C⸮^⸮⸮9⸮⸮⸮
⸮̄X⸮Hؽ⸮)⸮⸮Ύ⸮JJJ[8⸮H⸮g⸮⸮k⸮⸮⸮⸮9)g⸮⸮⸮⸮⸮JJJk⸮H⸮g⸮⸮k⸮⸮⸮⸮9)g⸮⸮JJJ{⸮⸮H⸮g⸮C⸮^$⸮9⸮⸮⸮
⸮̽X⸮Hؽ⸮Ɋ⸮Ύ⸮JJJ⸮M⸮⸮g⸮⸮k⸮⸮⸮⸮⸮?⸮JJ⸮
⸮M⸮⸮g⸮C⸮^⸮⸮9⸮⸮v
⸮̽X⸮Hؽ⸮)⸮⸮Ύ⸮JJ⸮
⸮M⸮⸮g⸮⸮k⸮⸮⸮⸮?⸮JJ⸮
⸮M⸮⸮g⸮C⸮^⸮⸮9⸮⸮⸮
⸮̽X⸮Hؽ⸮)⸮⸮Ύ⸮JJ⸮]⸮M⸮⸮g⸮⸮k⸮⸮⸮⸮⸮?⸮JJ⸮⸮⸮M⸮⸮g⸮#⸮^⸮⸮9⸮⸮⸮
⸮̽X⸮Hؽ⸮)⸮⸮Ύ⸮JJ⸮⸮⸮M⸮⸮g⸮⸮k⸮⸮⸮⸮9)g⸮R)⸮⸮JJ⸮)⸮M⸮⸮g⸮#⸮^⸮⸮9⸮⸮⸮
⸮̽X⸮Hؽ⸮)⸮⸮Ύ⸮JJe)⸮M⸮⸮g⸮⸮k⸮⸮⸮⸮⸮?⸮JJ⸮1M⸮⸮g⸮C⸮^⸮⸮9⸮⸮⸮
⸮̽X⸮Hؽ⸮)⸮⸮Ύ⸮JJ⸮⸮M⸮⸮g⸮⸮k⸮Ϙ⸮9)g⸮⸮)⸮⸮JJ⸮⸮M(⸮g⸮C⸮^⸮⸮9⸮⸮⸮
⸮̄X⸮Hؽ⸮)⸮⸮Ύ⸮JJ⸮⸮M⸮⸮g⸮⸮k⸮⸮⸮⸮9)g⸮⸮)⸮⸮JJ⸮*⸮M⸮⸮g⸮⸮k⸮⸮⸮⸮9)g⸮R)⸮⸮JJ⸮
⸮M⸮⸮g⸮!⸮^⸮⸮9⸮⸮⸮
⸮̽X⸮H⸮⸮⸮)⸮⸮Ύ⸮⸮JJJ⸮H⸮⸮)⸮
```

**(Please, remember to close the issue when the problem has been addressed)**

So I tried reflasing my Snoff SC with 6.4.1 but it didnt bring up the AP, hooked it up to the Arduino console, is shows the above chars, very wierd. Please advise. (Reflashing any version/erasing flash doesnt work)

andrethomas, 2019-01-21:
And the console is configured for 115200bps?

ezraholm50, 2019-01-21:
Recently switched to linux based esptool, previously I used my old windows machine. No settings user side needed. Now appearently with the linux tool it does (just followed the wiki instructions for linux)...

Tried it with the windows esptool and it work, that just cost me 2 hours searching for a solution. Must have been the settings.

closed 2019-01-21 by ezraholm50