arendst/Tasmota #5002: nodemcu active a check on gpio when reboot

misterita, 2019-01-22:
I use a nodemcu to control the relays and when it restart for any reason , for a few seconds it turns the relays on and off, I'd like to know  is it a nodemcu hardware check or something wrong in my configuration?
Tks

Jason2866, 2019-01-22:
https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#the-esp8266---hardware-more-info
The ESP cycles most GPIO pins at startup. It is not under control from Tasmota.
You can try GPIO 4 or 5. On these GPIO it should not happen.

andrethomas2, 2019-01-22:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-22 by andrethomas2

misterita, 2019-01-22:
tks