arendst/Tasmota #5003: Shelly 1 with switch and Tasmota 6.4.1

jimmyhall82, 2019-01-22:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

jimmyhall82, 2019-01-22:
Hi all, I have a Shelly 1 wired into a switch box and the original light switch used through the switch input per the Shelly wiring diagram. I have flashed to 6.4.1 and have the shelly 1 template used. The device responds through the toggle on the IP address accessed page and also through mqtt in home assistant, but every once in a while, I can toggle the switch either on the wall or in home assistant or the IP accessed page, and nothing happens. Over mqtt from home assistant, the toggle on the ui momentarily moves to on, but then moves back to off, without the light triggering or the Shelly relay making any noise. If i continually try, eventually i will get the relay to switch and the light turn on. Any suggestions as to what is going on, or do i have a bad unit.

arendst, 2019-01-22:
First provide requested info like output of command ``status 0``.

Second provide info about your Shelly 1 hardware version as document on the sticker on the unit.

Third you might want to try the following configuration:

[image]

ascillato, 2019-01-22:
Hi,

Sorry but the command you post is incomplete due the correct command has a space between status and the zero:

Status 0

jimmyhall82, 2019-01-22:
@ascillato see below:

19:14:25 CMD: status 0
19:14:25 MQT: stat/chandelier/STATUS = {"Status":{"Module":18,"FriendlyName":["chandelier"],"Topic":"chandelier","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:14:25 MQT: stat/chandelier/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:44:53","StartupUTC":"2019-01-22T16:29:32","Sleep":50,"BootCount":33,"SaveCount":176,"SaveAddress":"FB000"}}
19:14:25 MQT: stat/chandelier/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
19:14:25 MQT: stat/chandelier/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home Network",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
19:14:25 MQT: stat/chandelier/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":16,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
19:14:25 MQT: stat/chandelier/STATUS5 = {"StatusNET":{"Hostname":"chandelier-1677","IPAddress":"10.0.1.126","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"deleted","Webserver":2,"WifiConfig":5}}
19:14:25 MQT: stat/chandelier/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.145","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25A68D","MqttUser":"deleted","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:14:25 MQT: stat/chandelier/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 22 18:14:25 2019","Local":"Tue Jan 22 19:14:25 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:32","Sunset":"17:31"}}
19:14:25 MQT: stat/chandelier/STATUS10 = {"StatusSNS":{"Time":"2019-01-22T19:14:25","Switch1":"ON"}}
19:14:25 MQT: stat/chandelier/STATUS11 = {"StatusSTS":{"Time":"2019-01-22T19:14:25","Uptime":"0T01:44:53","Vcc":3.456,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"deleted","BSSId":"deleted","Channel":1,"RSSI":92}}}

jimmyhall82, 2019-01-23:
It appears that if the physical switch is left alone, the mqtt side of things works, as soon as a switch is used, everything starts going out of sync. If the light is on, and i physically switch it off, the mqtt does not report it as off. If I try physically turning it back on, it doesn't respond, till eventually, it reboots I think. Then mqtt works.

jimmyhall82, 2019-01-23:
I have two shelly 1, one noted as bedroom and one chandelier, see the two status 0 results below.... appologies for the extra data if it is not helpful.

Bedroom

05:22:30 CMD: status 0
05:22:30 MQT: stat/bedroom_light/STATUS = {"Status":{"Module":46,"FriendlyName":["Bedroom Light"],"Topic":"bedroom_light","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:22:30 MQT: stat/bedroom_light/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"11T01:23:19","StartupUTC":"2019-01-12T02:59:11","Sleep":50,"BootCount":13,"SaveCount":208,"SaveAddress":"F6000"}}
05:22:30 MQT: stat/bedroom_light/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
05:22:30 MQT: stat/bedroom_light/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home Network",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
05:22:30 MQT: stat/bedroom_light/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
05:22:30 MQT: stat/bedroom_light/STATUS5 = {"StatusNET":{"Hostname":"Bedroom_Light","IPAddress":"10.0.1.125","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"CC:50:E3:25:A6:93","Webserver":2,"WifiConfig":5}}
05:22:30 MQT: stat/bedroom_light/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.145","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25A693","MqttUser":"jimmyhall82","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
05:22:30 MQT: stat/bedroom_light/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 23 04:22:30 2019","Local":"Wed Jan 23 05:22:30 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:31","Sunset":"17:32"}}
05:22:30 MQT: stat/bedroom_light/STATUS10 = {"StatusSNS":{"Time":"2019-01-23T05:22:30","Switch1":"OFF"}}
05:22:30 MQT: stat/bedroom_light/STATUS11 = {"StatusSTS":{"Time":"2019-01-23T05:22:30","Uptime":"11T01:23:19","Vcc":3.460,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Home Network","BSSId":"B2:84:C6:08:EB:78","Channel":1,"RSSI":100}}}
05:22:35 MQT: stat/bedroom_light/STATUS = {"Status":{"Module":46,"FriendlyName":["Bedroom Light"],"Topic":"bedroom_light","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:22:35 MQT: stat/bedroom_light/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"11T01:23:24","StartupUTC":"2019-01-12T02:59:11","Sleep":50,"BootCount":13,"SaveCount":208,"SaveAddress":"F6000"}}
05:22:35 MQT: stat/bedroom_light/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
05:22:35 MQT: stat/bedroom_light/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home Network",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
05:22:35 MQT: stat/bedroom_light/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
05:22:35 MQT: stat/bedroom_light/STATUS5 = {"StatusNET":{"Hostname":"Bedroom_Light","IPAddress":"10.0.1.125","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"CC:50:E3:25:A6:93","Webserver":2,"WifiConfig":5}}
05:22:35 MQT: stat/bedroom_light/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.145","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25A693","MqttUser":"jimmyhall82","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
05:22:35 MQT: stat/bedroom_light/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 23 04:22:35 2019","Local":"Wed Jan 23 05:22:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:31","Sunset":"17:32"}}
05:22:35 MQT: stat/bedroom_light/STATUS10 = {"StatusSNS":{"Time":"2019-01-23T05:22:35","Switch1":"OFF"}}
05:22:35 MQT: stat/bedroom_light/STATUS11 = {"StatusSTS":{"Time":"2019-01-23T05:22:35","Uptime":"11T01:23:24","Vcc":3.458,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Home Network","BSSId":"B2:84:C6:08:EB:78","Channel":1,"RSSI":100}}}
05:22:43 MQT: stat/bedroom_light/STATUS = {"Status":{"Module":46,"FriendlyName":["Bedroom Light"],"Topic":"bedroom_light","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:22:43 MQT: stat/bedroom_light/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"11T01:23:32","StartupUTC":"2019-01-12T02:59:11","Sleep":50,"Bo [... cut ...]
05:24:03 MQT: stat/chandelier/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 23 04:24:03 2019","Local":"Wed Jan 23 05:24:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:31","Sunset":"17:32"}}
05:24:03 MQT: stat/chandelier/STATUS10 = {"StatusSNS":{"Time":"2019-01-23T05:24:03","Switch1":"ON"}}
05:24:03 MQT: stat/chandelier/STATUS11 = {"StatusSTS":{"Time":"2019-01-23T05:24:03","Uptime":"0T00:05:14","Vcc":3.475,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Home Network","BSSId":"AC:84:C6:08:E4:B4","Channel":6,"RSSI":84}}}

jimmyhall82, 2019-01-24:
Resolved with a firmware delete and reflash.

closed 2019-01-24 by jimmyhall82