arendst/Tasmota #5004: Automatic discovery for Homeassistant doesnt create sensors.

timota, 2019-01-22:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
When Enabling Home Assistant automatic discovery via option SetOption19 1 - sensor devices not 
announced to Home Assistant, no retained topics in mqtt either, lights and switches are created and managebleю

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [ x] _Hardware used_ : sonoff basic r1
- [x ] _Development/Compiler/Upload tools used_ : OTA
- [ x] _If a pre-compiled release or development binary was used, which one?_ : Sonoff-Tasmota 6.4.1.10, core 2.5.0-beta2
- [x ] _You have tried latest release or development binaries?_ : yes
- [x ] _Provide the output of command_``status 0`` :
```
15:47:03 MQT: f3-r1-light-table/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["f3-r1-light-table","f3-r1-light-table-ws"],"Topic":"f3-r1-light-table","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:47:03 MQT: f3-r1-light-table/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:16:59","StartupUTC":"2019-01-22T14:30:04","Sleep":50,"BootCount":9,"SaveCount":18,"SaveAddress":"FA000"}}
15:47:03 MQT: f3-r1-light-table/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.10(dcabb9c-sonoff)","BuildDateTime":"2019-01-22T00:12:21","Boot":31,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
15:47:03 MQT: f3-r1-light-table/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"192.168.1.201","LogPort":514,"SSId":["VM1126584",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
15:47:03 MQT: f3-r1-light-table/stat/STATUS4 = {"StatusMEM":{"ProgramSize":548,"Free":452,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
15:47:03 MQT: f3-r1-light-table/stat/STATUS5 = {"StatusNET":{"Hostname":"f3-r1-light-table-6518","IPAddress":"192.168.1.176","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:0A:D9:76","Webserver":2,"WifiConfig":4}}
15:47:03 MQT: f3-r1-light-table/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0AD97676","MqttUser":"master","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:47:03 MQT: f3-r1-light-table/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 22 14:47:03 2019","Local":"Tue Jan 22 15:47:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:32","Sunset":"17:31"}}
15:47:03 MQT: f3-r1-light-table/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-22T15:47:03","BME280":{"Temperature":20.3,"Humidity":44.5,"Pressure":991.7},"BH1750":{"Illuminance":0},"PressureUnit":"hPa","TempUnit":"C"}}
15:47:03 MQT: f3-r1-light-table/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-22T15:47:03","Uptime":"0T00:16:59","Vcc":3.435,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Dimmer":10,"Color":"191919","HSBColor":"0,0,10","Channel":[9,9,9],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"VM1126584","BSSId":"B0:39:56:D6:C2:2C","Channel":6,"RSSI":82}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
Connect sensors as showen below
```
{"GPIO1":"6 (I2C SDA)","GPIO2":"0 (None)","GPIO3":"5 (I2C SCL)","GPIO4":"0 (None)","GPIO14":"7 (WS2812)"}
```
run SetOption19 1

**Expected behavior**
_A clear and concise description of what you expected to happen._
Sensors have to be announced to HASS and Mqtt retain topics created

**Additional context**
_Add any other context about the problem here._
More logs and some info [https://github.com/arendst/Sonoff-Tasmota/issues/4988](url)

also, when running SetOption19 1, following topics have been created
```
Jan 22 14:30:00 f3-r1-light-table-6518 ESP-MQT: homeassistant/light/0AD976_LI_1/config =  (retained)
Jan 22 14:30:00 f3-r1-light-table-6518 ESP-MQT: homeassistant/switch/0AD976_RL_1/config = {"name":"f3-r1-light-table","cmd_t":"~cmnd/POWER1","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER1}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"0AD976_RL_1","device":{"identifiers":["0AD976"],"name":"f3-r1-light-table","model":"Sonoff Basic","sw_version":"6.4.1.10(dcabb9c-sonoff)","manufacturer":"Tasmota"}, "~":"f3-r1-light-table/"} (retained)
Jan 22 14:30:00 f3-r1-light-table-6518 ESP-MQT: homeassistant/switch/0AD976_RL_2/config =  (retained)
Jan 22 14:30:00 f3-r1-light-table-6518 ESP-MQT: homeassistant/light/0AD976_LI_2/config = {"name":"f3-r1-light-table-ws","cmd_t":"~cmnd/POWER2","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER2}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","bri_cmd_t":"~cmnd/Dimmer","bri_stat_t":"~tele/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","rgb_cmd_t":"~cmnd/Color2","rgb_stat_t":"~tele/STATE","rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join
Jan 22 14:30:00 f3-r1-light-table-6518 ESP-MQT: homeassistant/light/0AD976_LI_3/config =  (retained)
Jan 22 14:30:00 f3-r1-light-table-6518 ESP-MQT: homeassistant/switch/0AD976_RL_3/config =  (retained)
Jan 22 14:30:00 f3-r1-light-table-6518 ESP-MQT: homeassistant/light/0AD976_LI_4/config =  (retained)
Jan 22 14:30:00 f3-r1-light-table-6518 ESP-MQT: homeassistant/light/0AD976_LI_5/config =  (retained)
```
nothing related to sensors.

arendst, 2019-01-22:
The output you receive after executing ``setoption19 1`` is that all or did you forget to add it all.

A correct discovery message should at least contain 8 light/switch pairs like mine:
```
homeassistant/switch/C523C3_RL_1/config (null)
homeassistant/light/C523C3_LI_1/config {"name":"Ledunia2","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","bri_cmd_t":"~cmnd/Dimmer","bri_stat_t":"~tele/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","rgb_cmd_t":"~cmnd/Color2","rgb_stat_t":"~tele/STATE","rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join(',')}}","fx_cmd_t":"~cmnd/Scheme","fx_stat_t":"~tele/STATE","fx_val_tpl":"{{value_json.Scheme}}","fx_list":["0","1","2","3","4"],"uniq_id":"C523C3_LI_1","device":{"identifiers":["C523C3"],"name":"Ledunia2","model":"Generic","sw_version":"6.4.1.10(sonoff)","manufacturer":"Tasmota"}, "~":"ledunia2/"}
homeassistant/light/C523C3_LI_2/config (null)
homeassistant/switch/C523C3_RL_2/config (null)
homeassistant/light/C523C3_LI_3/config (null)
homeassistant/switch/C523C3_RL_3/config (null)
homeassistant/light/C523C3_LI_4/config (null)
homeassistant/switch/C523C3_RL_4/config (null)
homeassistant/light/C523C3_LI_5/config (null)
homeassistant/switch/C523C3_RL_5/config (null)
homeassistant/light/C523C3_LI_6/config (null)
homeassistant/switch/C523C3_RL_6/config (null)
homeassistant/light/C523C3_LI_7/config (null)
homeassistant/switch/C523C3_RL_7/config (null)
homeassistant/light/C523C3_LI_8/config (null)
homeassistant/switch/C523C3_RL_8/config (null)
homeassistant/sensor/C523C3_MHZ19B_Model/config {"name":"Ledunia2 MHZ19B Model","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","unit_of_meas":" ","val_tpl":"{{value_json['MHZ19B'].Model}}","uniq_id":"C523C3_MHZ19B_Model","device":{"identifiers":["C523C3"],"name":"Ledunia2","model":"Generic","sw_version":"6.4.1.10(sonoff)","manufacturer":"Tasmota"}, "~":"ledunia2/tele/"}
homeassistant/sensor/C523C3_MHZ19B_CarbonDioxide/config {"name":"Ledunia2 MHZ19B CarbonDioxide","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","unit_of_meas":" ","val_tpl":"{{value_json['MHZ19B'].CarbonDioxide}}","uniq_id":"C523C3_MHZ19B_CarbonDioxide","device":{"identifiers":["C523C3"],"name":"Ledunia2","model":"Generic","sw_version":"6.4.1.10(sonoff)","manufacturer":"Tasmota"}, "~":"ledunia2/tele/"}
homeassistant/sensor/C523C3_MHZ19B_Temperature/config {"name":"Ledunia2 MHZ19B Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","unit_of_meas":"°C","val_tpl":"{{value_json['MHZ19B'].Temperature}}","uniq_id":"C523C3_MHZ19B_Temperature","device":{"identifiers":["C523C3"],"name":"Ledunia2","model":"Generic","sw_version":"6.4.1.10(sonoff)","manufacturer":"Tasmota"}, "~":"ledunia2/tele/"}
```
Can you have a look at your MQTT broker's log and see if it receives all messages.

arendst, 2019-01-22:
From your status 0 info it still reports a version (dcabb9c) without the latest Hass Discovery fix:

[image]

Pls update to at least development version 6.4.0.10 with commit 7745e1e or later:

[image]

timota, 2019-01-22:
After upgrade to commit bda3126 with Build time - 20190122 15:12 GMT+2 (40min ago) it starts working.

And also - base topic switched automatically. No need to change it manually. 

Looks like issue has been fixed by you in latest commit. 
Many Thanks. 

Issue can be closed or do you need more info from me ?

arendst, 2019-01-22:
You raised the issue. You may/should also close the issue.

timota, 2019-01-22:
One more time - many thanks.

closed 2019-01-22 by timota