arendst/Tasmota #5005: Sonoff SC Sensors not available anymore Webgui/MQTT/Status 8

ezraholm50, 2019-01-22:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used : ESPTOOL windows
- [ ] Hardware used : Sonoff SC
- [ ] If a pre-compiled release or development binary was used, which one? : 6.4.1 / core 2.3.0 - 6.2.1 / 2.3.0 - 6.4.1 / 2.4.2 all with just sonoff.bin and sonoff-sensors.bin
- [ ] You have tried latest release or development binaries? : latest and different core (wifi issues on other devices)
- [ ] Provide the output of command ``status 0`` :
```
19:18:16 CMD: status 0
19:18:16 SRC: WebConsole from 192.168.3.3
19:18:16 RSL: Received Topic /status, Data Size 1, Data 0
19:18:16 RSL: Group 0, Index 1, Command STATUS, Data 0
19:18:16 MQT: sonoffsc/stat/STATUS = {"Status":{"Module":21,"FriendlyName":["Sonoff SC"],"Topic":"sonoffsc","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:18:16 MQT: sonoffsc/stat/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:15","StartupUTC":"2019-01-22T18:14:01","Sleep":50,"BootCount":20,"SaveCount":43,"SaveAddress":"F6000"}}
19:18:17 MQT: sonoffsc/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1(release-sensors)","BuildDateTime":"2018-12-24T15:53:42","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:18:17 MQT: sonoffsc/stat/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["HASS",""],"TelePeriod":300,"SetOption":["000A8009","558180C0","00000200"]}}
19:18:17 MQT: sonoffsc/stat/STATUS4 = {"StatusMEM":{"ProgramSize":534,"Free":468,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDEE794","0003A3A4","B7FFBFCC","0002BBC0"]}}
19:18:17 MQT: sonoffsc/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoffsc-7808","IPAddress":"192.168.6.106","Gateway":"192.168.6.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.5","Mac":"BC:DD:C2:FD:1E:80","Webserver":2,"WifiConfig":2}}
19:18:17 MQT: sonoffsc/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.3.9","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FD1E80","MqttUser":"htpc","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:18:17 MQT: sonoffsc/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 22 18:18:17 2019","Local":"Tue Jan 22 19:18:17 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:32","Sunset":"17:31"}}
19:18:17 MQT: sonoffsc/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-22T19:18:17"}}
19:18:17 MQT: sonoffsc/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-22T19:18:17","Uptime":"0T00:04:16","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"HASS","BSSId":"02:4E:26:12:37:B4","Channel":1,"RSSI":54}}}
19:18:28 WIF: Checking connection...
```

Status 8 should give me the info, but only reports time:
```
19:19:48 CMD: status 8
19:19:48 SRC: WebConsole from 192.168.3.3
19:19:48 RSL: Received Topic /status, Data Size 1, Data 8
19:19:48 RSL: Group 0, Index 1, Command STATUS, Data 8
19:19:48 MQT: sonoffsc/stat/STATUS8 = {"StatusSNS":{"Time":"2019-01-22T19:19:48"}}
```

This all did not happen after an upgrade or anything, just was gone, rebooting, reflashing etc etc doesnt work. Some more info:

```
19:21:26 CMD: I2Cscan
19:21:26 SRC: WebConsole from 192.168.3.3
19:21:26 RSL: Received Topic /I2Cscan, Data Size 0, Data 
19:21:26 RSL: Group 0, Index 1, Command I2CSCAN, Data 
19:21:26 MQT: sonoffsc/stat/RESULT = {"Command":"Unknown"}
```

```
19:21:45 CMD: Gpio
19:21:45 SRC: WebConsole from 192.168.3.3
19:21:45 RSL: Received Topic /Gpio, Data Size 0, Data 
19:21:45 RSL: Group 0, Index 1, Command GPIO, Data 
19:21:45 MQT: sonoffsc/stat/RESULT = {"GPIO2":"0 (None)"}
```

```
19:21:59 CMD: gpios
19:21:59 SRC: WebConsole from 192.168.3.3
19:21:59 RSL: Received Topic /gpios, Data Size 0, Data 
19:21:59 RSL: Group 0, Index 1, Command GPIOS, Data 
19:21:59 MQT: sonoffsc/stat/RESULT = {"GPIOs1":["0 (None)","18 (Button2)","91 (Button2n)","19 (Button3)","92 (Button3n)","20 (Button4)","93 (Button4n)","9 (Switch1)","82 (Switch1n)","10 (Switch2)","83 (Switch2n)","11 (Switch3)","84 (Switch3n)","12 (Switch4)","85 (Switch4n)","13 (Switch5)","86 (Switch5n)","14 (Switch6)","87 (Switch6n)","15 (Switch7)","88 (Switch7n)","16 (Switch8)","89 (Switch8n)","21 (Relay1)","29 (Relay1i)","22 (Relay2)","30 (Relay2i)","23 (Relay3)"]}
19:21:59 MQT: sonoffsc/stat/RESULT = {"GPIOs2":["31 (Relay3i)","24 (Relay4)","32 (Relay4i)","25 (Relay5)","33 (Relay5i)","26 (Relay6)","34 (Relay6i)","27 (Relay7)","35 (Relay7i)","28 (Relay8)","36 (Relay8i)","53 (Led2)","57 (Led2i)","54 (Led3)","58 (Led3i)","55 (Led4)","59 (Led4i)","37 (PWM1)","46 (PWM1i)","38 (PWM2)","47 (PWM2i)","39 (PWM3)","48 (PWM3i)","40 (PWM4)","49 (PWM4i)","41 (PWM5)","50 (PWM5i)","42 (Counter1)","94 (Counter1n)","43 (Counter2)","95 (Counter2n)"]}
19:21:59 MQT: sonoffsc/stat/RESULT = {"GPIOs3":["44 (Counter3)","96 (Counter3n)","45 (Counter4)","97 (Counter4n)","5 (I2C SCL)","6 (I2C SDA)","1 (DHT11)","2 (AM2301)","3 (SI7021)","4 (DS18x20)","7 (WS2812)","8 (IRsend)","51 (IRrecv)","105 (RFSend)","106 (RFrecv)","116 (RF Sensor)","73 (SR04 Tri)","74 (SR04 Ech)","79 (TM16 CLK)","80 (TM16 DIO)","81 (TM16 STB)","102 (HX711 SCK)","103 (HX711 DAT)","71 (SerBr Tx)","72 (SerBr Rx)","60 (MHZ Tx)","61 (MHZ Rx)","64 (SAir Tx)"]}
19:22:00 MQT: sonoffsc/stat/RESULT = {"GPIOs4":["65 (SAir Rx)","101 (SDS0X1 Tx)","70 (SDS0X1 Rx)","62 (PZEM0XX Tx)","63 (PZEM004 Rx)","98 (PZEM016 Rx)","99 (PZEM017 Rx)","75 (SDM120/220 Tx)","76 (SDM120/220 Rx)","77 (SDM630 Tx)","78 (SDM630 Rx)","69 (PMS5003)","104 (TX20)","100 (MP3 Player)","107 (Tuya Tx)","108 (Tuya Rx)"]}
```

ezraholm50, 2019-01-23:
@Jason2866 Also retrying it with esptool on linux i get this error:

```
root@wzm-PC:~# esptool --port /dev/ttyUSB0 erase_flash
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Enabling default SPI flash mode...
Erasing flash (this may take a while)...

A fatal error occurred: ESP8266 ROM does not support function erase_flash.
```

Jason2866, 2019-01-23:
So, you have hardware problem....

ezraholm50, 2019-01-23:
Right, what makes you believe that? Sensors not showing or unable to manually erase flash?

Jason2866, 2019-01-23:
If flash fails something is faulty in hardware. In the case that flash works in general for you
If flash fails in genearal you have a hardware problem too ;-)

ezraholm50, 2019-01-23:
The flash doesn't fail... Note that it happend without me doing anything "automagically" 
After i noticed the sensors had been gone, i reflashed, which just worked, selected the right module, and still nothing, can't be that 4 individual sensors are failing all at once...

Jason2866, 2019-01-24:
`A fatal error occurred: ESP8266 ROM does not support function erase_flash.`  ??!!
After erase flash command there has to be a message successfull and the time needed for erasing
Since this is missing something goes wrong during erase and probably flashing tasmota too

joba-1, 2019-01-24:
Hi,
the output just states the more or less obvious: you cannot write (or erase) Read Only Memory.
Maybe it helps if you add -vvv to the esptool command to find out more background.

ezraholm50, 2019-01-24:
@joba-1 Yes thanks, but again this is not an issue as the write commands perfectly erase the device. Flash works fine because i tried different versions and they show up properly in the info page at the webui

Jason2866, 2019-01-24:
If everything is okay, it is fine...

ezraholm50, 2019-01-24:
> If everything is okay, it is fine...

Yes it would be, but its not...

ascillato, 2019-01-24:
If you have already erased the flash and flashed again successfully, and you have selected the correct module in module config, and the issue just happens without any change, seems to be a hardware issue.

Please, check with a DMM if the sensors are receiving voltage.

ezraholm50, 2019-01-24:
> If you have already erased the flash and flashed again successfully, and you have selected the correct module in module config, and the issue just happens without any change, seems to be a hardware issue.
> 
> Please, check with a DMM if the sensors are receiving voltage.

Will do, will report back in a sec

mike2nl, 2019-01-25:
@ezraholm50 
Possible a stupid question:
- You have set back the jumpers in the right way, not crossed?

ezraholm50, 2019-01-25:
Hey Mike, thanks for the heads up, yeah there on there the right way!

ascillato, 2019-01-25:
@ezraholm50 

You have erased and reflashed it, you have checked voltages, you have put back the jumpers after flashing, so, as it fails just suddendly without anything changed from you, seems to be a type of hardware issue.

Let's check the console for more debugging. Please, use the command `weblog 4` in the console and restart your device. Please, then post the full output of the console. Thanks.

Tontze, 2019-01-25:
Hm, you are talking about jumpers, not jumper. In docs you remove only TX jumper for flashing purposes, not both ? 

I have similar problems tho, all sensors updates only once, when i add power to the module. Not even when i reboot device, only when i unplug/plug in power ...

ezraholm50, 2019-01-25:
@Tontze I've tried every option now, only TX off, both off, both on, nothing seems to make a difference.

@ascillato Thanks for your help, here is the output after a restart with weblog 4

```
00:00:00 CFG: Loaded from flash at F9, Count 64
00:00:00 APP: Set Baudrate to 19200
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff SC Version 6.4.1(release-sensors)-2_3_0
00:00:00 SER: Transmit AT+START
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 WaaromZoMoeilijk-HASS in mode 11N as sonoffsc-7808...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoffsc-7808.local with IP address 192.168.6.106
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: sonoffsc/tele/LWT = Online (retained)
00:00:05 MQT: sonoffsc/cmnd/POWER = 
00:00:05 MQT: Subscribe to sonoffsc/cmnd/#
00:00:05 MQT: Subscribe to sonoffs/cmnd/#
00:00:05 MQT: Subscribe to cmnd/DVES_FD1E80_fb/#
00:00:05 MQT: sonoffsc/tele/INFO1 = {"Module":"Sonoff SC","Version":"6.4.1(release-sensors)","FallbackTopic":"cmnd/DVES_FD1E80_fb/","GroupTopic":"sonoffs"}
00:00:06 MQT: sonoffsc/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffsc-7808","IPAddress":"192.168.6.106"}
00:00:06 MQT: sonoffsc/tele/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: homeassistant/light/FD1E80_LI_1/config =  (retained)
00:00:06 MQT: homeassistant/switch/FD1E80_RL_1/config =  (retained)
00:00:06 MQT: homeassistant/light/FD1E80_LI_2/config =  (retained)
00:00:06 MQT: homeassistant/switch/FD1E80_RL_2/config =  (retained)
00:00:06 MQT: homeassistant/light/FD1E80_LI_3/config =  (retained)
00:00:06 MQT: homeassistant/switch/FD1E80_RL_3/config =  (retained)
00:00:06 MQT: homeassistant/light/FD1E80_LI_4/config =  (retained)
00:00:06 MQT: homeassistant/switch/FD1E80_RL_4/config =  (retained)
00:00:06 MQT: homeassistant/light/FD1E80_LI_5/config =  (retained)
00:00:06 MQT: homeassistant/switch/FD1E80_RL_5/config =  (retained)
00:00:06 MQT: homeassistant/light/FD1E80_LI_6/config =  (retained)
00:00:06 MQT: homeassistant/switch/FD1E80_RL_6/config =  (retained)
00:00:06 MQT: homeassistant/light/FD1E80_LI_7/config =  (retained)
00:00:06 MQT: homeassistant/switch/FD1E80_RL_7/config =  (retained)
00:00:06 MQT: homeassistant/light/FD1E80_LI_8/config =  (retained)
00:00:06 MQT: homeassistant/switch/FD1E80_RL_8/config =  (retained)
00:00:10 APP: Boot Count 37
00:00:10 CFG: Saved to flash at F8, Count 65, Bytes 3584
00:00:14 MQT: sonoffsc/tele/STATE = {"Time":"1970-01-01T00:00:14","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"WaaromZoMoeilijk-HASS","BSSId":"02:4E:26:12:37:B4","Channel":1,"RSSI":58}}
00:00:17 APP: (UTC) Fri Jan 25 19:50:50 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
20:50:57 WIF: Checking connection...
20:50:57 WIF: Connected
20:51:17 WIF: Checking connection...
20:51:17 WIF: Connected
20:51:28 CMD: status 8
20:51:28 SRC: WebConsole from 192.168.3.3
20:51:28 RSL: Received Topic /status, Data Size 1, Data 8
20:51:28 RSL: Group 0, Index 1, Command STATUS, Data 8
20:51:28 MQT: sonoffsc/stat/STATUS8 = {"StatusSNS":{"Time":"2019-01-25T20:51:28"}}
```

Ive also entered status 8, its just the same as it was when i posted the issue. 
It doesnt really show anything more, right?

ascillato, 2019-01-25:
Thanks for share.

The ESP is sending the START to the other MCU, but the other is not responding.

Please, re check the jumpers, may be they are lose or there is something in the path of the the TX from the MCU to the ESP8266.

ezraholm50, 2019-01-25:
Hmm nothing seems out of the ordinary. Anything else i can try? Reflash the other chip?

ascillato, 2019-01-25:
So, please check with a DMM if the track of TX from the mcu to the esp8266 is ok. Also check that there is VCC and GND on tha mcu. If all that is ok, sorry but that mcu is dead. I don't know about flashing that other. Sorry.

ezraholm50, 2019-01-25:
Guess i'll have to leave it at that, there 4.9 volt flowing through both rx and TX and every VCC and GND are ok...

Any idea if i can salvage parts of other sonoffs? Thanks anyway so far!

ascillato, 2019-01-25:
On free gpios you can add other sensors. 

Sorry that this ends up without solution due to hardware issue.

If you feel there is nothing else to do, you can close the issue. Anyway, you can reopen it if you find another tip.

Tontze, 2019-01-25:
reopen this please, i have done some research with my sonoff sc. I came in the same exact situation as ezra, status8 doesnt output all it should, and weblog lvl4 shows : 

`21:14:36 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
21:14:36 SER: Transmit AT+SEND=fail`

and it should show : 

`14:15:31 SER: Received AT+UPDATE="humidity":87,"temperature":23,"light":9,"noise":2,"dusty":1`

after this i reflashed sonoff sc original firmware, and all sensor worked normally...

reflashed back to sonoff, everything worked normally, but then i changed mqtt settings, and again same situation, no sensor updates. Unplugged power few times, and working again ? Some instability in the air ?

joba-1, 2019-01-26:
what does your status4 say?
I see heap 12 in the other log, which always meant instability for me.
-- 
iphone so i typo

Tontze, 2019-01-26:
`10:02:51 CMD: status4
10:02:51 SRC: WebConsole from 192.168.8.100
10:02:51 RSL: Received Topic /status4, Data Size 0, Data 
10:02:51 RSL: Group 0, Index 4, Command STATUS, Data 
10:02:51 MQT: stat/sonoff/STATUS = {"Status":{"Module":21,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:02:57 SER: Received AT+STATUS`

ezraholm50, 2019-01-26:
@Tontze original as in itead firmware? Could you share the firmware ill tru it out... there is still hope, i guess..!

mike2nl, 2019-01-26:
@Tontze 
when you have the original firmware, please open an issue on:
- https://github.com/rt400/Sonoff-Firmware
and tell them that you have the binary for it. Tell them the version number of the board too.

ezraholm50, 2019-01-26:
@Tontze in the link mike posted? Thanks could you give a heads up? By power cycle you mean remove the power plug?

ezraholm50, 2019-01-26:
You can also use this, spun up a temp upload folder on my server.

https://nextcloud.waaromzomoeilijk.nl/s/x6fJWETSyxYKNiZ

Tontze, 2019-01-26:
yeah, remove power plug and plug it back.

Tontze, 2019-01-26:
> You can also use this, spun up a temp upload folder on my server.
> 
> https://nextcloud.waaromzomoeilijk.nl/s/
> Passwd: sonoffSC6666

need login&passwd

ezraholm50, 2019-01-26:
Lost the details of that email for now.. haha. Please send to sonoff@waaromzomoeilijk.nl
Thanks

ezraholm50, 2019-01-26:
Ok some i came up with this: https://www.itead.cc/wiki/File:FWTRX-TAMCUSC-SONOFFSC-ATMEGA328P.zip#filelinks

I'll give it a shot to try to reflash the ATMEGA328P chip

So far I've:
* Downloaded the zip and extracted
* Added the 2 zip lib's to Arduino IDE
* Opened sonoff_sc.ino
* Added ATMEGA board and selected ATMEGA328P
* Hooked up 5V TX RX GND to the ATMEGA serial pins (with flash mode enable / without, with the use of the button)

But i cant get it to flash yet... Any idea?

mike2nl, 2019-01-27:
@ezraholm50 
1st it is not the actual file it is for version 1.0
SO far you have v1.0 PCB board it is OK.

You have no bootloader on the ATMEGA. First you have to flash a bootloader for that.

arendst, 2019-01-27:
While looking at your status 0 info I see you have Seriallog set to 4.

As the comms between the esp and the Atmega uses this same serial interface it is most obvious you won't get any reasonable response from the Atmega.

Set seriallog to 0 and reboot. If no succes, turn off power, wait 10 seconds and turn power back on. A correct comms interface between esp and ATmega looks like this:
```
00:00:00 CFG: Loaded from flash at FB, Count 303
00:00:00 CFG: Saved to flash at FB, Count 304, Bytes 3584
00:00:00 SRC: Restart
00:00:00 Project sonoff Sc Version 6.4.1.11(sonoff)-2_5_0_BETA3
00:00:00 SER: Transmit AT+START
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sc...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:02 SER: Received AT+UPDATE="humidity":79,"temperature":26,"light":10,"noise":4,"dusty":1
00:00:02 SER: Transmit AT+SEND=ok
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sc with IP address 192.168.2.205
00:00:04 APP: (UTC) Sun Jan 27 14:36:54 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
15:36:56 MQT: Attempting connection...
15:36:56 MQT: Connected
15:36:56 MQT: tele/sc/LWT = Online (retained)
15:36:56 MQT: cmnd/sc/POWER = 
15:36:56 MQT: Subscribe to cmnd/sc/#
15:36:56 MQT: Subscribe to cmnd/sonoffs/#
15:36:56 MQT: Subscribe to cmnd/DVES_F8B22E2E_fb/#
15:36:56 MQT: tele/sc/INFO1 = {"Module":"Sonoff SC","Version":"6.4.1.11(sonoff)","FallbackTopic":"cmnd/DVES_F8B22E2E_fb/","GroupTopic":"sonoffs"}
15:36:56 MQT: tele/sc/INFO2 = {"WebServerMode":"Admin","Hostname":"sc","IPAddress":"192.168.2.205"}
15:36:56 MQT: tele/sc/INFO3 = {"RestartReason":"Software/System restart"}
15:37:00 APP: Boot Count 122
15:37:00 CFG: Saved to flash at FA, Count 305, Bytes 3584
15:37:02 SER: Received AT+STATUS?
15:37:02 SER: Transmit AT+STATUS=4
15:37:04 MQT: tele/sc/STATE = {"Time":"2019-01-27T15:37:04","Uptime":"0T00:00:14","Vcc":3.486,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"00:22:6B:FE:8E:20","Channel":6,"RSSI":100}}
15:37:04 MQT: tele/sc/SENSOR = {"Time":"2019-01-27T15:37:04","SonoffSC":{"Temperature":26,"Humidity":79,"Light":10,"Noise":40,"AirQuality":100},"TempUnit":"C"}
15:37:05 SER: Received AT+UPDATE="humidity":79,"temperature":26,"light":10,"noise":3,"dusty":1
15:37:05 SER: Transmit AT+SEND=ok
15:38:17 SER: Received AT+STATUS?
15:38:17 SER: Transmit AT+STATUS=4
15:38:43 SER: Received AT+UPDATE="humidity":79,"temperature":26,"light":10,"noise":4,"dusty":1
15:38:43 SER: Transmit AT+SEND=ok
```

ezraholm50, 2019-01-27:
@arendst thanks for your reply, I guess this tells me the comms between the 2 is broken somewhere then:
* Reset config
* Seriallog 0
* Weblog 4
* Syslog 4

```
00:00:00 CFG: Loaded from flash at F8, Count 12
00:00:00 APP: Boot Count 6
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_FD1E80, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 SER: Transmit AT+START
00:00:00 CFG: Saved to flash at F7, Count 13, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 WaaromZoMoeilijk-HASS in mode 11N as sonoff-7808...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff-7808.local with IP address 192.168.6.147
00:00:05 MQT: Attempting connection...
00:00:06 DNS: Query done. MQTT services found 0
00:00:06 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:00:13 APP: (UTC) Sun Jan 27 16:18:07 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
17:18:11 MQT: Attempting connection...
17:18:12 DNS: Query done. MQTT services found 0
17:18:12 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
17:18:21 WIF: Checking connection...
17:18:21 WIF: Connected
17:18:22 HTP: Main Menu
17:18:22 HTP: Console
17:18:23 MQT: Attempting connection...
17:18:24 DNS: Query done. MQTT services found 0
17:18:24 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
17:18:35 MQT: Attempting connection...
17:18:36 DNS: Query done. MQTT services found 0
17:18:36 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
```

Status 0 & 8
```

17:20:21 CMD: status 0
17:20:21 SRC: WebConsole from 192.168.3.3
17:20:21 RSL: Received Topic /status, Data Size 1, Data 0
17:20:21 RSL: Group 0, Index 1, Command STATUS, Data 0
17:20:21 RSL: stat/sonoff/STATUS = {"Status":{"Module":21,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:20:21 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:27","StartupUTC":"2019-01-27T16:17:54","Sleep":0,"BootCount":6,"SaveCount":13,"SaveAddress":"F7000"}}
17:20:21 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:20:21 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["WaaromZoMoeilijk-HASS",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
17:20:21 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
17:20:21 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7808","IPAddress":"192.168.6.147","Gateway":"192.168.6.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.5","Mac":"BC:DD:C2:FD:1E:80","Webserver":2,"WifiConfig":5}}
17:20:21 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FD1E80","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:20:21 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 27 16:20:21 2019","Local":"Sun Jan 27 17:20:21 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":1,"Sunrise":"08:27","Sunset":"17:38"}}
17:20:21 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-27T17:20:21"}}
17:20:21 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-27T17:20:21","Uptime":"0T00:02:27","Vcc":3.213,"Wifi":{"AP":1,"SSId":"WaaromZoMoeilijk-HASS","RSSI":64,"APMac":"02:4E:26:12:37:B4"}}}
17:20:29 CMD: status 8
17:20:29 SRC: WebConsole from 192.168.3.3
17:20:29 RSL: Received Topic /status, Data Size 1, Data 8
17:20:29 RSL: Group 0, Index 1, Command STATUS, Data 8
17:20:29 RSL: stat/sonoff/STATUS8 = {"StatusSNS":{"Time":"2019-01-27T17:20:29"}}
17:20:30 WIF: Checking connection...
17:20:30 WIF: Connected
17:20:33 MQT: Attempting connection...
17:20:34 DNS: Query done. MQTT services found 0
17:20:34 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
```

ezraholm50, 2019-01-27:
@arendst I see no support @ sensors for Sharp’s GP2Y1010AU0F dust sensor that is in the SC, is that because the ATMEGA handles it? Could i flash a d1 mini with tasmota and get it up and running that way perhaps? Just the dust sensor and Temp/hum that im interested in of the SC...

ascillato2, 2019-01-27:
Sorry that your device ends up to be broken. Sometimes that happens. Anyway the alternative that Jason have explained is really good, better than buying another sonoff sc.

Thanks for all the testings. If you need any more help, please just ask here o in the Tasmota Support Chat.

closed 2019-01-27 by ascillato2

Tontze, 2019-01-27:
@ezraholm50 If you still can try original firmware, i have now sent it to github page mentioned before. OP will propably put it there when he/she receives it.

ezraholm50, 2019-01-27:
@Tontze Great, will try it out! thanks!

Tontze, 2019-01-27:
Andi can confirm that setting log levels to zero, means that sonoff sc will boot as it should :) No need to replug power.

Tontze, 2019-02-24:
@ezraholm50 Out of curiosity, did you try original firmware ?

Joshfindit, 2019-06-17:
Odd, possibly related, issue as it may come down to communication between the ATMega328P and recent versions of Tasmota :

On `6.5.0`, the `SENSOR` values do not change. They seem to update once at boot, then not again until the next full power-cycle. Have tested it with an uptime of multiple days.

Console log:
```
23:44:46 CMD: Syslog 4
23:44:46 SRC: WebConsole from 192.168.7.100
23:44:46 RSL: Received Topic /Syslog, Data Size 1, Data 4
23:44:46 RSL: Group 0, Index 1, Command SYSLOG, Data 4
23:44:46 MQT: DVES_68A1A2_SC/stat/RESULT = {"SysLog":"4 (Active 4)"}
23:44:47 MQT: DVES_68A1A2_SC/tele/STATE = {"Time":"2019-06-17T23:44:47","Uptime":"0T00:04:26","Vcc":3.112,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"The Fro","BSSId":"AC:22:0B:CE:C3:50","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:44:47 MQT: DVES_68A1A2_SC/tele/SENSOR = {"Time":"2019-06-17T23:44:47","SonoffSC":{"Temperature":17.0,"Humidity":41.0,"Light":10,"Noise":60,"AirQuality":100},"TempUnit":"C"}
23:44:47 CFG: Saved to flash at F8, Count 36, Bytes 3584
23:44:47 WIF: Checking connection...
23:44:47 WIF: Connected
23:44:47 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
23:44:47 SER: Transmit AT+SEND=fail
23:44:55 MQT: DVES_68A1A2_SC/tele/HASS_STATE = {"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff SC","RestartReason":"Power on","Uptime":"0T00:04:34","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:04","MqttCount":1,"BootCount":19,"SaveCount":36,"IPAddress":"192.168.7.84","RSSI":"100","LoadAvg":19}
23:44:57 MQT: DVES_68A1A2_SC/tele/STATE = {"Time":"2019-06-17T23:44:57","Uptime":"0T00:04:36","Vcc":3.113,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"The Fro","BSSId":"AC:22:0B:CE:C3:50","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:44:57 MQT: DVES_68A1A2_SC/tele/SENSOR = {"Time":"2019-06-17T23:44:57","SonoffSC":{"Temperature":17.0,"Humidity":41.0,"Light":10,"Noise":60,"AirQuality":100},"TempUnit":"C"}
23:44:57 SER: Received AT+STATUS?
23:44:57 SER: Transmit AT+STATUS=4
23:45:00 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
23:45:00 SER: Transmit AT+SEND=fail
23:45:05 MQT: DVES_68A1A2_SC/tele/HASS_STATE = {"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff SC","RestartReason":"Power on","Uptime":"0T00:04:44","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:04","MqttCount":1,"BootCount":19,"SaveCount":36,"IPAddress":"192.168.7.84","RSSI":"100","LoadAvg":19}
23:45:07 MQT: DVES_68A1A2_SC/tele/STATE = {"Time":"2019-06-17T23:45:07","Uptime":"0T00:04:46","Vcc":3.114,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"The Fro","BSSId":"AC:22:0B:CE:C3:50","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:45:07 MQT: DVES_68A1A2_SC/tele/SENSOR = {"Time":"2019-06-17T23:45:07","SonoffSC":{"Temperature":17.0,"Humidity":41.0,"Light":10,"Noise":60,"AirQuality":100},"TempUnit":"C"}

```

arendst, 2019-06-19:
No problem at all:

```
00:00:00 CFG: Loaded from flash at FB, Count 308
00:00:00 CFG: Saved to flash at FB, Count 309, Bytes 3584
00:00:00 SRC: Restart
00:00:00 Project sonoff Sc Version 6.5.0.16(sonoff)-2_4_2
00:00:00 SER: Transmit AT+START
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sc...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:02 SER: Received AT+UPDATE="humidity":95,"temperature":23,"light":9,"noise":2,"dusty":1
00:00:02 SER: Transmit AT+SEND=ok
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sc with IP address 192.168.2.195
14:07:59 APP: (UTC) Wed Jun 19 12:07:59 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
14:08:00 MQT: Attempting connection...
14:08:00 MQT: Connected
14:08:00 MQT: tele/sc/LWT = Online (retained)
14:08:00 MQT: cmnd/sc/POWER = 
14:08:00 MQT: Subscribe to cmnd/sc/#
14:08:00 MQT: Subscribe to cmnd/sonoffs/#
14:08:00 MQT: Subscribe to cmnd/DVES_F8B22E_fb/#
14:08:00 MQT: tele/sc/INFO1 = {"Module":"Sonoff SC","Version":"6.5.0.16(sonoff)","FallbackTopic":"cmnd/DVES_F8B22E_fb/","GroupTopic":"sonoffs"}
14:08:00 MQT: tele/sc/INFO2 = {"WebServerMode":"Admin","Hostname":"sc","IPAddress":"192.168.2.195"}
14:08:00 MQT: tele/sc/INFO3 = {"RestartReason":"Software/System restart"}
14:08:01 HTP: Main Menu
14:08:02 APP: Boot Count 125
14:08:02 CFG: Saved to flash at FA, Count 310, Bytes 3584
14:08:04 SER: Received AT+STATUS?
14:08:04 SER: Transmit AT+STATUS=4
14:08:04 HTP: Console
14:08:08 MQT: tele/sc/STATE = {"Time":"2019-06-19T14:08:08","Uptime":"0T00:00:16","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
14:08:08 MQT: tele/sc/SENSOR = {"Time":"2019-06-19T14:08:08","SonoffSC":{"Temperature":23,"Humidity":95,"Light":20,"Noise":20,"AirQuality":100},"TempUnit":"C"}
14:08:14 SER: Received AT+STATUS?
14:08:14 SER: Transmit AT+STATUS=4
14:08:18 SER: Received AT+UPDATE="humidity":82,"temperature":23,"light":9,"noise":2,"dusty":1
14:08:18 SER: Transmit AT+SEND=ok
14:08:18 WIF: Checking connection...
14:08:18 WIF: Connected
14:08:22 SER: Received AT+UPDATE="humidity":76,"temperature":23,"light":9,"noise":2,"dusty":1
14:08:22 SER: Transmit AT+SEND=ok
14:08:28 SER: Received AT+UPDATE="humidity":85,"temperature":23,"light":9,"noise":2,"dusty":1
14:08:28 SER: Transmit AT+SEND=ok
14:08:32 SER: Received AT+UPDATE="humidity":90,"temperature":23,"light":9,"noise":2,"dusty":1
14:08:32 SER: Transmit AT+SEND=ok
14:08:36 SER: Received AT+UPDATE="humidity":87,"temperature":23,"light":9,"noise":2,"dusty":1
14:08:36 SER: Transmit AT+SEND=ok
14:08:38 WIF: Checking connection...
14:08:38 WIF: Connected
14:08:40 SER: Received AT+UPDATE="humidity":77,"temperature":23,"light":9,"noise":2,"dusty":1
14:08:40 SER: Transmit AT+SEND=ok
14:08:46 SER: Received AT+UPDATE="humidity":82,"temperature":23,"light":9,"noise":2,"dusty":1
14:08:46 SER: Transmit AT+SEND=ok
14:08:50 SER: Received AT+UPDATE="humidity":92,"temperature":23,"light":9,"noise":2,"dusty":1
14:08:50 SER: Transmit AT+SEND=ok
14:08:56 SER: Received AT+UPDATE="humidity":80,"temperature":23,"light":9,"noise":2,"dusty":1
14:08:56 SER: Transmit AT+SEND=ok
```

DasUrmel, 2019-11-17:
I know this is an old issue.... but I stumpled into this today as well.

After switching from Ubuntu 16.04 to Mint 19.2 I realized I couldn't erase my flash of an ESP8266 any more. It took my almost two hours to realise the provided esptool from Mint is Version 2.1 while the most current version on github is 2.8!

So... "apt remove esptool", "apt install pip; pip install setuptools wheel pyaes esptool" - voila!
I can use "erase_flash" again.

I hope this helps someone.

Thorsten