arendst/Tasmota #5006: MPR121 ha

phdelodder, 2019-01-22:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : VSC -> Platform.io
- [x] Hardware used : MPR121 Nodemcu
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? : 6.4.1.10
- [x] Provide the output of command ``status 0`` :
```
19:30:08 CMD: status 0
19:30:08 MQT: MCU/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["MCU Greenhouse"],"Topic":"MCU","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
19:30:08 MQT: MCU/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"External System","Uptime":"0T00:29:54","StartupUTC":"2019-01-22T18:00:14","Sleep":50,"BootCount":48,"SaveCount":114,"SaveAddress":"F8000"}}
19:30:08 MQT: MCU/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.10(sonoff)","BuildDateTime":"2019-01-22T18:55:52","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
19:30:08 MQT: MCU/stat/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DelodderVanTorre",""],"TelePeriod":300,"SetOption":["000A8229","558180C0","00000200"]}}
19:30:08 MQT: MCU/stat/STATUS4 = {"StatusMEM":{"ProgramSize":556,"Free":444,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE794","040383A0","23B697CE","00003BF3"]}}
19:30:08 MQT: MCU/stat/STATUS5 = {"StatusNET":{"Hostname":"MCU-Greenhouse","IPAddress":"192.168.0.67","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:52:91:FE","Webserver":2,"WifiConfig":5}}
19:30:08 MQT: MCU/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.38","MqttPort":1883,"MqttClientMask":"MCU","MqttClient":"MCU","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:30:08 MQT: MCU/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 22 18:30:08 2019","Local":"Tue Jan 22 19:30:08 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:32","Sunset":"17:31"}}
19:30:09 MQT: MCU/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-22T19:30:08","BME280":{"Temperature":14.9,"Humidity":49.6,"Pressure":999.7},"ADS1115":{"A0":3295,"A1":3331,"A2":3294,"A3":3293},"MCP230XX":{"D0":0,"D1":0,"D2":0,"D3":0,"D4":0,"D5":0,"D6":0,"D7":0,"D8":0,"D9":0,"D10":0,"D11":0,"D12":0,"D13":0,"D14":0,"D15":0},"MPR121A":{"Button0":0,"Button1":0,"Button2":0,"Button3":0,"Button4":0,"Button5":0,"Button6":0,"Button7":0,"Button8":0,"Button9":0,"Button10":0,"Button11":0,"Button12":0},"PressureUnit":"hPa","TempUnit ...
19:30:09 MQT: MCU/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-22T19:30:09","Uptime":"0T00:29:55","Vcc":2.538,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"DelodderVanTorre","BSSId":"7C:FF:4D:DC:CD:1F","Channel":11,"RSSI":100}}}
```

I have a MPR121 attached to a NODEMCU, I have executed 'setoptions19 1' and the MPR121 is visable in hass but as 12 different sensors, these are sensor and never change state when the button is pressed. My first question is why is the MPR121 reported as a sensor instead of a binary_sensor? 

```
homeassistant/sensor/5291FE_MPR121A_Button9/config {"name":"MCU Greenhouse MPR121A Button9","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","unit_of_meas":" ","val_tpl":"{{value_json['MPR121A'].Button9}}","uniq_id":"5291FE_MPR121A_Button9","device":{"identifiers":["5291FE"],"name":"MCU Greenhouse","model":"Generic","sw_version":"6.4.1.10(sonoff)","manufacturer":"Tasmota"}, "~":"MCU/tele/"}
MCU/stat/RESULT {"MPR121A":{"Button9":1}}
MCU/stat/RESULT {"MPR121A":{"Button9":0}}
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-22:
>My first question is why is the MPR121 reported as a sensor instead of a binary_sensor?

Because was never implemented.

renne, 2019-06-16:
MPR121 has no HomeAssistant function. I suggest to listen for MQTT RESULT messages.

effelle, 2019-09-28:
@phdelodder could you gently test this experimental build? I don't have a MPR121 and I need a feedback before proceeding. 
I've enabled ADS1115 and MCP230XX too, your Node should add all as expected.
If is working fine I'll add it on the list of the next possible HA integration updates.

[MPR121.zip](https://github.com/arendst/Sonoff-Tasmota/files/3665954/MPR121.zip)

phdelodder, 2019-09-29:
@effelle I have flashed it but don't see the values appear in the HA integration section.

effelle, 2019-09-29:
When you run setoption19 the binary sensores appears on console? If yes try to create an entity. Could be just an issue with the sensor list under Discovery. If not please paste here the result of setoption19.

phdelodder, 2019-09-29:
`16:07:29 CMD: setoption19 1
16:07:29 MQT: sonoff/stat/RESULT = {"SetOption19":"ON"}
16:07:31 MQT: homeassistant/binary_sensor/5291FE_MPR121A_Button5/config = {"name":"Sonoff MPR121A Button5","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"5291FE_MPR121A_Button5","device":{"identifiers":["5291FE"],"connections":[["mac","DC:4F:22:52:91:FE"]]},"~":"sonoff/tele/","unit_of_meas":" ","val_tpl":"{{value_json['MPR121A'].Button5}}"} (retained)
16:07:31 MQT: homeassistant/binary_sensor/5291FE_MPR121A_Button6/config = {"name":"Sonoff MPR121A Button6","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"5291FE_MPR121A_Button6","device":{"identifiers":["5291FE"],"connections":[["mac","DC:4F:22:52:91:FE"]]},"~":"sonoff/tele/","unit_of_meas":" ","val_tpl":"{{value_json['MPR121A'].Button6}}"} (retained)
16:07:31 MQT: homeassistant/binary_sensor/5291FE_MPR121A_Button7/config = {"name":"Sonoff MPR121A Button7","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"5291FE_MPR121A_Button7","device":{"identifiers":["5291FE"],"connections":[["mac","DC:4F:22:52:91:FE"]]},"~":"sonoff/tele/","unit_of_meas":" ","val_tpl":"{{value_json['MPR121A'].Button7}}"} (retained)
16:07:31 MQT: homeassistant/binary_sensor/5291FE_MPR121A_Button8/config = {"name":"Sonoff MPR121A Button8","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"5291FE_MPR121A_Button8","device":{"identifiers":["5291FE"],"connections":[["mac","DC:4F:22:52:91:FE"]]},"~":"sonoff/tele/","unit_of_meas":" ","val_tpl":"{{value_json['MPR121A'].Button8}}"} (retained)
16:07:31 MQT: homeassistant/binary_sensor/5291FE_MPR121A_Button9/config = {"name":"Sonoff MPR121A Button9","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"5291FE_MPR121A_Button9","device":{"identifiers":["5291FE"],"connections":[["mac","DC:4F:22:52:91:FE"]]},"~":"sonoff/tele/","unit_of_meas":" ","val_tpl":"{{value_json['MPR121A'].Button9}}"} (retained)
16:07:31 MQT: homeassistant/binary_sensor/5291FE_MPR121A_Button10/config = {"name":"Sonoff MPR121A Button10","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"5291FE_MPR121A_Button10","device":{"identifiers":["5291FE"],"connections":[["mac","DC:4F:22:52:91:FE"]]},"~":"sonoff/tele/","unit_of_meas":" ","val_tpl":"{{value_json['MPR121A'].Button10}}"} (retained)
16:07:31 MQT: homeassistant/binary_sensor/5291FE_MPR121A_Button11/config = {"name":"Sonoff MPR121A Button11","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"5291FE_MPR121A_Button11","device":{"identifiers":["5291FE"],"connections":[["mac","DC:4F:22:52:91:FE"]]},"~":"sonoff/tele/","unit_of_meas":" ","val_tpl":"{{value_json['MPR121A'].Button11}}"} (retained)
16:07:31 MQT: homeassistant/binary_sensor/5291FE_MPR121A_Button12/config = {"name":"Sonoff MPR121A Button12","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"5291FE_MPR121A_Button12","device":{"identifiers":["5291FE"],"connections":[["mac","DC:4F:22:52:91:FE"]]},"~":"sonoff/tele/","unit_of_meas":" ","val_tpl":"{{value_json['MPR121A'].Button12}}"} (retained)
16:07:31 MQT: homeassistant/sensor/5291FE_status/config = {"name":"Sonoff status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"5291FE_status","device":{"identifiers":["5291FE"],"connections":[["mac","DC:4F:22:52:91:FE"]],"name":"Sonoff","model":"Generic","sw_version":"6.6.0.14(sensors)","manufacturer":"Tasmota"},"~":"sonoff/tele/"} (retained)`

effelle, 2019-09-29:
Then tasmota marked them as binary sensors. Did you try to look on unused entities if the buttons are on binary-sensor section? I have a bunch of special sensor that are not added on HA discovery page but works just fine on Home Assistant (need a deeper investigation).

phdelodder, 2019-09-29:
Yes, I have done some investigation on it, can't seem to find them.

effelle, 2019-09-29:
Thanks for your feedback Philippe. 
Sadly is a little bit more complicated to have binary_sensor in the correct format, I ordered a MPR121 and I'll have a look within a couple of weeks.

effelle, 2019-10-11:
@phdelodder,
sadly the MPR121 I got was defective. Anyway I found the bug, but to complete the task and merge a PR I need to see the output of the console when you press any of the buttons on your MPR121. Could you gently provide it and paste here? Thanks!

ascillato2, 2019-10-11:
@effelle In the first comment I think is the information you need.

effelle, 2019-10-11:
Ok, it uses `/stat/RESULT` as with MCP230XX. 
Thanks Adrian.