arendst/Tasmota #5007: MCP230XX in ha

phdelodder, 2019-01-22:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : VSC -> Platform.io
- [x] Hardware used : MPR121 Nodemcu
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? : 6.4.1.10
- [x] Provide the output of command ``status 0`` :
```
19:30:08 CMD: status 0
19:30:08 MQT: MCU/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["MCU Greenhouse"],"Topic":"MCU","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
19:30:08 MQT: MCU/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"External System","Uptime":"0T00:29:54","StartupUTC":"2019-01-22T18:00:14","Sleep":50,"BootCount":48,"SaveCount":114,"SaveAddress":"F8000"}}
19:30:08 MQT: MCU/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.10(sonoff)","BuildDateTime":"2019-01-22T18:55:52","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
19:30:08 MQT: MCU/stat/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DelodderVanTorre",""],"TelePeriod":300,"SetOption":["000A8229","558180C0","00000200"]}}
19:30:08 MQT: MCU/stat/STATUS4 = {"StatusMEM":{"ProgramSize":556,"Free":444,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE794","040383A0","23B697CE","00003BF3"]}}
19:30:08 MQT: MCU/stat/STATUS5 = {"StatusNET":{"Hostname":"MCU-Greenhouse","IPAddress":"192.168.0.67","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:52:91:FE","Webserver":2,"WifiConfig":5}}
19:30:08 MQT: MCU/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.38","MqttPort":1883,"MqttClientMask":"MCU","MqttClient":"MCU","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:30:08 MQT: MCU/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 22 18:30:08 2019","Local":"Tue Jan 22 19:30:08 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:32","Sunset":"17:31"}}
19:30:09 MQT: MCU/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-22T19:30:08","BME280":{"Temperature":14.9,"Humidity":49.6,"Pressure":999.7},"ADS1115":{"A0":3295,"A1":3331,"A2":3294,"A3":3293},"MCP230XX":{"D0":0,"D1":0,"D2":0,"D3":0,"D4":0,"D5":0,"D6":0,"D7":0,"D8":0,"D9":0,"D10":0,"D11":0,"D12":0,"D13":0,"D14":0,"D15":0},"MPR121A":{"Button0":0,"Button1":0,"Button2":0,"Button3":0,"Button4":0,"Button5":0,"Button6":0,"Button7":0,"Button8":0,"Button9":0,"Button10":0,"Button11":0,"Button12":0},"PressureUnit":"hPa","TempUnit ...
19:30:09 MQT: MCU/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-22T19:30:09","Uptime":"0T00:29:55","Vcc":2.538,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"DelodderVanTorre","BSSId":"7C:FF:4D:DC:CD:1F","Channel":11,"RSSI":100}}}
```

I have a MCP23018 attached to a NODEMCU, I have executed 'backlog sensor29 reset5; setoptions19 1' and the MCP28018 is visable in hass but as 16 different sensors. I was expecting this to be a switch so I could toggle/enable/disable it. 

```
homeassistant/sensor/5291FE_MCP230XX_D11/config {"name":"MCU Greenhouse MCP230XX D11","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","unit_of_meas":" ","val_tpl":"{{value_json['MCP230XX'].D11}}","uniq_id":"5291FE_MCP230XX_D11","device":{"identifiers":["5291FE"],"name":"MCU Greenhouse","model":"Generic","sw_version":"6.4.1.10(sonoff)","manufacturer":"Tasmota"}, "~":"MCU/tele/"}
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-01-22:
MCP is not a switch. Please read the wiki article.
Imho there will no config that makes sense for HA with Autodiscovery. It should be excluded from.

phdelodder, 2019-01-22:
@Jason2866 the MCP is not a switch but it can be set as an output that can be toggle to 0 or 1. So I was assuming it would work.

andrethomas, 2019-01-22:
I tried installing HA at some stage but I was too stupid and couldn't get it to work so I went with node-red instead...

I suppose it could be adapted I'd have to learn how to use HA first... wait, isn't it that thing that uses yaml?

MadDoct, 2019-06-11:
@phdelodder , why don't you simply configure it manually in home assistant?!

phdelodder, 2019-06-11:
For now I have done everything manually, it's easier to have it automated.

kobuki, 2019-09-02:
I also had a similar request but then I realized it's easier to add controlling the MCP I/O endpoints to an MQTT switch. The states of those as inputs or outputs are autodetected and available in HA so both control and switch state can be updated. You might have done the same but I can post a sample if you want. Though I think it rather belongs to the HA forums.

effelle, 2019-09-10:
@kobuki if you want you can share your configuration anytime. 
Someone could benefit from it.

kobuki, 2019-09-10:
It's nothing fancy, just a basic MQTT switch config.

```
- platform: mqtt
  name: "MCP switch"
  command_topic: "remote-test/cmnd/sensor29"
  state_topic: "remote-test/stat/RESULT"
  value_template: "{{ value_json.S29cmnd_D5.STATE }}"
  payload_on: "5,off"
  payload_off: "5,on"
  state_on: "OFF"
  state_off: "ON"
  retain: false
```

effelle, 2019-10-11:
I had a deeper look on your issue but the amount of lines of codes needed just to change the MCP230XX behavior will be excessive. 
I don't think a patch will be merged, at least not made by me.
@ascillato if you agree please add `wont/can't fix` tag and close this issue, thanks.

closed 2019-10-11 by ascillato2