arendst/Tasmota #5010: Can't connect to Wifi with special characters in the password

andreas-bulling, 2019-01-23:
Hi,

I'm using the latest version of the firmware on a D1 mini but there seems to be a problem with special characters in the Wifi password.

What is the recommended way of using passwords with characters like
=,:,\,+
?

closed 2019-01-23 by andreas-bulling

ascillato2, 2019-01-25:
Sorry, not supported by the Arduino core.

ascillato2, 2019-01-25:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

spock64, 2019-02-15:
No, the Arduino Core does support the use of "*" - if the password is set via a serial connection, it works. The bug is in how the web UI handles the password - and this is a recent issue as Tasmota was always happy with "*" before.