arendst/Tasmota #5013: WS2812FX

YaLiC, 2019-01-23:
Very much need integration with the library WS2812FX to create festive garlands. Even if it will be built-in effects. Such a Union would have been excellent, convenience and functionality firmware Sonoff-Tasmota and capabilities of the library WS2812FX. I want it so badly, I'm willing to donate.

ascillato, 2019-01-23:
The led strip ws2812B is already implemented in Tasmota.

ascillato, 2019-01-23:
The request for the ws2812fx is already open at https://github.com/arendst/Sonoff-Tasmota/issues/3480

YaLiC, 2019-01-23:
> The led strip ws2812B is already implemented in Tasmota.

I'm grateful for that! But number of effects, sorry, very little for normal garland. It is more suitable for demonstrating support

ascillato2, 2019-01-23:
Ok, let's continue in the already open request at https://github.com/arendst/Sonoff-Tasmota/issues/3480

closed 2019-01-23 by ascillato2

a-roz, 2020-04-17:
Have finished this
https://github.com/a-roz/Tasmota/tree/ws2812fx

s-hadinger, 2020-04-18:
Brilliant. Would it make sense to submit it as a PR rather than a fork?
@a-roz

a-roz, 2020-04-18:
@s-hadinger This fork is too young, but in case no bugs after some time, why not ... The first I have to check it myself in everyday life how stable is this. Moreover, you know, I'm nub in git technologies, and have no time to understand ALL requirements to make pull request properly. The future shows.

s-hadinger, 2020-04-18:
@a-roz Thanks, I understand. I know just enough of git to survive :)

I understand that DMA requires to use GPIO3, which sacrifices Serial input. Have you tried non-DMA? I know there are lots of people stating that it doesn't work well, but I found this interesting blog post saying that you can relax some timings a lot: https://www.google.fr/amp/s/wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/amp/

a-roz, 2020-04-18:
@s-hadinger Thank you for the link to post. It's understandable. Answering your question - no, I did not try, and will not. Indeed, if CPU do not doing nothing except shifting bits in bit-bang manner, it can work, that's not surprised me. I can tell you, even with simple AVR processors  you can output stable video overlay to real-time CVBS video signal, but nothing else. I had this experience before. If we talking about realtime system with wifi processing, mqtt, sensors, rules etc, it's impossible to achieve smooth animation using bit-bang transmission. Quasi OS "Tasmota" works with external drivers only 1/4 of second (for every assigned drivers, not just xdrv_04_light), and light animation calls only every 50 ms inside this frame of 250 ms, mean 5 times for a second at most. (ONLY !) That's why the embedded Schemes are so poor, the are intended to use in bit-bang mode as well as DMA mode. To solve this puzzle have to change ideology of Tasmota OS, giving more time to light driver, sacrificing the rest. And also to change the library ws2812fx, using other approach for timing. This project I was need to finish fast and effective, what I've did. About **GPIO3**, using templates user can reassign Software Tx and Rx to any other free pins, this is not an argument.