arendst/Tasmota #5015: Set led color is not working with the last pixel

gabrielklein, 2019-01-23:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
I have a 16 pixels led WS2812

I configure the number of leds
Pixels 16

I send the following message: "/sonoff/Geiger/cmnd/Led1"
With the following content: 000000 000000 000000 000000 000000 000000 000000 000000 000000 000000 000000 000000 000000 070000 000700 000007

I expect the last 3 led to be set to red, green, blue.

It works perfectly except that the last led is not set to blue.

If I do a Pixels 17, it works perfectly - but I only have 16 leds.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ x ] _Searched the problem in issues and in the wiki_
- [ x ] _Hardware used_ : Esp8266 NodeMCU v3
- [ x ] _Development/Compiler/Upload tools used_ : Plateform IO, master branch then release
- [ x ] _If a pre-compiled release or development binary was used, which one?_ : sonoff.bin
- [ x ] _You have tried latest release or development binaries?_ : no
- [ x ] _Provide the output of command_``status 0`` :
```
00:39:26 CMD: status 0
00:39:26 MQT: /sonoff/Geiger/STATUS = {"Status":{"Module":18,"FriendlyName":["Geiger"],"Topic":"Geiger","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:39:26 MQT: /sonoff/Geiger/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"1T05:56:58","StartupUTC":"2019-01-22T17:42:28","Sleep":50,"BootCount":19,"SaveCount":95,"SaveAddress":"F5000"}}
00:39:26 MQT: /sonoff/Geiger/STATUS2 = {"StatusFWR":{"Version":"6.4.1.8(sonoff)","BuildDateTime":"2019-01-17T22:15:39","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:39:26 MQT: /sonoff/Geiger/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Klein","KleinAP"],"TelePeriod":15,"SetOption":["00000009","558180C0","00000000"]}}
00:39:26 MQT: /sonoff/Geiger/STATUS4 = {"StatusMEM":{"ProgramSize":519,"Free":484,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDAE794","240383A0","23B617CE","00003BC0"]}}
00:39:26 MQT: /sonoff/Geiger/STATUS5 = {"StatusNET":{"Hostname":"Geiger-0071","IPAddress":"10.0.0.227","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"5C:CF:7F:2B:E0:47","Webserver":2,"WifiConfig":4}}
00:39:26 MQT: /sonoff/Geiger/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.60","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2BE047","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:39:26 MQT: /sonoff/Geiger/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 23 23:39:26 2019","Local":"Thu Jan 24 00:39:26 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:05","Sunset":"17:23"}}
00:39:26 MQT: /sonoff/Geiger/STATUS10 = {"StatusSNS":{"Time":"2019-01-24T00:39:26","COUNTER":{"C1":143299},"BME280":{"Temperature":20.9,"Humidity":31.9,"Pressure":900.6},"PressureUnit":"hPa","TempUnit":"C"}}
00:39:26 MQT: /sonoff/Geiger/STATUS11 = {"StatusSTS":{"Time":"2019-01-24T00:39:26","Uptime":"1T05:56:58","Vcc":3.042,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":69,"Color":"AF0A00","HSBColor":"3,100,69","Channel":[68,3,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Klein","BSSId":"E8:D1:1B:15:E5:30","Channel":11,"RSSI":100}}}
00:39:26 MQT: /sonoff/Geiger/RESULT = {"Led1":"000000"}
00:39:29 MQT: /sonoff/Geiger/STATE = {"Time":"2019-01-24T00:39:29","Uptime":"1T05:57:01","Vcc":3.041,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":69,"Color":"AF0A00","HSBColor":"3,100,69","Channel":[68,3,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Klein","BSSId":"E8:D1:1B:15:E5:30","Channel":11,"RSSI":86}}
00:39:29 MQT: /sonoff/Geiger/SENSOR = {"Time":"2019-01-24T00:39:29","COUNTER":{"C1":143299},"BME280":{"Temperature":20.9,"Humidity":31.9,"Pressure":900.6},"PressureUnit":"hPa","TempUnit":"C"}
```

**To Reproduce**

I have a 16 pixels led WS2812

I configure the number of leds
Pixels 16

I send the following message: "/sonoff/Geiger/cmnd/Led1"
With the following content: 000000 000000 000000 000000 000000 000000 000000 000000 000000 000000 000000 000000 000000 070000 000700 000007

**Expected behavior**

I expect the last 3 led to be set to red, green, blue.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-01-24:
Verified. Will investigate.

arendst, 2019-01-24:
Should be fixed now in latest dev release (commit eab6be8).

gabrielklein, 2019-01-25:
Yes sorry. Thank you !

closed 2019-01-25 by gabrielklein