arendst/Tasmota #5017: SONOFF FIRMWARE 2.6.0

guile1994, 2019-01-24:
Hello, I tried to flash 3 units of sonoff basic and after that finish does not care anymore, what can it be? the first two realized the flash through the ARDUINO IDE and after 100% did not restart, it did not even call. At last I did a firmware backup through esptool.py and after it did not work anymore. Current version of the original 2.6.0 firmware.

Jason2866, 2019-01-24:
Erase whole flash with esptool. AFTER that flash Tasmota.
Just do it as described in wiki.

ascillato2, 2019-01-24:
If you need further assistant please find us in the Tasmota Chat. Github issues are for bug reporting. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-24 by ascillato2