arendst/Tasmota #5018: ssh console for remote serial debug and commands

wolfgangr, 2019-01-24:
Problem:
If we are short on RAM, and want to secure our installation, we have to drop the webserver.
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/my_user_config.h
```
  #error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
```
This ruins our web console.
When mqtt is not running properly, we have no more remote access for debug and configuration.
There are also situations where I'd like a remote debug console which does not interfere with mqtt.

I haven't find anything in the code, in the issues or in the wiki.

**Idea for solution**
Have a **ssh server running on tasmota** and redirect the serial console to that.

SSH is available on any client machine used to service network installations.
TLS - the successor of the SSL-protocol is built upon - is available in MQTT-TLS anyway. 
So I'd hope there were not much overhead in flash an memory usage.

Did anybody work on this already?
Any pointer is welcome.

wolfgangr, 2019-01-24:
https://www.wolfssl.com/products/wolfssh/
Oviously they offer a proprietary binary library.
The ESP8266 is not on their list of supported platforms
Will rise licensing questions as well.

https://matt.ucc.asn.au/dropbear/dropbear.html
>  runs on a variety of POSIX-based platforms

I think tasmota is not posix-based, right?

> suitable for memory-constrained environments – Dropbear can compile to a 110kB statically linked binary 

110 kB is still quite a figure.
There are features (X-forwarding, client agent forwarding, authorized keys) we do not need or even do not make sense in tasmota.

ascillato, 2019-01-24:
With secure mqtt there is no more ram left. Sorry. You should use mqtt as a console for commands.

ascillato2, 2019-01-24:
Anyway you are free to test it.

closed 2019-01-24 by ascillato2

wolfgangr, 2019-01-24:
> Both needs working wifi connection. 
> Mqtt is leightweight and the chance that it works is bigger than SSH...

mqtt needs a running server in a properly configured network.
And the server IP configured in the device.
And a mqtt client on the client machine.

ssh needs only an IP address which it might receive over DHCP.
So you could go with an generic config and do some kind "bootstrap" your config.

> closed this

never mind...

Jason2866, 2019-01-24:
Tasmota design goal is the use of mqtt...

ascillato, 2019-01-24:
@wolfgangr

I agree with you. Would be great to have other features working along with mqtt tls. But the problem is that now mqtt tls leaves no free RAM. So, the esp8266 will not be able to manage two encriptions at the same time. That is the only issue. 
Now, if you enable mqtt tls, you need to disable most sensors. If you enable some sensors (that uses few bytes of ram), mqtt tls does not work and the esp8266 restarts with an exception (not enough ram) 

If it is added ssh to work alone (no mqtt, and no webserver) It should work because all ram is available.

The issue with encription and the esp8266 is that all has to be done by code. There is no hardware encrypter/decrypter as for example the esp32 that it has dedicated hardware for that. So, in the case of esp32, all that is possible because it don't use ram.

wolfgangr, 2019-01-25:
Hi Adrian @ascillato  ,
> I agree with you. Your idea is cool. Would be great ... But .....

thanks :-)
So I feel encouraged to continue thinking together :-O

I have a **running setup** with **both MQTT TLS and web server**.
`Uptime	11T10:15:53`
Looks like there is **sufficient RAM** for that.
**Disabled anything** I did not know or did not have any use for:
https://github.com/wolfgangr/Sonoff-Tasmota/blob/LCT_noblock/sonoff/my_user_config.h
Not the "one binary for all", but I hink this is OK from a users point of view:
ssh is a requirement for development, and people who do this will be able to perform and own build

some further thoughts
- Do you have a rough idea what's the **RAM saving by dropping http?**
- And the amount of **RAM requirement of a second encrypted** connection?
- could we **save RAM** in the secure layer, e.g. by going to **shorter keys** or simpler algorithms or dropping default features?
( I thnik we protect against the casual hacker, not the physical intruder nor high intelligence)
- could we **reuse** allocated encrytion **RAM** for more than one connection?
basically boils to the question, whether encryption is done blockwise or bytewise, I think
- does TLS for MQTT as implemented now provide for more than one connection simultanously
(my guess is no - I see no necessity)
- and if we drop encryption for MQTT during development in favour?
sorry, that's a silly idea - I could as well drop encryption for ssh, rendering **plain telnet**

ascillato, 2019-01-25:
To use MQTT TLS with webserver you had modified several parts of the code. There are several keys checkings that don't allow both to be compiled in the same firmware. 

But Anyway, if you have that already working, there is no need for ssh, right? 

About your questions:

1- it is in my_user_config.h comments. 8K ram is used by webserver and 15k is used by MQTT TLS
2- at least 15K (as mqtt tls)
3- no
4- I don't think so. 
5- no
6- no

Remember that Tasmota is mqtt based for home automation. SSH is not the Tasmota core business.

wolfgangr, 2019-01-25:
Many thanks for your answers!
> Anyway, if you have that already working, there is no need for ssh, right?

well, I still have other ideas for the deployment of ram.... ;-)

sislakd, 2019-01-28:
I'm using TLS without WEB on many devices as well. For debugging, I use syslog where you can get any debug messages you need if syslog with debug is enabled. Syslog is not encrypted, but for debug purposes only it is sufficient. When done, you can disable syslog again. The only drawback compared to serial debug is that your network has to work (but for WEB or SSH you need it as well). Working MQTT connection is need as well otherwise you will loose ability to control your device (this is usually risky only if you change MQTT configuration, otherwise it is quite stable).