arendst/Tasmota #5020: IDE Setup for remote debugging

bubibubi, 2019-01-25:
Hi,
I installed Visual Studio Code and Platform IO so I can edit/build/upload/run tasmota on a NodeMCU (ESP8266).
The question is, is it possible to remote debug the code?

I've read that Platform IO can't debug ESP8266 code but some weeks ago I've read on a post of @arendst that he used VSCode+PlatformIO for remote debugging. I can't find that post anymore so I'm starting to think that I've never read it...

I also read that PlatformIO can remote debug ESP32 (I could not try it because I'm waiting for delivery from HK) but, I've also read that ESP32 porting of tasmota is not supported (and is not part of the tasmota master code).

ascillato, 2019-01-25:
Is it possible to remote debug the code?

Sorry, it is not possible in esp8266 AFAIK.

ascillato2, 2019-01-25:
I could not find any information to make that. Sorry.

closed 2019-01-25 by ascillato2

bubibubi, 2019-01-25:
About ESP32?

ascillato, 2019-01-25:
About IDE Setup for remote debugging for ESP8266.

Tasmota is for ESP8266