arendst/Tasmota #5021: add  a retain flag for the LWT topic

iomax, 2019-01-25:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
There seems to be no option to set the retain flag to the LWT topic 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
add the option to define retain flag to the LWT topic like PowerRetain, ButtonRetain, SwitchRetain and/or SensorRetain.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**
 
Please evaluate the option to add the option to configure the retain flag  to the LWT topic or, be patient Tasmota newbie here, please address me to the right option to set it to false.

Thanks

ascillato, 2019-01-25:
Hi,

By definition the LWT is a retained message to work. 

What is the use case of a non-retained LWT ? Why you need that?

iomax, 2019-01-25:
LWT messages are keep into mqtt server db forever and, play around several cards / config / firmware, many "temporary", or in any case not more valid, "devices" will still give their state (  "offline" of course )  to every new clients that will subscribe someway the LWT topic ( e.g. subscribing to the "general" # topic ).
Not a special uses cases now but I'd like to think to it as a configurable option to better match the needs case by case.

iomax, 2019-01-25:
You are right. The "problem" came when the devices will be definitively removed and not more connected due to whatever reason.
In this case, you still get "offline" info about it every time you will subscribing "#" topic.

ascillato, 2019-01-25:
Ok. I understand. If you have a device that will not longer be part of your network, you have the old LWT.

So, the solution for that is to delete that retained message sending nothing to that topic and the broker will delete it. 

A LWT without retain is not doing its purpose. I think that adding that option will confuse people as it can't be used for normal use. Sorry.

Anyway, thanks for sharing your ideas :+1:

PR #7744 (2020-02-16): MQTT_NO_RETAIN option for brokers that will drop connection otherwise
## Description:

The built-in MQTT broker within the Losant platform (and possibly others) will drop any connection that attempts to set a retain flag (https://docs.losant.com/mqtt/overview/#mqtt-version-and-limitations). This results in a continuous connected-dropped-connected cycle due to the LWT message.

This pull-request provides a (default disabled) compile-time option to suppress all MQTT retain flags, invoking the code provided for AWS IoT.

No command (SetOption) is provided. I figure this is an edge-case that can b [... cut]
closed 2019-01-25 by ascillato2