arendst/Tasmota #5035: Flashing TUYA Device - no control possible

nukleuz80, 2019-01-26:
Hi there,

today i'm flashing a tuya device (MAXCIO RGB Bulb) with Sonoff Tasmota. After that there is no control possible.
When i toggle between ON an OFF nothing happens...

Here are some information about it:

Maxcio RGB Bulb: https://www.amazon.de/Maxcio-Gl%C3%BChbirne-Kompatibel-Fernsteuerung-Szene-Modus/dp/B07JMR3ZZ9

Tutorial flashing: https://www.heise.de/ct/artikel/Tuya-Convert-IoT-Geraete-ohne-Loeten-vom-Cloud-Zwang-befreien-4283623.html

ChipID: e325fb
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipID: 1440a1
FlashChipSize: 1048576
FlashChipMode: 0

Sonoff-Tasmota Version: 6.4.1 -> running Basic Module and tried Generic an Tuya Dimmer.

Do you have any help for me please?

nukleuz

ascillato2, 2019-01-27:
Hi

Sorry, flashing with QIO does not work most of the time. You need to follow the Tasmota wiki for that. 

Please, read the wiki and if you need further assistant please find us in the Tasmota Support Chat. Github issues are for bug reporting. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-27 by ascillato2

federkicco, 2019-03-14:
I managed to make this type of bulb to work using Tasmota `6.4.1.21` (loaded for the template support) and using the following template:  `{"NAME":"Lyasi PT-BW09","GPIO":[17,255,255,255,143,144,0,0,0,0,0,0,0],"FLAG":0,"BASE":27}`