arendst/Tasmota #5039: Add MP3Player commands

AUyttendaele, 2019-01-27:
Hi the MP3Player is working fine but is lacking commands to be truly functional

Addition of Following basic commands would be great
NEXT_SONG | 0x01 | Play next song.
PREV_SONG | 0x02 | Play previous song.
SHUFFLE_PLAY | 0x18 | Playback shuffle mode. Data is 0 to enable, 1 to disable.

These would allow a basic MP3 player functionality.

The Folder commands would be nice but are not of the same importance.
PLAY_FOLDER_FILE | 0x0f | Play the file with the specified folder and index number
SHUFFLE_FOLDER | 0x28 | Playback shuffle mode for folder specified. Data high byte is the folder index.

Thanks

ascillato2, 2019-01-27:
Hi, 

This was already requeted in https://github.com/arendst/Sonoff-Tasmota/issues/3628
Anyway, I have just copied your request there. 

@mike2nl was working on that. You can ask him if you want in the Tasmota Support Chat

Thanks for sharing your ideas. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-27 by ascillato2

AUyttendaele, 2019-01-28:
Hi I read the other post but as it was closed I wrote this one. I see this more a request for enhancement rather than a closed issue.