arendst/Tasmota #5043: PZEM004T and display

fragolinux, 2019-01-27:
hi, is there PZEM004T support in the DISPLAY precompiled firmware? I'd like to add a 1602 i2c display to show actual values of the PZEM004T sensor...

or, how to know what features are enabled in standard precompiled firmwares?
Thanks

fragolinux, 2019-01-27:
nope, no pzem info in main page, even if setup as such:

[image]

this is status 0 output
```
13:40:36 CMD: status 0
13:40:36 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:40:36 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-IT.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:19","StartupUTC":"2019-01-27T12:40:17","Sleep":50,"BootCount":17,"SaveCount":39,"SaveAddress":"FB000"}}
13:40:36 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(display)","BuildDateTime":"2018-12-24T14:47:48","Boot":12,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
13:40:36 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Shark-HOME2","SharkAzzo2"],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
13:40:36 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":504,"Free":496,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FAAA59C","00038FA0","23B617CA","00003BC0"]}}
13:40:36 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2375","IPAddress":"192.168.1.241","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:C8:89:47","Webserver":2,"WifiConfig":4}}
13:40:36 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C88947","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:40:36 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 27 12:40:36 2019","Local":"Sun Jan 27 13:40:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:27","Sunset":"17:38"}}
13:40:36 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-27T13:40:36"}}
13:40:36 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-27T13:40:36","Uptime":"0T00:00:19","Vcc":2.952,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":76,"POWER":"ON","Wifi":{"AP":1,"SSId":"Shark-HOME2","BSSId":"C4:EA:1D:0C:19:51","Channel":8,"RSSI":100}}}
```

fragolinux, 2019-01-27:
while just switching from the DISPLAY edition to the standard (IT version) one, the pzem appeared on main page... so it seems i just need to compile my own firmware with both display and pzem support :)

[image]

arendst, 2019-01-27:
@fragolinux Correct!

To make the display binary as small as possible (prefferably < 500k) I remove the ENERGY_SENSOR functionality. You can find this in file sonoff_post.h lines 225 - 248

closed 2019-01-27 by arendst

fragolinux, 2019-01-27:
> @fragolinux Correct!
> 
> To make the display binary as small as possible (prefferably < 500k) I remove the ENERGY_SENSOR functionality. You can find this in file sonoff_post.h lines 225 - 248

why the display firmware allows selecting a not included device, anyway? Firmwares list any supported device in their combo boxes, even though not compiled in the installed version? Thanks, i'll post my setup once completed :)

arendst, 2019-01-27:
> why the display firmware allows selecting a not included device, anyway? Firmwares list any supported device in their combo boxes, even though not compiled in the installed version?

That's a matter of admin. I will add the (un)defines to solve this.

fragolinux, 2019-01-27:
ehm, need a little help with display... compiled binary, all fine, i've both display and pzem working now...
in console i get a line as such, every 5 minutes (ignore the 0 values, i've disconnected serial for now):
`15:45:13 MQT: tele/PZEM/SENSOR = {"Time":"2019-01-27T15:45:13","ENERGY":{"TotalStartTime":"2019-01-27T15:37:43","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}`
and added a rule as such (and enabled it), but nothing appears on display... i don't know how to extract the parts of the ENERGY field...
`rule1 on tele-PZEM-SENSOR#ENERGY do DisplayText [s1l0c0]%value% endon`

fragolinux, 2019-01-28:
adding photo so i'll create a wiki page with project

[image]

[image]

[image]

[image]

fragolinux, 2019-01-28:
created a page with all details: https://github.com/arendst/Sonoff-Tasmota/wiki/PZEM004T,-Wemos-D1-Mini-and-a-1602-I2C-display