arendst/Tasmota #5044: help in extracting data in json

fragolinux, 2019-01-27:
i need a little help with display values from a connected pzem004 energy monitor... compiled binary to have both display and pzem working, no problems here...

in console i get a line as such, every 5 minutes (ignore the 0 values, i've disconnected serial for now):

15:45:13 MQT: tele/PZEM/SENSOR = {"Time":"2019-01-27T15:45:13","ENERGY":{"TotalStartTime":"2019-01-27T15:37:43","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

and added a rule as such (and enabled it), but nothing appears on display... i don't know how to extract the parts of the ENERGY field...

`rule1 on tele-PZEM-SENSOR#ENERGY do DisplayText [zx0y0]%value% endon`

i'd like to display on 1st line voltage and current, and on 2nd one power and energy, if possible... don't know how to extract those sub-parts of the ENERGY json object...

Jason2866, 2019-01-27:
You will find some working rules examples here https://github.com/arendst/Sonoff-Tasmota/files/2365181/DisplayText.docx
Please respect this is not a forum. This is github issues for Tasmota Not the right place....
If you have questions please visit us at Discord Chat. Thx.

fragolinux, 2019-01-27:
> You will find some working rules examples here https://github.com/arendst/Sonoff-Tasmota/files/2365181/DisplayText.docx
> Please respect this is not a forum. This is github issues for Tasmota Not the right place....
> If you have questions please visit us at Discord Chat. Thx.

it's the second time today that you answer without actually answering at all, and pointing to non relevant stuff: before to release notes which lacked what i asked (features embedded in "display" edition), and now to the display documentation while i'm asking about json data extraction... thanks for pointing me to the discord, i'll join it right now, but please read what people asks before answering as such...

Jason2866, 2019-01-27:
Why offend? In the linked document are examples how json parsing is done...

fragolinux, 2019-01-27:
not the kind of extraction i'm searching, a json object which is a values of an other object, otherwise i'd not asked... i'm on discord right now

fragolinux, 2019-01-27:
in case anybody needs it, this is how to display the 4 values i needed from that json: 
`rule1 on tele-ENERGY#Power do DisplayText [z] [x0y0]%value%W endon on tele-ENERGY#Today do DisplayText [x8y0]%value%Wh endon on tele-ENERGY#Voltage do DisplayText [x0y1]%value%V endon on tele-ENERGY#Current do DisplayText [x8y1]%value%A endon`