arendst/Tasmota #5047: Multiple AM2301

matija84, 2019-01-27:
Hello,
I connect 5 AM2301 but only 3 sensors appeared on startup screan and console say:

20:51:59 MQT: tele/d1mini2/SENSOR = {"Time":"2019-01-27T20:51:59","AM2301-00":{"Temperature":24.2,"Humidity":46.6},"AM2301-01":{"Temperature":25.5,"Humidity":44.3},"AM2301-02":{"Temperature":24.3,"Humidity":53.7},"TempUnit":"C"}

how to se all 5 or more AM2301 connected sensors?

ascillato2, 2019-01-28:
Please, complete the troubleshooting template in order to have more information for properly help you. Thanks.

ascillato2, 2019-01-28:
Anyway, 3 sensor is the maximum amount of supported sensors in Tasmota. Sorry.

closed 2019-01-28 by ascillato2

sublime24, 2022-10-21:
Hello, Sorry to reopen so late. I am setting up an esp8266 with tasmota and trying to connect the max amount of AM2301 sensors. I currently only see 4 of the 5. Is Do you know the max connections?

barbudor, 2022-10-21:
Max is 4

barbudor, 2022-10-21:
I don't see any reason for not making this overridable for user compiled binaries

barbudor, 2022-10-21:
Once the PR is merged you can compile from development branch with a `#define DHT_MAX_SENSORS 5` or any value you want (reasonable of course) in your `user_config_override.h`

matija84, 2022-10-22:
Hello, better solution is to use ESPHome and you can connect 10 or even
more AM2301 sensors on D1 Mini. https://esphome.io/
Hers is code:

esphome:
  name: "mikroklima"
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "********"
  password: "********"
web_server:
  port: 80
sensor:
  - platform: dht
    pin: GPIO12
    temperature:
      name: "temperatura1"
    humidity:
      name: "vlaga1"
    update_interval: 20s
    model: AM2302

 - platform: dht
    pin: GPIO13
    temperature:
      name: "temperatura2"
    humidity:
      name: "vlaga2"
    update_interval: 20s
    model: AM2302

  - platform: dht
    pin: GPIO14
    temperature:
      name: "temperatura3"
    humidity:
      name: "vlaga3"
    update_interval: 20s
    model: AM2302

  - platform: dht
    pin: GPIO15
    temperature:
      name: "temperatura4"
    humidity:
      name: "vlaga4"
    update_interval: 20s
    model: AM2302

  - platform: dht
    pin: GPIO16
    temperature:
      name: "temperatura5"
    humidity:
      name: "vlaga5"
    update_interval: 20s
    model: AM2302

sublime24, 2022-10-22:
Thank you all for your responses. Quick questions.
1) tasmota can now be complied to support multiple sensors, greater than 4? @barbudor
2) What is the Benifit of using D1 mini and esphome? Just simplicity in configuration with esphome? 
In the past I've used esphome with a good amount of success. The only issue I found was when using temp sensors configured via esphome, hassio would sometimes get stuck if the sensor did not provide a temperature. I found tasmota is a little more stable.

barbudor, 2022-10-22:
1. Yes, but you need to self compile with `#define DHT_MAX_SENSORS 5` in your user_config_override.h and use the development branch, not the release

2. This is a Tasmota forum so I would say that Tasmota is best 🤣 
I don't like the approach of esphome to require compiling a dedicated binary for each device
With Tasmota you may have to compile a binary for some specific features but you can generally have a single binary for all of your needs and just do configuration.

So ESPhome is different but in no way better
It's generally better for maintenance to have all of your devices running the same firmware

sublime24, 2022-10-22:
@barbudor All good points. I agree. Going to attempt with 10 sensors. Thank you for the information!

sublime24, 2022-10-23:
I have created a tasmota .bin with the overrides. I connected 8 sensors on pins D0-8. 7 of the 8 work but pins D7 and D0... rx,tx do not work give me a reading. Any ideas?

barbudor, 2022-10-23:
GPIO16 (D0) is a very special pin lacking the pull-up feature, I would not be suprised that this prevent this pin to be useable

GPIO12 (D6) should be ok. I don't see any reason for it not to work
Can you try with only 1 sensor connected to that pin ? (and remove AM2301 configuration for the other pins)

sfromis, 2022-10-23:
On a dev board, TX/RX are connected to the USB/serial adapter chip, with serial communication being active low, I'd not expect those two gpio to be unencumbered when it comes to connecting other stuff.

sublime24, 2022-10-23:
@barbudor I removed all sensors and set the configuration to none, except for D6. The sensor shows null. @sfromis I agree, terminate just showed gibberish makes sense.

barbudor, 2022-10-23:
No idea why GPIO12/D6 doesn't work
Can you try with another board ?

For TX/RX, you should be able to use them as far as you do not connect USB and power the board from VIN/+5V pin

sublime24, 2022-10-23:
restored from one board to a new board and all pins work D1-D8. When rt/tx pins in use the board acted funny. So just left rt/tx none.