arendst/Tasmota #5048: OTA firmware not working after arduino IDE Flash

postaanerik, 2019-01-27:
Hi,

I'm having difficulties upgrading sonoff devices if they are initially flashed via the Arduino IDE
( flash size, flash mode, iWip are set correct )
The initial flash is successful (all functions working properly), but after a firmware upgrade the sonoff device is unresponsive, no LED, no serial out.

( when I flash using the ESPEasy tool with a Bin file there is no issue with OTA)

My new sonoff basic devices are delivered with esp8285 so I have to use the Arduino IDE

Any help is welcome.

Thanks,

Erik

jziolkowski, 2019-01-27:
No, you don't have to use Arduino IDE to flash ESP8285. Esptool works fine. Follow these instructions:
https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool

ascillato, 2019-01-28:
May be you have flashed with a different memory map, so ota will not work. Please, follow the wiki.

If you need further assistant please find us in the Tasmota Support Chat. Github issues are for bug reporting. Thanks

closed 2019-01-28 by ascillato2