arendst/Tasmota #5050: Issues connecting to Linksys router with DD-WRT

apastuszak, 2019-01-28:
I swapped out my router for a Linksys WRT3200ACM with DD-WRT installed on it.

On the router I configured the 5 Ghz band to be N/AC-Only
I configured the 2.4 Ghz band to be mixed and Dynamic 20/40.

I then created a virtual interface on the 2.4 Ghz network and called it "SonoffNET" to remove any spaces in the name and gave it a password.

I rebooted the router, and I cannot get it to connect to either network, the 2.4 Ghz network, or the virtual interface I created for it.

Are there recommended settings for DD-WRT I should use to get Tasmota to connect to it?

ascillato, 2019-01-28:
Sorry, there isn't. That seems to be an issue with your router and the esp8266 core.

You can try with Core 2.5.0

ascillato2, 2019-01-28:
You can also ask in the gitter chat from arduino repository, may be they could help you with that as the wifi connection protocol is beyond Tasmota and it is handled by the core and the sdk.

Jason2866, 2019-01-28:
Have you asked for help from the DD-WRT group? 
This issue isnt a Tasmota problem. 
Either it is Arduino ESP8266 core or DD-WRT or both....

ascillato2, 2019-01-28:
Closing this issue as it isn't a Tasmota bug. Anyway, if you find a solution, please share it for other users using that. Thanks. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-28 by ascillato2

apastuszak, 2019-01-28:
I found the solution!  I went to the SSID I made, SonoffNET and disabled WMM under the advanced settings.  All the Tasmota devices were then able to connect to the network!

Please update the wiki, or can I do that?

Jason2866, 2019-01-28:
You can update the wiki

apastuszak, 2019-01-28:
It is on by default.  According to the D-WRT forums it needs to be on for N and faster.

Jason2866, 2019-01-28:
This is nonsense. In general it slows down wifi performance.
Just google and you will find many issues related to enabled WMM....