arendst/Tasmota #5053: KNX Communication - uncalled UDP response

HomeAsASystem, 2019-01-28:
I'm using Tasmota 6.4.1 with KNX integration on several ESP8266 based devices. ESP Platform is 2.4.2. I have a full featured KNX TP installation with a MDT KNX-IP Gateway as well.
The integration of the Tasmota devices with the KNX TP installation works quite well. 
However there is a strange behaviour of the ESP devices. 
Whenever my IP Gateway or any other device sends a KNX IP message to the KNX Multicast address, the Tasmota devices respons with an UDP Message with 0 length to the KNX multicast address.
[image]

I've even isolated two ESP devices in a private WIFI net without any other connections, but the same behaviour was shown.
It seems that this is related to the IGMP implementation of the ESP platform, but sofar I couldn't locate the source of the uncalled UDP reponse.

All this causes a lot of additional traffic on the network an confuses the KNX binding of my OpenHAB server.

Tasmota 6.4.1 is used.

Jason2866, 2019-01-28:
Could you try http://thehackbox.org/tasmota/020500/sonoff-knx.bin
if this behaviour is in the upcoming core 2.5.0. It is in Beta3 stage and runs stable with Tasmota

HomeAsASystem, 2019-01-28:
This didn't solve the issue. 
used 

Program Version 6.4.1.12(e63741a-knx)
2019-01-28T18:13:42
2_5_0_BETA3/3.0.0-dev(c0f7b44)

[image]

As long as the device hasn't send KNX data, it responds to IP 0.0.0.0 Port 0. Only if it send once data to the KNX multicast address the UDP messages replies to the KNX mutlicast address
[image]

ascillato2, 2019-01-29:
Closing this issue as it is not a Tasmota bug. It is from the core. For follow up see referenced issue.
Thanks for sharing your tests.

closed 2019-01-29 by ascillato2

d-a-v, 2019-01-29:
@ascillato which version of lwIP is selected in 6.4.1.12 (described above) ?

BBBits, 2020-02-18:
Sorry to dig this up again, I am having the same problem.
In WiFiUDP::beginMulticast if I comment out 
    if (igmp_joingroup(interfaceAddr, multicast)!= ERR_OK) {
        return 0;
    }
The problem goes away.
Can you give me a heads up on how to use a lwip lib in PIO that doesn't have this issue ?

HomeAsASystem, 2020-02-18:
The issue is still  for me. Even with the later Tasmota Version using the 2.6.1 core the effect remains. So I’m still working on core 2.3.0.

7.2.0(KNX)
2019-12-22T11:41:52
2_6_1/2.2.2-dev(38a443e)

BBBits, 2020-02-18:
ok , 
so core 2.3 is ok then ?