arendst/Tasmota #5055: Cross-lock Interlock

meingraham, 2019-01-28:
First, thank you for the new TASMOTA Interlock feature!

**Have you looked for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
A relay cannot be part of two different interlocked relay groups.

The new `Interlock` feature ported from this fork over to TASMOTA main allows great flexibility in defining the groups of relays to be interlocked via the software configuration. One can define multiple groups and can define which relays are part of each group. However, a relay cannot be part of two different interlock groups.

**Describe the solution you'd like**
In my use case, I need to be able to interlock a relay individually with two other relays as separate interlocking groups. I have a ceiling fan speed setting scenario where:

- Low - 1 is on by itself
- Medium - 1&2 are on together
- High - 3 is on by itself

In this scenario, relay 1 can act independently from relay 2. However, relay 3 should never be latched when EITHER relay 1 or relay 2 are latched. Using the current `Interlock` syntax, the groups would be defined as:

`Interlock 1,3 2,3`

In other words, define the interlock groups as follows:

A. Relay 1 & 3
B. Relay 2 & 3
C. Do not interlock Relay 1 & 2 

`Interlock 1,2,3` won't allow 1&2 to be on at the same time, which the circuitry in the fan requires for selecting Medium speed.

**Describe alternatives you've considered**
Use Interlock for one of the relay pairs and use Rules for the other.

**Additional context**
https://github.com/arendst/Sonoff-Tasmota/pull/5014#issuecomment-457644164
https://github.com/arendst/Sonoff-Tasmota/pull/5014#issuecomment-457684473

This enhancement will simplify the setup and make maintaining the configuration easier. My use case is likely but one of many others that other makers have in their environments.

Thank you for your consideration of this enhancement.

Mike

arendst, 2019-01-29:
Won't do as there are too many situations possible which each have to be managed in code. Use rules for your situation.

closed 2019-01-29 by arendst

meingraham, 2019-01-29:
Understood. I wasn't sure if there might be "concise" logic that would allow implementation of the bit-level required for interlock/cross-lock permutations for up to the 8 channels you have allocated for in TASMOTA.

Thought I'd ask. Nothing ventured, nothing gained ;-) Perhaps someone will have an epiphany and come up with an elegant solution sometime down the road.

Regards.

Mike

quasi8, 2019-08-23:
He just gave you the solution but it requires you actually doing work and apparently you want someone to do the work for you,

joba-1, 2019-08-23:
lol, after 7 months?

You obviously don’t know him or you would be aware that he is the one that does the work for others - piles of it.

Go over to the tasmota discord chat if you need proof.

Have a nice day :)

meingraham, 2019-08-23:
@joba-1 

Yeah, not sure who that post was directed to. If it was me, I ain't too proud to admit, he's right, I did want someone to do the work for me :wink: Primarily because I didn't have a clue as to how one might go about designing the logic, much less programming it.

But, if it was directed at Theo, then @quasi8 just earned himself a big 'ol red flag :wink:

Cheers!

Mike

quasi8, 2019-08-23:
Didn't mean for the confusion was directed at @meingraham , investing time and learning things can be very rewarding

meingraham, 2019-08-23:
@quasi8 

Thank you for the advice. I'll try harder.

Are you involved with Tasmota much?

I guess I fooled someone on this project into granting me these badges.
[image]

Below is a small sample of some of the contributions. You can scan the full wiki revision History for hundreds of other examples of my contributions to the project.
[image]
[image]
[image]
[image]

Regards.

Mike

P.S. I hereby award you the Flagged Badge.

ascillato2, 2019-08-23:
Issue locked due to lack of respect from @quasi8 to @meingraham 

This is a community doing a collaborative work on Tasmota. That type of behaviour is against the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

User reported to github.