arendst/Tasmota #5059: New Device: King Art - King Q4 Cover switch

seea2005, 2019-01-29:
I would like to control by MQTT two cover switch which are shown on ewelink as king Q4:

[https://es.aliexpress.com/item/EWeLink-App-UE-Reino-Unido-Wifi-Smart-Touch-cortina-interruptor-de-Control-de-voz-por-Alexa/32956776611.html?spm=a2g0s.9042311.0.0.3fc163c0VL8sRB](https://es.aliexpress.com/item/EWeLink-App-UE-Reino-Unido-Wifi-Smart-Touch-cortina-interruptor-de-Control-de-voz-por-Alexa/32956776611.html?spm=a2g0s.9042311.0.0.3fc163c0VL8sRB)

From ewelink we can:

- Open (state of device --> ON)
- Close (state of device --> OFF)
- Stop (state of device --> Pause)
- Set a position (attribute setclose --> between 0 and 100)

Don't know really what else I have to send to develop it.

That's the board:

[image]

[image]

Thanks!

Jason2866, 2019-01-29:
Sorry, same answer as in https://github.com/arendst/Sonoff-Tasmota/issues/5057#issuecomment-458489218

andrethomas2, 2019-01-29:
It's difficult to add support for these unless someone with coding experience can get their hands on a physical device to figure out how it could fit into Tasmota.

Closing it for now but users can still post here if there is more information forthcoming.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-29 by andrethomas2

mclaudiopt, 2019-04-25:
hi. anyone knows the correct wiring for flashing? there are 2 3.3v and 2 gnd :-(

DoraDeu, 2019-08-23:
Flashing is easy. At the last picture you see on the upper half the pins you need:
3.3V
RX
TX
GND
SW (is connected to GPIO0)

Then the normal procedure: Connect SW to GND at power on time and the device is in flash mode.
But I am still looking for the right setup of Tasmota for this device. (GPIO assignment and configuration).

AWiseAd, 2019-10-21:
I managed to flash the device with Tasmota like above. What I was able to discover:
1. There is a separate MCU taking care of most local functionality:
  - Control of the touch buttons
  - Steering the relays
  - Managing timing
  - Reporting the status
2. The MCU communicates throught Rx.Tx, with commands like:
  - AT+UPDATE="switch":"on","setclose":0  (open button 1 pressed, current cover position = 0)
  - AT+UPDATE="switch":"off","setclose":100  (close button 3 pressed, current cover position = 100)
  - AT+UPDATE="switch":"pause","setclose":91 (pause button 2 pressed, current cover pos =91)
3. The status of the cover is reported every ~10 sec, with messages as above.

As I flashed the device with tasmota I lost the communication from the ESP to the other MCU and have not been able to extract the commands to control the MCU. I expect it to be a similar command protocol (i.e. AT+.... ="switch":".....","setclose":...)  but trial and error did not prove it.

Is anybody with an 'original' device able to find the commands from ESP to the MCU?
Connection is like @DoraDeu wrote on Aug 23. Serial speed to 19200.

DoraDeu, 2019-10-21:
Sounds interesting!
Unfortunately I have just this device that I already flashed with Tasmota. Would it help to check if I saved the original firmware before flashing?

AWiseAd, 2019-10-21:
If you are able to flash the original again, you could read the Serial commands by sending the Ewlink commands and monitor the Tx/Rx

acrmelo, 2019-10-27:
I can not flash it and I have 4 devices with original firmware, can you help?
Tx and Rx just flash bit (5%) if I set baudrate to 19200.
I connect sw to GND. maybe I am flashing the separate MCU?!?

Thanks
[image]

Jason2866, 2019-10-27:
Probably, RESI is wrong.

AWiseAd, 2019-10-28:
Like @Jason2866 said, RESI is wrong.  SW is should be connected to ground to flash.  Also you should set the baudrate for flashing high. The co processor grabs the Rx/Tx line after ~30 seconds. The flashing should complete before that takes place. 

Before you flash them all, can you 'snif' the Rx/Tx commands between the co-processor and ESP.. ?

acrmelo, 2019-10-29:
What do you mean RESet1 is wrong? I connect SW to GND on boot (1 second only) and RES1 always to GND. Set baudrate to high 115200 but no flashing. 
My sonoff.bin file is less than 500kb and tried with DIO and DOUT no luck...
(PS: flashing sonoff bridge only works with DOUT and never with DIO, don't know why)

Setting back to 19200 and I see commands from MCU to ESP like this when I push touch buttons:

AT+UPDATE="switch":"off","setclose":0
AT+UPDATE="switch":"off","setclose":32
(...)
AT+UPDATE="switch":"pause","setclose":86
(...)
AT+UPDATE="switch":"on","setclose":100
AT+UPDATE="switch":"on","setclose":0

Tomorrow with more time, I will get a fresh device, configure it on ewelink and snif the comms from ewelink and come back and post the results.

PS: I'm newbie to this flashing scene, till now just flashed ok one esp01-relay, one sonoff bridge and one mini.

Thanks for your help

acrmelo, 2019-10-31:
The protocol I have found is something like this (I only capable of snif one comm at a time):

```
(power on)
ESP>AT+START[1B]

(send open from ewelink)
ESP>AT+UPDATE="sequence":"1572536577552","switch":"on"[1B]
MCU>AT+RESULT="sequence":"1572536188859"[1B]
ESP>AT+SEND=ok[1B]
(...window opening...)

(send pause from ewelink)
ESP>AT+UPDATE="sequence":"1572536601645","switch":"pause"[1B]
MCU>AT+RESULT="sequence":"1572536191808"[1B]
ESP>AT+SEND=ok[1B]
MCU>AT+UPDATE="switch":"pause","setclose":15[1B]
ESP>AT+SEND=ok[1B]

(send close from ewelink)
ESP>AT+UPDATE="sequence":"1572536766275","switch":"off"[1B]AT+SEND=ok[1B]
MCU>AT+RESULT="sequence":"1572536191808"[1B]
ESP>AT+SEND=ok[1B]
(...window fully closed)
MCU>AT+UPDATE="switch":"off","setclose":100[1B]

(set to 63% in ewelink)
ESP>AT+UPDATE="sequence":"1572536873586","setclose":63[1B]AT+SEND=ok[1B]
MCU>AT+RESULT="sequence":"1572537093019"[1B]
ESP>AT+SEND=ok[1B]
(... window openig to until 63%)
MCU>AT+UPDATE="switch":"on","setclose":63[1B]
ESP>AT+SEND=ok[1B]

(set to 80% in ewelink)
ESP>AT+UPDATE="sequence":"1572536873586","setclose":80[1B]AT+SEND=ok[1B]
MCU>AT+RESULT="sequence":"1572537093019"[1B]
ESP>AT+SEND=ok[1B]
(... window closing to until 80%)
MCU>AT+UPDATE="switch":"off","setclose":80[1B]
ESP>AT+SEND=ok[1B]
```

Notes:
1) At power on I don't know if START cmd is necessary
2) EOL is ESC char 0x1B
3) Sequences values are a timestamps but are wrong in this example, because I snif only one direction each time
4) Physical touch buttons on MCU send command as I stated before.
5) I will try to emulate this with tasmota but I still don't have a working device.

Hope this helps.

acrmelo, 2019-10-31:
Good news...
My flshing does not work because I defined on device a very small running time for curtains go to 100% and that was sropping flash procedure.
After flashing tasmota all commands worked ok...

Use this on console...
Baudrate 19200
SerialDelimiter 27

(turn on)
SerialSend2 AT+UPDATE="sequence":"1572542635565","switch":"on"
(turn off)
SerialSend2 AT+UPDATE="sequence":"1572542635565","switch":"off"
(pause)
SerialSend2 AT+UPDATE="sequence":"1572542635565","switch":"pause"
(move to XX)
SerialSend2 AT+UPDATE="sequence":"1572542635565","setclose":XX

The MCU try each 10s the last message waiting for a "SerialSend2 AT+SEND=ok".

Anyone knows how to create the final template and rules? And also process the MCU messages to update the tasmota state. 

Many thanks

voodooo82, 2019-11-27:
I have the same switch curtain, I've flash with tasmota and only solution for use this switch is with sonoff t1 3ch mode but only with button...

corvidus, 2019-12-01:
I went through much the same process as @AWiseAd, @acrmelo elo et al based on suggestions on the espurna board. Taking that information and reverse-engineering the PS-16-DZ driver (effectively removing non-relevant commands and changing the serial commands used) I knocked this together to replicate the Dimmer behaviour, but sending the shutter commands. It will need thorough checking by someone who knows these things better than me, but hopefully I've made it a little easier? 

Re: template, this is what I've been using in testing. Only change from PS-16-DZ is inverting the LED at GPIO13. You could probably re-do it based on Generic with the SendSerial commands sent as Rules, but I haven't got that far. 

{"NAME":"KingArt Cover","GPIO":[255,148,255,149,255,255,0,0,255,56,255,255,255],"FLAG":0,"BASE":58}

Hope that all helps. 

[xdrv_99_ka_cover.txt](https://github.com/arendst/Tasmota/files/3907786/xdrv_99_ka_cover.txt)

acrmelo, 2019-12-05:
Thanks for helping, can you send the precompiled .bin so I test it on my devices?

Thanks again...

acrmelo

acrmelo, 2019-12-07:
I worked a bit on the code and removed the button function and leave only de dimmer. Maybe if pause is really need can be done with rules. I replaced completely xdrv_19_ps16dz_dimmer.ino file with my code.

So I have my 5 kingart curtains with this new code. I use node red to connect to a dashboard and to google assistant.

Hope someone with more skills in tasmota to help a final version.

PS: Flash minimal tasmota first to make room for this 562kb .bin file.

Happy holidays 4u all. 
[ka-curtain.zip](https://github.com/arendst/Tasmota/files/3935640/ka-curtain.zip)

erelor, 2019-12-12:
I confirm the acrmelo firmware works very well also with openhab and MQTT! 
It will be great if this new device will be added to official Tasmota

voodooo82, 2019-12-12:
Is possible convert to .bin?

acrmelo, 2019-12-19:
Hi again,

@voodooo82 bin is in zip file.

Now I added a dummy relay to control pause, changing template to:
{"NAME":"KingArt Curtai","GPIO":[255,148,255,149,255,255,0,0,255,56,255,255,21],"FLAG":0,"BASE":58}

On console add this rule, and activate it:
rule1 on Power1#State=1 do backlog SerialSend2 AT+UPDATE="sequence":"1576774195000","switch":"pause" ; power1 0 endon
rule1 1 

I now can user MQTT to send dimmer and pause with:
/cmnd/<yourtopic>/dimmer <position>
/cmnd/<youttopic>/power1 1

voodooo82, 2019-12-19:
[image]
Command unknow

acrmelo, 2019-12-20:
Hi,

Comand starts in "rule1..." all to "endon"

`rule1 on Power1#State=1 do backlog SerialSend2 AT+UPDATE="sequence":"1576774195000","switch":"pause" ; power1 0 endon`

Good luck

voodooo82, 2019-12-31:
Thanks is very good 😍😍😍😍

jonhgaspar, 2020-02-01:
> I worked a bit on the code and removed the button function and leave only de dimmer. Maybe if pause is really need can be done with rules. I replaced completely xdrv_19_ps16dz_dimmer.ino file with [...]

is there a possibility to turn the LED's on and off the buttons?
Currently only WIFI is possible

acrmelo, 2020-02-05:
> is there a possibility to turn the LED's on and off the buttons?
> Currently only WIFI is possible

WiFi led is connected to ESP and can be controlled, but led buttons are controlled by the MCU and I think there is no way changed it.

jonhgaspar, 2020-02-06:
But the code works, my question was for example the LEDs of the 3 buttons only lit up at night. During the day they are off. It's a good idea but I don't know much about code.

jonhgaspar, 2020-02-06:
> WiFi led is connected to ESP and can be controlled, but led buttons are controlled by the MCU and I think there is no way changed it.

But the code works, my question was for example the LEDs of the 3 buttons only lit up at night. During the day they are off. It's a good idea but I don't know much about code.

erelor, 2020-04-07:
I noticed that if you move the curtain using buttons the dimmer value isn't updated. I use OpenHAB to control my 4 curtains, sometimes I ask to move a curtain but anything happen because if you move the curtain via touch buttons the stored Dimmer value in OpenHAB is different from the real curtain position. Could you confirm this?

pisti222, 2020-05-06:
HI, Timer is not working, because curtain moving alvays a same time. How can i calibrate?

pisti222, 2020-05-06:
> Hi again,
> 
> @voodooo82 bin is in zip file.
> 
> Now I added a dummy relay to control pause, changing template to:
> {"NAME":"KingArt Curtai","GPIO":[255,148,255,149,255,255,0,0,255,56,255,255,21],"FLAG":0,"BASE":58}
> 
> On console add this rule:
> `rule1 on Power1#State=1 do backlog SerialSend2 AT+UPDATE="sequence":"1576774195000","switch":"pause" ; power1 0 endon`
> 
> And activate it:
> `rule1 1 `
> 
> I now can use MQTT to send dimmer:
> `cmnd/<yourtopic>/dimmer <position>`
> 
> And to pause with:
> `cmnd/<youttopic>/power1 1`

How works the timer settings? My curtains is long and need more time about mooving.

BerniApple, 2020-05-12:
> I can not flash it and I have 4 devices with original firmware, can you help?
> Tx and Rx just flash bit (5%) if I set baudrate to 19200.
> I connect sw to GND. maybe I am flashing the separate MCU?!?
> 
> Thanks
> [image]

how did you put it in flashing mode?
i connected lt like this but nothing happen, it just boot up normally
3.3V -> 3.3v
RX -> TC
TX -> RX
GND -> GND
SW -> GND

pisti222, 2020-05-12:
> how did you put it in flashing mode?
> i connected lt like this but nothing happen, it just boot up normally
> 3.3V -> 3.3v
> RX -> TC
> TX -> RX
> GND -> GND
> SW -> GND

Hi,
Just cut RX and TX lane next to the "big hole". After that it will work to upload tasmota.

DoraDeu, 2020-05-12:
The pin setup looks good so I don´t see the reason why it does not go into flash mode.
Maybe there are different hardware version available but in my case I did not cut any lanes to flash the device.

BerniApple, 2020-05-12:
> Hi,
> Just cut RX and TX lane next to the "big hole". After that it will work to upload tasmota.

thanks for the quick reply, i tried cutting the lane next to the big hole but it does nothing anyway, other suggest?

DoraDeu, 2020-05-12:
If the original firmware is tuya compatible maybe OTA flashing would be an option.

BerniApple, 2020-05-12:
ok i got the mistake...

3.3V -> 3.3v
RX -> **RX**  instead of RX -> **TX**
TX -> **TX** instead of TX -> **RX**
GND -> GND
SW -> GND

DoraDeu, 2020-05-12:
But this would not prevent the device to go into flash mode. You are just not able to write the image to the flash memory.

BerniApple, 2020-05-12:
i thought it wasn't in flash mode because the buttons led were on... thanks for the help anyway.
what are the usable GPIO?

pisti222, 2020-05-12:
RES1 not need to connect. SW  pont to put 5 sec when power up, after not need SW in round point. Flash mode will be active. Check Rx and TX pont in the right connect not reverse.
Should it will work.

BerniApple, 2020-05-12:
> But this would not prevent the device to go into flash mode. You are just not able to write the image to the flash memory.

i thought it wasn't in flash mode because the buttons led were on... thanks for the help anyway.
dp you know what are the usable GPIO?

pisti222, 2020-05-12:
Led link is not sign the flash mode. You can see the data if try to upload new flash.
Ill take a pictures, but need to prepare a new swich.
If you cut RX -TX from the button it will work and you will se the comm leds in your flasher.
[image]

BerniApple, 2020-05-12:
> Led link is not sign the flash mode. You can see the data if try to upload new flash.
> Ill take a pictures, but need to prepare a new swich.
> If you cut RX -TX from the button it will work and you will se the comm leds in your flasher.
> [image]

i got it working and now esphome fw is running on it, do you know what are the usable GPIO?

pisti222, 2020-05-12:
> i got it working and now esphome fw is running on it, do you know what are the usable GPIO?

Yes.
{"NAME":"KingArt Curtai","GPIO":[255,148,255,149,255,255,0,0,255,56,255,255,21],"FLAG":0,"BASE":58}
Comunication need PS-16-DZ driver. Please checked flash file what **### acrmelo** made. You can find complete tasmota in the zip attachment.

BerniApple, 2020-05-12:
> Yes.
> {"NAME":"KingArt Curtai","GPIO":[255,148,255,149,255,255,0,0,255,56,255,255,21],"FLAG":0,"BASE":58}
> Comunication need PS-16-DZ driver. Please checked flash file what **### acrmelo** made. You [...]

what does that mean? do you have something like this?

GPIO0 | Touchpad #1 (inverted)
GPIO9 | Touchpad #2 (inverted)
GPIO10 | Touchpad #3 (inverted)
GPIO12 | Relay #1 and Touchpad #1 backlight
GPIO5 | Relay #2 and Touchpad #2 backlight
GPIO4 | Relay #3 and Touchpad #3 backlight
GPIO13 | Blue LED (inverted)

pisti222, 2020-05-12:
> what does that mean? do you have something like this?
>
> GPIO0 | Touchpad #1 (inverted)
> GPIO9 | Touchpad #2 (inverted)
> GPIO10 | Touchpad #3 (inverted)
> GPIO12 | Relay #1 and Touchpad #1 backlight
> GPIO5 [...]

That has no direct connection, just communicate RX and TX with touch chip which controll the button and relays. Protocol is under the PS-16-DZ mode.
You can not use direct GPIO connection.

pisti222, 2020-05-12:
> That has no direct connection, just communicate RX and TX with touch chip which controll the button and relays. Protocol is under the PS-16-DZ mode.
> You can not use direct GPIO connection.
> That is the comrutine:
> (turn on)
> SerialSend2 AT+UPDATE="sequence":"1572542635565","switch":"on"
> (turn off)
> SerialSend2 AT+UPDATE="sequence":"1572542635565","switch":"off"
> (pause)
> SerialSend2 AT+UPDATE="sequence":"1572542635565","switch":"pause"
> (move to XX)
> SerialSend2 AT+UPDATE="sequence":"1572542635565","setclose":XX

I do not know how works with ESP Home. I think Tasmota is similar and you can used the methode.

pisti222, 2020-05-13:
Timer settings process if somebody needs to calibrate curtain lenght:

After installed and connect the switch with the motor. For the first time power up the switch, it will automatically turn close the curtain to the maximum length
Press the touch interface mild pause button around 4seconds till the red background led come up and blinking, then press the open touch button, the curtain start to open, the switch start to calculate the time.
When the curtain track open completely, please press Open or Close button stop the time calculating. the switch record the races time.
Example as above photo, the 2meter track races close and open is 8seconds, so each time the switch open/close maximum time is 8+2=10seconds.
Remark: the switch MCU will plug additional 2seconds base on the calculating time, so if the user want the switch auto stop immediately and Accurate percentage curtain status interface on the mobile app. The user can stop the calculating when the curtain near close completely.

pisti222, 2020-05-13:
[image]

acrmelo, 2020-05-13:
> I noticed that if you move the curtain using buttons the dimmer value isn't updated. I use OpenHAB to control my 4 curtains, sometimes I ask to move a curtain but anything happen because if you move [...]

If I use buttons to close all to end, dimmer position is updated... (in configuration menu if set web log to debug I get this in console):
18:42:37 PSZ: Rcvd AT+UPDATE="switch":"off","setclose":100
18:42:37 SRC: Switch
18:42:37 CMD: Group 0, Index 1, Command "DIMMER", Data "100"
18:42:37 SRC: Light
18:42:37 MQT: stat/janela2/RESULT = {"POWER2":"ON","Dimmer":100}
18:42:37 PSZ: Send AT+SEND=ok
18:42:37 SRC: MQTT
18:42:37 CMD: Group 0, Index 1, Command "DIMMER", Data "100"
18:42:37 MQT: stat/janela2/RESULT = {"POWER2":"ON","Dimmer":100}

erelor, 2020-05-25:
> If I use buttons to close all to end, dimmer position is updated
I tried, but if I press a button to close or open the curtain anything happens. 
This is the log when I send the close command via MQTT. It's slightly different from your log. Maybe I'm not flashed the latest firmware release? I got the zip file from your Dec 7th post.

```
17:09:27 SRC: MQTT
17:09:27 CMD: Group 0, Index 1, Command "CHANNEL", Data "100"
17:09:27 MQT: stat/tasmota_566ED9/RESULT = {"POWER":"ON","Dimmer":100}
17:09:27 PSZ: Send AT+UPDATE="sequence":"1590426567014","setclose":100
```

asterixonline, 2020-06-17:
I know this topic is closed, but I can't get my King switch module in flash mode. Tried everything as described, but no luck :(. Anyone have an option? the switch does work on 220v. but on my flash units there are no leds blinking at all. seems dead, but 220v input works.

asterixonline, 2020-06-19:
> what does that mean? do you have something like this?
>
> GPIO0 | Touchpad #1 (inverted)
> GPIO9 | Touchpad #2 (inverted)
> GPIO10 | Touchpad #3 (inverted)
> GPIO12 | Relay #1 and Touchpad #1 backlight
> GPIO5 [...]

How did you flash the unit? i tryed cutting, but the board is not powered, also there are no buttons colored blue or red when connected to the flash unit. On 220v everything seems okay.

pisti222, 2020-06-20:
> I know this topic is closed, but I can't get my King switch module in flash mode. Tried everything as described, but no luck :(. Anyone have an option? the switch does work on 220v. but on my flash units there are no leds blinking at all. seems dead, but 220v input works.

Can you describe your problem? If you connect 3,3v and gnd you can see the buton colour. You have to cut lane after rx-tx connection before flash the system, and need to connect back when you are using tashmota. If you take short video we can help.

asterixonline, 2020-06-21:
> Can you describe your problem? If you connect 3,3v and gnd you can see the buton colour. You have to cut lane after rx-tx connection before flash the system, and need to connect back when you are [...]

If i connect 3.3v and ground it is not working at all. so no leds blink or are lit.
That's the strange thing.

So i hoped to revert back to original fw and then with the use of the new custom_component from [AlexxIT](https://github.com/AlexxIT/SonoffLAN), but i think i have to solder the cutted lanes. But how small is that. cutting is no problem :) but can't see how to connect it back :(

pisti222, 2020-06-21:
> If i connect 3.3v and ground it is not working at all. so no leds blink or are lit.
> That's the strange thing.
>
> So i hoped to revert back to original fw and then with the use of the new [...]

Where you cut, you can soldered back. Just clean the surface and use thin cable. Good if 2 lanes not cut in same point.

asterixonline, 2020-06-21:
> Where you cut, you can soldered back. Just clean the surface and use thin cable. Good if 2 lanes not cut in same point.

i managed to flash it, seems my ground cable didn't work.
now the fine resolder part :)

asterixonline, 2020-06-24:
> Where you cut, you can soldered back. Just clean the surface and use thin cable. Good if 2 lanes not cut in same point.

is it still posible to calibrate using the buttons on the switch?

pisti222, 2020-06-25:
> is it still posible to calibrate using the buttons on the switch?

You can calibrate timing, but you can not change button sensitivity. Please check my May 13 post.

asterixonline, 2020-06-25:
> Hi again,
>
> @voodooo82 bin is in zip file.
>
> Now I added a dummy relay to control pause, changing template to:
> {"NAME":"KingArt [...]

so if i don't use bin file, and just default tasmota, can i use these commands?

Thanks.

pisti222, 2020-06-25:
> so if i don't use bin file, and just default tasmota, can i use these commands?
>
> Thanks.

NO. Tasmota needs some modification. Important are:
PS-16-DZ
And add this settings:
{"NAME":"KingArt Curtai","GPIO":[255,148,255,149,255,255,0,0,255,56,255,255,21],"FLAG":0,"BASE":58}

[57 more replies]