arendst/Tasmota #5061: SonOff T1 3 Gang - Button LEDs won't turn OFF completely

MrAlester, 2019-01-29:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE
- [x] Hardware used : SonOff T1 3 Gang
- [x] If a pre-compiled release or development binary was used, which one? : Using firmware version 6.4.1
- [x] You have tried latest release or development binaries? : Nope
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

Hello, first of all thanks for all the hard work.

My plan is to change all the normal switches from my home to SonOff T1.  The only problem is the blue LED (from the buttons) being slightly ON when the relays are OFF. I'm not talking about the WIFI LED, i'm refering to the surface where you press to turn the relays OFF.

I already tried LedPower 0 and LedState Off, but those commands only disable the wifi led.

Wife won't accept the change until I "fix" that issue, since the led being slightly ON is "bothersome".

Thanks!

wongnam, 2019-01-29:
@MrAlester My Sonoff T1-US 2C and Sonoff T1-US 3C  are the sample situation, it's hardware design, but it's useful for me to find the correct position of the button in dark.

Jason2866, 2019-01-29:
Please close this issue because it is hardware related and not under the control of Tasmota.
Thx.

MrAlester, 2019-01-29:
I'm sad to hear that. Perhaps I will use tape to disable the leds.

closed 2019-01-29 by MrAlester