arendst/Tasmota #5062: Unable to Parse home assistant discovery json

phdelodder, 2019-01-29:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : magichome
- [x] _Development/Compiler/Upload tools used_ : esptool to flash, default firmware 2.5.0 beta3
- [x] _If a pre-compiled release or development binary was used, which one?_ : 2.5.0 beta3 6.4.1.11
- [x] _You have tried latest release or development binaries?_ :
- [x] _Provide the output of command_``status 0`` :
```
18:43:21 CMD: status 0
18:43:21 MQT: MagicHome-2/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Sfeerverlichting Slaapkamer"],"Topic":"MagicHome-2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:43:21 MQT: MagicHome-2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"2T09:46:28","StartupUTC":"2019-01-27T07:56:53","Sleep":0,"BootCount":51,"SaveCount":438,"SaveAddress":"F7000"}}
18:43:21 MQT: MagicHome-2/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.11(f5a2053-sonoff)","BuildDateTime":"2019-01-26T14:52:48","Boot":7,"Core":"2_5_0_BETA3","SDK":"3.0.0-dev(c0f7b44)"}}
18:43:21 MQT: MagicHome-2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DelodderVanTorre",""],"TelePeriod":300,"SetOption":["400A8009","558180C0","00000200"]}}
18:43:21 MQT: MagicHome-2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":551,"Free":452,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
18:43:21 MQT: MagicHome-2/stat/STATUS5 = {"StatusNET":{"Hostname":"MagicHome-2","IPAddress":"192.168.0.75","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"84:0D:8E:57:06:DF","Webserver":2,"WifiConfig":4}}
18:43:21 MQT: MagicHome-2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.38","MqttPort":1883,"MqttClientMask":"MagicHome-2","MqttClient":"MagicHome-2","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:43:21 MQT: MagicHome-2/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 29 17:43:21 2019","Local":"Tue Jan 29 18:43:21 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:24","Sunset":"17:42"}}
18:43:21 MQT: MagicHome-2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-29T18:43:21"}}
18:43:21 MQT: MagicHome-2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-29T18:43:21","Uptime":"2T09:46:28","Vcc":3.442,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"OFF","Dimmer":18,"Color":"0,0,0,45","HSBColor":"0,0,0","Channel":[0,0,0,17],"Scheme":2,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"DelodderVanTorre","BSSId":"7C:FF:4D:DC:CD:1F","Channel":11,"RSSI":56}}}
```

**To Reproduce**

setoption19 1

**Expected behavior**

json is parsable and understandable for Home Assistant

**Additional context**
_Add any other context about the problem here._

```
2019-01-29 18:36:45 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON 5706DF_LI_1: '{"name":"Sfeerverlichting Slaapkamer","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","bri_cmd_t":"~cmnd/Dimmer","bri_stat_t":"~tele/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","rgb_cmd_t"
    :"~cmnd/Color2","rgb_stat_t":"~tele/STATE","rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join(',')}}","fx_cmd_t":"~cmnd/Scheme","fx_stat_t":"~tele/STATE","fx_val_tpl":"{{value_json.Scheme}}","fx_list":["0","1","2","3","4"],"whit_val
    _cmd_t":"~cmnd/White","whit_val_stat_t":"~tele/STATE","white_value_scale":100,"whit_val_tpl":"{{ value_json.Channel[3] }}","uniq_id":"5706DF_LI_1","device":{"identifiers":["5706DF"],"name":"Sfeerverlichting Slaapkamer","model":"Generic"
    ,"sw_version":"6.4.1.11(f5a2053-sonoff)","manufacturer":"Tasmota"}'
```

**(Please, remember to close the issue when the problem has been addressed)**

jziolkowski, 2019-01-29:
Can you please post the actual payload sent from Tasmota?

phdelodder, 2019-01-30:
```
homeassistant/switch/5706DF_RL_1/config (null)
homeassistant/light/5706DF_LI_1/config {"name":"Sfeerverlichting Slaapkamer","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","bri_cmd_t":"~cmnd/Dimmer","bri_stat_t":"~tele/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","rgb_cmd_t":"~cmnd/Color2","rgb_stat_t":"~tele/STATE","rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join(',')}}","fx_cmd_t":"~cmnd/Scheme","fx_stat_t":"~tele/STATE","fx_val_tpl":"{{value_json.Scheme}}","fx_list":["0","1","2","3","4"],"whit_val_cmd_t":"~cmnd/White","whit_val_stat_t":"~tele/STATE","white_value_scale":100,"whit_val_tpl":"{{ value_json.Channel[3] }}","uniq_id":"5706DF_LI_1","device":{"identifiers":["5706DF"],"name":"Sfeerverlichting Slaapkamer","model":"Generic","sw_version":"6.4.1.11(f5a2053-sonoff)","manufacturer":"Tasmota"}
homeassistant/switch/5706DF_RL_2/config (null)
homeassistant/light/5706DF_LI_2/config (null)
homeassistant/switch/5706DF_RL_3/config (null)
homeassistant/light/5706DF_LI_3/config (null)
homeassistant/switch/5706DF_RL_4/config (null)
homeassistant/light/5706DF_LI_4/config (null)
homeassistant/switch/5706DF_RL_5/config (null)
homeassistant/light/5706DF_LI_5/config (null)
homeassistant/switch/5706DF_RL_6/config (null)
homeassistant/light/5706DF_LI_6/config (null)
homeassistant/switch/5706DF_RL_7/config (null)
homeassistant/light/5706DF_LI_7/config (null)
homeassistant/switch/5706DF_RL_8/config (null)
homeassistant/light/5706DF_LI_8/config (null)
```

jziolkowski, 2019-01-30:
I've checked 5706DF_LI_1 node and indeed the JSON parser (http://json.parser.online.fr/) reports missing closing } bracket et EOL. 

[image]

I've enabled setoption19 1 on one of my 6.4.1.9 devices and the output is slightly different:

`08:28:32 MQT: homeassistant/switch/7A3BC0_RL_1/config = {"name":"S20 Biurko Igor","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"7A3BC0_RL_1","device":{"identifiers":["7A3BC0"],"name":"S20 Biurko Igor","model":"Sonoff S2X","sw_version":"6.4.1.10(bda3126-sonoff)","manufacturer":"Tasmota"}, "~":"s20_igor/"}
`
[image]

Assuming you've copied the entire string, indeed the code needs to be checked.
@andrethomas2 I've checked recent commits and so no changes to HA AD code, could you please verify/advise?

arendst, 2019-01-30:
The code is fine. The string length is the cause of the problem. The failed JSON message is not only missing the "}" but also some additional text.

The total allowed MQTT string size is 893. That's exactly the number of characters received by Hass. 

To solve this issue latests Hass supports abbreviated JSON names which are supported by Tasmota too when SetOption19 is set.

So due to the Hass enormous JSON hunger use SetOption19 to make it satisfied by using abbreviated JSON names.

phdelodder, 2019-01-30:
Is the size going to be increased in the future?

arendst, 2019-01-30:
Nope

phdelodder, 2019-01-30:
Reduce "Friendly Name 1" was the solution

closed 2019-01-30 by phdelodder