arendst/Tasmota #5066: Anoopsyche JH-G01B1 modify S31 config

Lawrencezarb, 2019-01-29:
I have the same issue as per #5056. Following the instruction on that thread I tried to modify the S31 config in sonoff_template.h, but it had no effect.

I assume I am doing something wrong with the config.

Can someone point me in the right direction in what I need to change in sonoff_template.h.

The smart plug requires:

REL1 = GPIO14
Button1 = GPIO12
LED_INV - GPIO13

Lawrencezarb, 2019-01-29:
Yes, but the Sonoff basic does not include energy monitoring, which the S31 does, and that part works out of the box.

andrethomas, 2019-01-29:
Ah ok, then make the same changes on the S31.
```
  { "Sonoff S31",      // Sonoff S31 (ESP8266 - CSE7766)
     GPIO_KEY1,        // GPIO00 Button
     0,                // GPIO01 Serial RXD 4800 baud 8E1 CSE7766 energy sensor
     0,
     0,                // GPIO03 Serial TXD
     0, 0,
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_KEY1,        // GPIO12 Red Led and Relay (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Green Led (0 = On, 1 = Off) - Link and Power status
     GPIO_REL1,        // GPIO14
     0, 0, 0
  },
```

andrethomas, 2019-01-29:
Awesome, well done.

andrethomas2, 2019-01-29:
Closing this issue as it will not be merged as a new template but a workaround is possible as an interim solution.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-29 by andrethomas2

loglux, 2019-02-01:
Is it possible to get a firmware release or code that would be working on ANOOPSYCHE JH-G01B1?

loglux, 2019-02-02:
Many thanks for the config customisation code. I have managed to compile and flash it.

blakadder, 2019-03-19:
It is now added as a [template](https://blakadder.github.io/templates/anoopsyche_JH-G01B1.html)

loglux, 2019-03-19:
Great news! Thanks

stevel23, 2019-05-23:
Apologies if I have made a mistake here but the template does not seem to work. When I apply it, the relay and led work correctly but the main menu on the device web page does not display any energy monitoring information as it does with the S31 - is this correct and how can I access the energy info. Thanks

blakadder, 2019-05-23:
Which binary are you using?

stevel23, 2019-05-23:
6.5.0(release-basic)

meingraham, 2019-05-23:
You have flashed sonoff-basic.bin it appears. Flash sonoff.bin

stevel23, 2019-05-24:
HI - Same issue I'm afraid - took latest sonoff.bin from https://github.com/arendst/Sonoff-Tasmota/releases. Information page now shows 6.5.0(release-sonoff). If I select S31 I get poower information but no working relay. If I apply and activate the JH-G01B1  template from https://blakadder.github.io/templates/anoopsyche_JH-G01B1.html  I get working relay but no power information.

stevel23, 2019-05-24:
Hi - Did some playing around and I think I have fixed the issue it seems that in the template GPIO1 and GPIO3 are the wrong way round - to be specific if I change GPIO1 to CSE7766 TX (145) and GPIO3 to CSE7766 RX  (146) all seems to work OK. Thanks for your help

blakadder, 2019-05-24:
I changed the template. Thanks for the research

cflocke, 2019-10-03:
Hi - thanks a lot. These settings also worked for me for the JH-G01E Model!