arendst/Tasmota #5067: Flashing Luani HVIO Board

jannnfe, 2019-01-29:
Hello and especially hello to @tom211
I've been trying to get the ESP8266 HVIO board from Luani running for days. I can flash the ESP normally over the Luani board but if I turn off the "Flash Mode", the blue LED on the ESP flashes wildly and no reach to ESP. I do not think I have a soldering error, as I have paid close attention to it. How did you manage to install Tasmota?
I would appreciate an answer.
Thanks!

jziolkowski, 2019-01-30:
It's not a code issue. Please join our support chat over at https://discord.gg/Ks2Kzd4

andrethomas2, 2019-01-30:
Closing this issue as it is not a firmware/feature/bug related issue but a new device flash related issue.

Feel free to continue posting here when there is more information or visit the chat group for more guidance if need be.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-30 by andrethomas2